nodemcu/nodemcu-firmware #557: http server - setting headers

roccomuso, 2015-07-15:
In the simple http server example, how to send a response to the client setting the headers?

Where may i find a complete documentation?

TerryE, 2015-07-15:
@roccomuso, this is absolutely nothing to with NodeMCU.  This is straight HTTP protocol stuff that you will find in the RRCs, Wikipedia and many tutorials on this subject.  This isn't the place to seek this sort of help.

marcelstoer, 2015-07-15:
This issue list is for bugs/features of the firmware. Please ask general Lua/NodeMCU questions at http://www.esp8266.com/viewforum.php?f=17 or http://bbs.nodemcu.com/c/nodemcu.

zerog2k, 2015-09-04:
the example is here:
https://github.com/nodemcu/nodemcu-firmware/blob/master/lua_modules/http/http-example.lua#L30-L33

closed 2015-09-04 by TerryE