nodemcu/nodemcu-firmware #559: Copy lua file (*.lua or *.lc) to ROM when flashing

vudangngoc, 2015-07-17:
How can I create a folder and flash tool always copy file from this folder to ROM? I think this function is very helpful

TerryE, 2015-07-17:
Whilst nodeMCU is based on eLua, there are some material differences. One is the eLua supports a ROMfs the content of which you initialise though the build process (and this can include lua or lc files).  
Whilst the ESP8266 does have a true ROM, this contains the Espressif boot loader, but for normal execution, the firmware is loaded from the on-module Flash memory, not ROM or eeROM.  nodeMCU uses SPIFFS to provide application access to available SPI Flash.  This is blank at build and there currently isn't a mechanism for pre installing files onto this, so this is a feature request rather than a bug.   My current work around is that I have a PC-based Lua script which writes a short provisioning bootstrap onto my ESP modules, so that when they are powered on for the first time, they download their initial config from an RPi on my home network.

MarsTechHAN, 2015-07-17:
Yes, we can... But you need to add them when compilig the bin file at present. We are now trying to firgure out better way.

TerryE, 2015-07-17:
@MarsTechHAN, Martin, surely the easiest method would be to have a Lua utility which runs on the host and takes a directory as an argument, plus the base `0x10000.bin` file and produces a separate `esptool` 0xNNNN.bin file with the SPIFFS image in it?  That way you can run the tool to produce the image without needing to do your own custom build.

nodemcu, 2015-07-30:
pls checkout github.com/xlfe/spiffs

closed 2015-07-30 by nodemcu

vudangngoc, 2015-07-30:
https://github.com/xlfe/spiffs --> page not found

nodemcu, 2015-07-30:
opps, sorry
https://github.com/xlfe/spiffy

vudangngoc, 2015-07-30:
thank you for your information ^_^

jmattsson, 2015-07-31:
Doh, wish I'd found that one before I wrote one myself. Ah well, here's another option for you: https://github.com/DiUS/spiffsimg

kad, 2015-07-31:
spiffy looks good, however, is there easy way to get start address of SPIFS on running nodemcu device ?

TerryE, 2015-08-01:
As well as [DiUS's spiffsimg](/DiUS/spiffsimg) this also links to the original [SPIFS project](/pellepl/spiffs).  I note that this has quite a few updates and fixes since nodeMCU regressed its changes.  It might be worth reviewing these to work out which should be merged into nodeMCU.

jmattsson, 2015-08-03:
@TerryE The dev branch has SPIFFS 0.3.2 as of the other week merged in.

TerryE, 2015-08-03:
@jmattsson, sorry missed this one.  :flushed: