nodemcu/nodemcu-firmware #561: mqtt:publish callback not working correclty

doringaiu, 2015-07-17:
mqtt:publish( topic, payload, qos, retain, function(client) )
*function(client): callback fired when PUBACK received.* - doesn't work correclty

I have a piece of code:
tmr.alarm(0, 2000, 1, function()
       if(busyFlag == 0) then
       busyFlag = 1 
        m:publish("some topic", "some message", 0, 0, function(client)
          busyFlag = 0
        end)
        print(counter)
        counter = counter + 1
      end
      end)

The problem is that sometimes the callback function is not called when the PUBACK is received. The busyFlag remains always 1 and i can't publish again. 

How to solve this issue ?

marcelstoer, 2015-07-17:
If it's a dupe of #554 then please close this one.

doringaiu, 2015-07-17:
No .. In my case i don't receive puback or the callback function doesn't work as it should.

jasonhector, 2015-07-20:
As you mention, and as the docs state, the callback is only called on receiving a PUBACK. Now with QOS 0 in your publish msg, the broker wont send any PUBACK, and my guess is there is some timeout on waiting for PUBACKS, and when its not received in this time, the callback is called anyway. I saw timeout times of about 1 sec [which is really a long time for me]. 
My suggestion is to put the QOS to 1 in your publish messages. The time for the PUBACK to come back from the broker is tens of ms, which far outstrips the 1000 ms timout, so your code will run faster and be more responsive. The cost of this is one more network msg per publish.

doringaiu, 2015-07-21:
Thanks jasonhector. I set qos to 1 and removed the flags and now it seems to work.

closed 2015-07-21 by doringaiu