nodemcu/nodemcu-firmware #563: dev120: depending on number of selected modules build fails

marcelstoer, 2015-07-17:
Not to be impatient but just to keep track of bugs...

Depending on the (number of) modules you select in `user_modules.h` the dev120 build may fail with 

>  `section `.text' will not fit in region `iram1_0_seg'

``` 
xtensa-lx106-elf-gcc  -L../lib -Wl,--gc-sections -Xlinker -Map=mapfile -nostdlib -T../ld/eagle.app.v6.ld -Wl,--no-check-sections -Wl,--wrap=_xtos_set_exception_handler -Wl,-static -Wl,--start-group -lc -lgcc -lhal -lphy -lpp -lnet80211 -llwip -lwpa -lmain -ljson -lsmartconfig -lssl user/.output/eagle/debug/lib/libuser.a driver/.output/eagle/debug/lib/libdriver.a json/.output/eagle/debug/lib/libjson.a upgrade/.output/eagle/debug/lib/libupgrade.a platform/.output/eagle/debug/lib/libplatform.a libc/.output/eagle/debug/lib/liblibc.a lua/.output/eagle/debug/lib/liblua.a coap/.output/eagle/debug/lib/coap.a mqtt/.output/eagle/debug/lib/mqtt.a u8glib/.output/eagle/debug/lib/u8glib.a smart/.output/eagle/debug/lib/smart.a wofs/.output/eagle/debug/lib/wofs.a spiffs/.output/eagle/debug/lib/spiffs.a cjson/.output/eagle/debug/lib/libcjson.a crypto/.output/eagle/debug/lib/libcrypto.a dhtlib/.output/eagle/debug/lib/libdhtlib.a modules/.output/eagle/debug/lib/libmodules.a -Wl,--end-group -lm -o .output/eagle/debug/image/eagle.app.v6.out 
/esp-open-sdk/xtensa-lx106-elf/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: .output/eagle/debug/image/eagle.app.v6.out section `.text' will not fit in region `iram1_0_seg'
collect2: error: ld returned 1 exit status
make[1]: *** [.output/eagle/debug/image/eagle.app.v6.out] Error 1
make[1]: Leaving directory `app'
make: *** [.subdirs] Error 2
```
OK example: node,file,gpio,wifi,net,pwm,tmr,adc,uart
NOK example (see above): node,file,gpio,wifi,net,pwm,tmr,uart,cjson

Could this be the reason @vowstar disabled most modules in https://github.com/nodemcu/nodemcu-firmware/commit/0b3ac33eb1510c9a8955aa56dc2512b55b21b711?

TerryE, 2015-09-12:
Marcel, this is really symptomatic of a wider issue that we are struggling to square, and that I've really just come to get to grips with looking at Dev120.

*     The 0.95 SDK has a stable Lua build, but it also has a lot of bugs which the later SDK releases fix.
*     However, the later SDKs continue to add a lot more functionality and so with each release the footprint of available iram1 slot (.data .bss,) dram0 (dynamic RAM), irom0 (.text) and irom1(.irom0.text) utilisation increases.
*     Essentially we can't fit a functional 1.2 based Lua build within the memory limits of a 512Kb flash ESP8266, and its extremely marginal that we can fit one in a 1Mb or 4Mb Flash version because we are still crippled by the 32Kb iram1.  

If the SDK was open source then we might be able to make some trade-offs, but the 1.2 SDK takes up all by 20 bytes of iram1, so in practice attempting to compile *any* module which requires iram1 will cause the build to fail.

TerryE, 2015-09-13:
Marcel, I've added a summary of the 0x40100000 at this [gist](https://gist.github.com/TerryE/e8df8f3f100b971f1c68) or as an Openoffice Spreadsheet [here](http://files.ellisons.org.uk/eps8266/0x40100000-segment-SDK1.2-build.ods).  Looking through this list of RAM-cache .text segments, there's nothing that really leaps out as not belonging naturally in the irom0 slot.  It's all high frequency or low time latency stuff.  It's just that it adds up to 32K (less 20 bytes).  There's just too much of it for this architecture.  The ODS file includes a pivot table summary.  

The big ticket items are: 

*    the SDK libraries: libpp.a (13,862), libmain.a (4,720), libm.a (1,074), liblwip.a (1,032)
*    The gcc RTS: libgcc.a(7,528), libc.a (3,534)

and its hard to see how to cut any of this out.  Integer only might help if we removed *all* references floating point to avoid the associated support routines being pulled in, but this is probably a thankless task.  Likewise is we has the SDK opensource then there's almost certainly initialisation and error handling code on constant data that code be hoisted to flash, but we don't have this.  

Need to ponder what to do next.

marcelstoer, 2015-09-13:
Puhhh, doesn't look good.

> Need to ponder what to do next.

Do any of the NodeMCU core contributors have connections to Espressif?

TerryE, 2015-09-13:
Note sure.  Better than the wider community anyway. @nodemcu (zeroday) and @vowstar have gone pretty dark on this repository for the last 2-4 weeks (not sure why), so even if they have then we still need them to engage.  Looking at the [SDK 1.3 release notes](http://espressif.com/new-sdk-release/), the first bullet item is 

> Memory optimization to save 12KBytes!

so my instinct is to forget SDK 1.2 and move my dev120 release to build over 1.3 and see if this is usable.  Even 2-4K with Johnnie's patch should give us some wiggle room to get workable builds back up and running.  The 512Kb limit on the Flash on the common modules is still a big problem, but enven a 8mbit 1Mb Flash config is more than workable.

My only other concern is that a lot of the later ESP modules switch from QIO to DIO Flash to free up an extra 2 pins for GPIO.  Given than the firmware is effectively running out of Flash this effectively pretty much halves the MIPs of the chip.  But one issue at a time, I guess.

georgkreimer, 2015-09-13:
Hey guys, I've switched for my stuff to https://github.com/pvvx/EspLua which is a fork that's already using the 1.3.0 SDK. Maybe also worth having a look at.

TerryE, 2015-09-13:
Yup have a look at pvvx/EspLua#2.  It's worth taking a look at, but I would a little happier if Victor's approach was a little less anarchic.

georgkreimer, 2015-09-13:
Agreed. That's not the right approach. Will comment on the issue you created.

TerryE, 2015-09-13:
Back to the main topic, I've just built dev120 against SDK 1.3.  I've done no testing but there is a small improvement in the irom0 segment -- up from 20 byte free to 196.   The simplest low hanging fruit that we might attack is #635.  If we could define memcpy to c_memcpy, etc so that we don't include the gcc versions of  in the irom0  segment then we might save another 1K or so.

jmattsson, 2015-09-14:
I saw a comment somewhere from vowstar to the effect that he was [going to be?] away for a number of weeks. Can't remember where it was a I saw that though.

As for memcpy & friends, it's possible to achieve that (redirect them to ets_memcpy & friends) with some additional linker command line options. Last time I tinkered with it there didn't seem to be enough benefit in doing so, but with 1.3 that might have changed. Basically what you need is to add a whole bunch of -Wl,--defsym=strcpy=ets_strcpy entries early in LINKFLAGS_eagle.app.v6 variable in app/Makefile. By doing so you're forcing the linker to simply reference the ROM routines rather than pull in libgcc's versions.

TerryE, 2015-10-31:
I will be deleting dev120 tomorrow unless any of the committers objects.

closed 2015-10-31 by TerryE