nodemcu/nodemcu-firmware #565: GPIO NEW TABLE in dev096 doesn't match NodeMCU devkit

marcoskirsch, 2015-07-19:
I upgraded to dev096 and from ESP-01 to NodeMCU devkit v1.

A program I had that toggled GPIO0 and GPIO2 was sending values 3 and 4 respectively to gpio.write() as described in the documentation: https://github.com/nodemcu/nodemcu-firmware/wiki/nodemcu_api_en-(dev096)

 After I upgraded, things didn't work. If I send 0 and 2, the pins labeled D0 and D2 on the hardware would toggle. This makes sense to me and I never understood why we needed a GPIO TABLE to map the parameter to the hardware pin in the first place.

In any case, assuming I understand everything correctly, the documentation is incorrect. The GPIO NEW/OLD TABLEs should be deleted and the information under the functions needs updating.

Let me know if I am correct and I'll be happy to do a pull request.

MarsTechHAN, 2015-07-20:
= =I cannot understand your question...
```
PIN0->D0->GPIO16
PIN2->D2->GPIO4
PIN3->D3->GPIO0
PIN4->D4->GPIO2
```
Using PIN instead the GPIO will make the new user better understand, and also...
The most serious problem to change it is that it has already been a stangard setting, NodeMCU using it, Arduino for ESP8266 using it, and so on. And this pin map will be easier to use and design.

marcoskirsch, 2015-07-20:
I misunderstood. I thought D0 was the label for GPIO0 and so forth. Closing.

closed 2015-07-20 by marcoskirsch