nodemcu/nodemcu-firmware #566: loose WiFi after first TCP(http) connection to ESP 12E

kabauterman, 2015-07-20:
Hello,

I'm using ESP 12E Modules with the NodeMCU Firmware 0.9.6-dev_20150704

First, I have my code running on a ESP 201 Module so I think it is some compatibility issue between NodeMCU and the 12E Modul.

Each time I connect to the WiFi and try to open the website on the ESP, the WiFi dissapears and will not come back until I restart the modul.

I not had this issue with the ESP 07 and ESP 201 moduls, just the 12E will not handle the tcp connection.

dnc40085, 2015-07-20:
The esp-07, esp-12E and esp 201 all use the same chip "ESP8266EX"
The issue you're having sounds similar to issue #541, are they not the same?

kabauterman, 2015-07-20:
No it's not the same, after a reset the AP is back there again, I can connect but as soon as I try to open the IP in my browser, the WiFi is gone and I have to reset the Modul to bring the WiFi back again.

I do not have this problem on the 07 and 201 moduls they are working just fine like they should, but I bought 5x 12E modules and tought they will work in the same way, but it seems like some hardware incompatibility.

MarsTechHAN, 2015-07-20:
First can you take a picture of your using enviroment, including the power supply, the board and the serial port to USB moudule(if have).(Take a picture is the fastest way I believe). 

Second, trying to using the newest firmware(in dev120). To see weather the issue cotinuing appear.

Third, paste your core code there, and maybe there is a BUG inside.

NOTICED HERE
ALL THE NODEMCU DEV BOARD V2 USING ESP-12E, AND THEY ALL WORK FINE...

kabauterman, 2015-07-20:
It's nothing special, a usual bread board 3,3V power supply at a 12V 2A power brick.
CH340 USB to Serial Converter. Will make a picture later.

I wanted to try the dev120 but I can't find a .bin to download and the cloud build service will not make any dev120 builds anymore.

I will post the code and a picture later.

But if there is a bug in my code, why is it working on 201 and 07?

kabauterman, 2015-07-21:
I just made a picture of my esp, it is soldered to a pin breakout board.

[image]

Init.lua

wifi.ap.config({ssid="TotalControl",pwd="TotalControl"})
iopin = {[0]=3,[1]=10,[2]=4,[3]=9,[4]=2,[5]=1,[9]=11,[10]=12,[12]=6,[13]=7,[14]=5,[15]=8,[16]=0}
dir = iopin[0]
step = iopin[2]
slprst = iopin[16]
m0 = iopin[13]

ms = 1

chdir0 = iopin[4]
chdir1 = iopin[5]

gpio.mode(dir, gpio.OUTPUT)
gpio.mode(step, gpio.OUTPUT)
gpio.mode(slprst, gpio.OUTPUT)
gpio.mode(m0, gpio.OUTPUT)
gpio.mode(chdir0, gpio.INPUT, gpio.PULLUP)
gpio.mode(chdir1, gpio.INPUT, gpio.PULLUP)

dofile("server.lua")

server.lua

srv=net.createServer(net.TCP)
srv:listen(80,function(conn)
    conn:on("receive", function(client,request)
        local buf = "";
        local _, _, method, path, vars = string.find(request, "([A-Z]+) (.+)?(.+) HTTP");
        if(method == nil)then
            _, _, method, path = string.find(request, "([A-Z]+) (.+) HTTP");
        end
        local _GET = {}
        if (vars ~= nil)then
            for k, v in string.gmatch(vars, "(%w+)=(%w+)&*") do
                _GET[k] = v
            end
        end

		buf = [===[
			<h1> WLAN SLIDER CONTROL</h1>
			<form method="get"><input type="submit" name="enable" value="Enable"></form>
			<form method="get"><input type="submit" name="direction" value="Forward"><input type="submit" name="direction" value="Backward"></form>
			<form method="get"><input type="number" name="delay" value="1"><input type="submit"></form>
			<form method="get"><input type="submit" name="start" value="Start"><input type="submit" name="start" value="Stop"></form>
   ]===]

if(_GET.delay ~= nil)then
 ms = _GET.delay
 end

if(_GET.start == "Start")then
              tmr.alarm(1, ms, 1, function() 
                   gpio.write(step, gpio.HIGH);
                   gpio.write(step, gpio.LOW);               
                   tmr.wdclr()
              end )            

elseif(_GET.start == "Stop")then
               tmr.stop(1);

elseif(_GET.enable == "Enable")then
              gpio.write(slprst, gpio.LOW);              
              gpio.write(m0, gpio.HIGH);

elseif(_GET.direction == "Forward")then
              gpio.write(dir, gpio.LOW);

elseif(_GET.direction == "Backward")then
              gpio.write(dir, gpio.HIGH);                                           

        end
        client:send(buf);
        client:close();
        collectgarbage();
    end)
end)

MarsTechHAN, 2015-07-24:
I briefly review your program, here are some bugs inside and may cause fatal error.
Here is some advice.
1. put conn:close() out of callback function (may cause fatal error)
2.use attribute mark 'local' in every verbal that you wont use out of context.
3.compile serve.lua, and dofile('serve.lc')
4.replace
'name'..name
to
local buf = ['name',name]
and use function concat to combine

ESP12E works at DIO mode, so the speed will be a little bit slower.
And I strongly suggest add a 220uF TANT Cap on your power supply.

ctvarlan, 2015-07-24:
@kabauterman I have an identic power supply and never worked for ESP8266. I had to build my own able to give 2Amp bursts.

TerryE, 2015-07-24:
Well I use a bog standard 5V 2A USB plug and with a AMS1117-3.3V based DC-DC Module to drop Vcc to 3.3V and this works fine with a 220uF TANT Capacitor across Vcc to Gnd. The whole rig cost about $10 off eBay.

zerog2k, 2015-08-05:
+1 for suggestion that it could be power droop related. Definately put at least 100uF cap across vcc and ground at the module. I had problems with that very same breadboard powersupply. I think the AMS1117 chips are fake. The 5v AMS1117-5.0 shorted through, and I dont think the 3.3v reg was up to the job for even 500mA. I wound up replacing both regs with TI LM1117 equivalents and it's solid !

jmattsson, 2015-08-06:
Contrary to the official specs which say up to 400mA, we've measured up to ~700mA bursts here, and I wouldn't be surprised if it can go even higher under other circumstances.

kabauterman, 2015-08-06:
As I said, this power supply works with any other kind of ESP's. And the ESP12E I use works great with any other software then the NodeMCU.

I've even testet other power supplys, not 1117 based with up to 2A and it also faild after I open the webpage.

For now I will no longer use NodeMCU, I've switched to the Arduino IDE for ESP8266, wrote my code again and it works flawlessly and fine without any troubles. 

So why should it be the PSU if it happens with any kind of PSU? 
And why should it be the PSU if it works with ESP 07 and ESP 201 on the same PSU?
And why should it not be the Firmware (NodeMCU) if it also works with any other firmware?

closed 2015-08-06 by kabauterman