nodemcu/nodemcu-firmware #569: Can't "require('dht_lib')" in http://frightanic.com/nodemcu-custom-build/ images with dht module

borromeotlhs, 2015-07-21:
I created and flashed the int version of the firmware built by http://frightanic.com/nodemcu-custom-build/.  I included the dht module, and when I try to do:
```lua
DHT = require('dht_lib')
```
I get a failure that the module doesn't exist and that it can't load either the .lua or .lc extension either.

However, when I then do:
```lua
DHT = require('dht')
```
I can see that I get a pointer into the romtable.  I hope that this means that my require call _is_ working, and not an indication of yet another issue, however, I'd like to report that for whatever reason the Example code in the documentation will not work as-is.

Here is the details of the build:

NodeMCU custom build by frightanic.com
	branch: master
	commit: aaca524ce02fca3cccdeaf7a6ffd0c4640c84ecf
	SSL: false
	modules: node,file,gpio,wifi,net,pwm,tmr,adc,uart,cjson,crypto,dht
 	built on: 2015-07-16 06:02
  powered by Lua 5.1.4

Please advise as to whether the module mechanisms require an update to the naming scheme in the documentation, or if the dht source is out of step on automated builds such that 'dht_lib' is not actually being built.

Also, does the automated build interface hosted at frightanic.com have a help email?  I feel bad posting this seeming automated build error to this repo's issue tracker :(

borromeotlhs, 2015-07-21:
Also, if this is a simple documentation fix, request that all documentation move to the syntax:
```lua
DHT = require '<module_name>'
```
in preparation for lua 5.2 _if_ nodemcu decides to move to it :)

TerryE, 2015-07-21:
(a) The module name is `dht` so why should `dht_lib` work any more than `dht_what_the_f`?

(b) Why is this a nodeMCU issue?

borromeotlhs, 2015-07-21:
Thanks!,
a.) OK, just needed confirmation that the module _is_ 'dht'.  This indicates that this is potentially a misnomer in the Documentation at: https://github.com/nodemcu/nodemcu-firmware/blob/master/lua_modules/dht_lib/README.md <-- as to my assumption, please note 'dht_lib' for module, and that the file is dht_lib.lua, _and_ that the module name is define as
```lua
local moduleName = ...
```
which I currently am unable to decipher as to what this resolves to as I don't know the variadic args passed in when this is called (yet)

b.) https://github.com/nodemcu/nodemcu-firmware/blob/master/README.md state's that you can, optionally, utilize Online firmware custom build by Marcel. In light of information you provided, I can now also confirm that the online build tool is _not_ at issue (nor is it _your_ issue even if it were somehow wrong), though your documentation _is_. ;)

marcelstoer, 2015-07-21:
> Also, does the automated build interface hosted at frightanic.com have a help email?

Have you tried the email address you get build notifications from?

> I feel bad posting this seeming automated build error to this repo's issue tracker :(

Good for you ;-)

TerryE, 2015-07-21:
@borromeotlhs, my apologies for my ratty reply earlier.  It was uncalled for and I've deleted it.  The confusion is because there are two DHT modules: one is written in C an is at [app/dhtlib] (/nodemcu/nodemcu-firmware/tree/master/app/dhtlib).  This is the one that is included in the Frightanic build when you tick the DHT box, and is pathed in by a `require("dht")`.  The second is the [Lua version](/nodemcu/nodemcu-firmware/tree/master/lua_modules/dht_lib) which needs to be copied to your FLash filesystem, and once this is done, then you don't need the C option and this is pathed in by the  `require("dht_lib")`.

borromeotlhs, 2015-07-21:
No prob!,
You choose how to spend your time, and _any_ response is fine by me :)  BTW, I am reading your amazing post at: http://www.esp8266.com/wiki/doku.php?id=nodemcu-unofficial-faq to ensure I don't miss stuff like this.  For the life of me, I couldn't figure out why getTemperature was failing with 'nil' even though my requires succeeded. . .

Appreciate the heads up, and I definitely request that this 'issue' be closed as "user error" in light of the fact that I misunderstood what frightanic's build was including.

R/
TJ

@marcelstoer , I sent an email reply back to the build bot just to see that it wouldn't bounce.  I will definitely keep this in mind should I come across an _actual_ issue ;)

TerryE, 2015-07-22:
@borromeotlhs 

> I definitely request that this 'issue' be closed

It's a lot easier if you do this yourself.  Most of us don't have the "privileges" to do this for other's issues.

closed 2015-07-22 by borromeotlhs

ScheissSchiesser, 2015-12-04:
@borromeotlhs 
I apologize if this is not the proper forum, but I am having the same issue that you mentioned in your comment. I use the syntax

```lua
DHT = require('dht')
```
but `getTemperature()` returns `nil`

```lua
dht22 = require("dht")
dht22.read(PIN)
t = dht22.getTemperature()
```
returns
```
dht22example.lua:5: attempt to call field 'getTemperature' (a nil value)
```

So apparently the C module doesn't have the same commands (obviously).
OK, that problem is solved (sorry, this is sort of stream-of-consciousness style).

But now after loading the dht_lib.lua module, I run the following:
```lua
PIN = 2 --  data pin, GPIO2

dht22 = require("dht_lib")
dht22.read(PIN)
t = dht22.getTemperature()
h = dht22.getHumidity()

if h == nil then
  print("Error reading from DHT22")
else
  -- temperature in degrees Celsius  and Farenheit
  -- floating point and integer version:
  print("Temperature: "..((t-(t % 10)) / 10).."."..(t % 10).." deg C")
  -- only integer version:
  print("Temperature: "..(9 * t / 50 + 32).."."..(9 * t / 5 % 10).." deg F")
  -- only float point version:
  print("Temperature: "..(9 * t / 50 + 32).." deg F")

  -- humidity
  -- floating point and integer version
  print("Humidity: "..((h - (h % 10)) / 10).."."..(h % 10).."%")
end

-- release module
dht22 = nil
package.loaded["dht_lib"]=nil
print("module released")
```
which returns 
`Error reading from DHT22`

So I can't seem to get a reading. Now I am realizing this is definitely not the right place for this comment. But I will post it anyways in case someone else finds this the same way I did.