nodemcu/nodemcu-firmware #574: Web Server and BMP180

jpbarba, 2015-07-23:
This code is a web server that accept request to get data from a BMP180. The problem is
that sometimes the MCU restarts just when the sensor init is executed. There is a getData() function
that gets the data from the sensor. If I set an timer to executed every second there isnt any problem
This is my code

wifi.setmode (wifi.STATION) 
wifi.sta.config ("-", "-")
wifi.sta.connect()
data={}
t=tmr.now()
tmr.alarm(1, 1000, 1, function()
    if wifi.sta.getip()== nil then
        print("IP unavaiable, Waiting...")
    else
        tmr.stop(1)
        ip, nm, gw = wifi.sta.getip()
        print("Modo: " .. wifi.getmode())
        print("Dirección MAC cliente: " .. wifi.sta.getmac())
        print("Dirección IP cliente: "..ip)
        print("Máscara de red: "..nm)
        print("Dirección IP AP: "..gw)
        print("Tiempo en establecer conexión: "..(tmr.now()-t)/1000000 .. "." .. (tmr.now()-t)%1000000 .. "s")
    end
 end)

function getData()
    print("Obteniendo datos del sensor")
    bmp085 = require("bmp085")
    sda = 6 --GPIO12
    scl = 7 --GPIO13
    bmp085.init(sda, scl)
    -- Get string of temperature.
    data["temp"] = bmp085.getUT(false)
    print("Temperature: "..data["temp"].." ºC")
    -- Get number of temperature.
    p = bmp085.getUP()
    data["presion"]=tonumber(p)/100 .."."..tonumber(p)%100
    -- Don't forget to release it after use
    bmp085 = nil
    package.loaded["bmp085"]=nil
end

-- Las funciones relacionadas con los recursos 
requests = {}

requests['index'] = function(socket, req)
    socket:send("HTTP/1.0 200 OK\r\nContent-Type: text/html\r\n\r\n")
    socket:send("<h1>Bienvenido al Servidor HTTP</h1>\r\n")
    socket:send("<a href='info'>Ver info</a>\r\n")
end

requests['error'] = function(socket, resource)
    socket:send("HTTP/1.0 404 OK\r\nContent-Type: text/html\r\n\r\n")
    socket:send("<h1>El recurso "..resource.." no es accesible</h1>\r\n")
end        
    
requests['info'] = function(socket, resource)
    socket:send("HTTP/1.0 200 OK\r\nContent-Type: text/html\r\n\r\n")
    majorVer, minorVer, devVer, chipid, flashid, flashsize, flashmode, flashspeed = node.info();
    socket:send("<table border='1'>")
    socket:send("<tr><td>Chip ID</td><td>"..chipid.."</td></tr>")
    socket:send("<tr><td>Flash ID</td><td>"..flashid.."</td></tr>")
    socket:send("<tr><td>Flash Size</td><td>"..flashsize.."</td></tr>")
    socket:send("<tr><td>Flash Mode</td><td>"..flashmode.."</td></tr>")
    socket:send("<tr><td>Flash Speed</td><td>"..flashspeed.."</td></tr>")
    socket:send("<tr><td>Heap</td><td>"..node.heap().."</td></tr>")
    socket:send("</table>")
    
    socket:send(html)
end

requests['reset'] = function(socket)
    html="HTTP/1.0 200 OK\r\n"
    html=html.."Refresh: 5; url=/\r\n"
    html=html.."Content-type: text/html\r\n"
    html=html.."\r\n"
    html=html.."Reseteando ..."
    socket:send(html)
    --tmr.delay(5000000);
    --node.restart()
    tmr.alarm(1, 2000, 0, function()
        node.restart()
    end)
end

requests['sensor'] = function(socket)
    getData()
    socket:send("HTTP/1.0 200 OK\r\n")
    socket:send("Refresh: 5; url=/sensor\r\n")
    socket:send("Content-Type: text/html\r\n\n\r")
    socket:send("<table border='1'>")
    socket:send("<tr><td>Temperatura</td><td>"..data["temp"].."</td></tr>") 
    socket:send("<tr><td>Presion</td><td>"..data["presion"].."</td></tr>") 
    socket:send("</table>") 

end

requests['sensordata'] = function(socket)
    getData()
    socket:send("HTTP/1.0 200 OK\r\n")
    socket:send("Refresh: 5; url=/sensor\r\n")
    socket:send("Content-Type: application/json/\r\n\n\r")
    json={temp=data["temp"], presion=data["presion"]}
    socket:send(cjson.encode(json))
end

-- Trocea una cadena utilizando un separador
local function split(str,sep)
    local array = {}
    local reg = string.format("([^%s]+)",sep)
    for mem in string.gmatch(str,reg) do
        table.insert(array, mem)
    end
    return array
end

svr = net.createServer (net.TCP)
    svr:listen (80, function(conn)
    conn:on ("receive", function(socket, request)
        -- Partimos la peticiones en cadenas 
        local req_parts = split(request,"\n") 
        -- Partimos la primera cadena en partes
        local req_type = split(req_parts[1], " ")
        local method=req_type[1]; 
        local resource = string.sub(req_type[2],2)
        
        if(method=="GET") then
            
            if(resource=="") then
                resource="index"
            end
            if(requests[resource]==nil) then
                resource="error"
            end    
            -- ejecutamos la función asociada
            requests[resource](socket, resource)            
        end
    
        socket:close()
        collectgarbage()
    end)
end)

TerryE, 2015-07-23:
@jpbarba, This is a "why does my Lua code cause a reboot?" Q rather than "here is a bug in nodeMCU firmware.  This list is only for the latter.  Please close this call and ask the Q on a forum such as esp8266.com or StackOverflow. Thank-you.

closed 2015-07-23 by jpbarba

marcoskirsch, 2015-07-23:
@jpbarba for what you're trying to do maybe you should check out Project https://github.com/marcoskirsch/nodemcu-httpserver.

jpbarba, 2015-07-23:
Hello Marcos,
I have used it before and it worked. This was an experiment. I have test an nodemcu-httpserver but the performance is worst that a web server is implement for example with Arduiono IDE for ESP8266. I have tested with JMeter and the difference is important. I think this is normal because Lua VM is not the same that program directly in C/C++.

marcoskirsch, 2015-07-23:
Yeah, performance of nodemcu-httpserver is not great. I've seen it take several seconds to respond to requests and the times vary wildly. This is what you get for running Lua scripts on a web server written in Lua on a small microcontroller with very little memory. It is what it is.

jpbarba, 2015-07-23:
What Ive seen is that compiling the last version from rep only with needed modules, performance is better.

jpbarba, 2015-07-23:
Problems I had are related with manipulation of strings. 

data["presion"]=tonumber(p)/100 .."."..tonumber(p)%100

It is strange because when I use the function with a timer there is no problem.