nodemcu/nodemcu-firmware #577: Suggestions of how to make nodemcu project organized! call for help.

nodemcu, 2015-07-26:
well, It's not a good place to discuss here on a issue maybe:)
but I need some professional suggestions and help of how to make this project clean and organized,
as #449 discussed. and more to sort out.
1, what is the good/popular branches setup. the clearly documented contribution guide?
let people know how to build a stable/test or most new features bin.
2, how to sort out issues, what kind of issues can be opened and stay as a "real" issue here?
 some issues like this one, can it be posted here or some where else like forum?
3, I want to make this project a long-run project, but some times too busy for this, currently we still do part-time in this project. any suggestions of how to make this project a really good opensource iot project.
4, what is a good police for those cool guys who want to participate more?
5, for now me and others are from China, not English native speaker, some time it's a problem. anyone can help this?
thank you guys in advance.  any advise would help!

clixx-io, 2015-07-26:
Hello,

I've been working on that recently on my project clixx.io

Basically what I do is put every project into a directory off ~/IoT. Sorry
for the brief answer.

One of the big problems with Arduino is they don't give you a management
interface for your devices. This is something I feel that's needed for the
ESP8266.

I'm busy writing, and testing the documentation. But I'm just showing a few
of the ideas that I have been working on so far.

clixx-io, 2015-07-26:
probably this is a better shot of the web interface:

On Mon, Jul 27, 2015 at 1:25 AM, David Lyon <david.lyon.preisshare@gmail.com

nodemcu, 2015-07-26:
dir? web interface?
@clixx-io I can't quite understand you.
I mean how to run/manage this project. 
am I misunderstanding?

clixx-io, 2015-07-26:
Every ESP8266 project goes into a project directory under ~\IoT.

There's also a configuration file that goes in there to hold some values,
such as the MAC or IP address.

The web server sits on a desktop or embedded-linux device and just makes
the network management easier. It also logs and can graph sensor values.

Basically, what I'm developing this for is eLua development. More for
organising nodemcu elua projects and devices.

I will come back in a few days once the documentation is online.

This is entirely a new concept to manage ESP8266 devices and their source
code - hard to explain in a few words.

clixx-io, 2015-07-26:
On Mon, Jul 27, 2015 at 1:35 AM, zeroday <notifications@github.com> wrote:

> dir? web interface?
>
> Both. The source and configuration files sit in a directory.

Then when your esp8266 is 'in-production', you can easily jump over to it
from the web-interface from your mobile phone/tablet.

marcelstoer, 2015-07-26:
> @clixx-io I can't quite understand you.

You're not alone with this. I don't see how those comments are relevant in discussing the structure and organization of the NodeMCU firmware project/community.

> It's not a good place to discuss here on a issue maybe.

I agree, http://bbs.nodemcu.com/ would probably be more appropriate but since that forum is mostly Chinese I suspect many contributors aren't active there (myself included). You can still open a thread there, post a link to it here, and then close the issue if you like.

Anyway, thank you very much for reaching out to us. It's a relief to see some of the project owners realize the project may need some guidance, governance or rules in order to prosper and continue to be relevant. I have pretty clear ideas on how a GitHub project should be run and I'm happy to share them once you decide whether you want to continue the discussion here or elsewhere.

TerryE, 2015-07-26:
I feel that the project could benefit from some the knowledge / experience / contributions from the wider contributors outside the core node MCU dev team.  At the moment we have info spread across multiple repositories: this git source hierarchy, the wiki, the nodeMCU online docs, the nodeMCU hosted resources such as it bbs and third party resources such as those hosted on esp8266.com.  Some rationalisation and normalisation of these would help, e.g. around some git hierarchy so that the same git contribution techniques could be used across the entire set.

We should also properly separate out end-user support from firmware development, with end-user support left to the nodeMCU bbs, esp8266.com and StackOverflow.  Any issues which are user "howto" Qs should be immediately closed referring the OP to the above.

nodemcu, 2015-07-27:
@marcelstoer , thank you! 
for a wider participation, I will leave this discussion mainly here. 
the bbs need some registration. many interested people probably have a github account.
@TerryE yes, indeed. a good point! thank you!

MarsTechHAN, 2015-07-27:
As for the first problem, Guide is always a big problem for almost any opensource project. And the dealling ways are always the same. We can let the users to write them. 
Our BBS need to have a great change and need manger from users.
Address:
```
http://bbs.nodemcu.com
```
2.We need to have more labels for issue list. To sort them clearly, and have more manger.
......

So as for me, the problem is at 
<hr>
The guys who manage this project dont have enough time to maintain, but the guys who have time don't have authority to do so.
<hr>

The methods that I think will work are->
1.  Find some participants who both have time and enough knowledge to set up nuclear dev team who will in chagre to dealling the issues, arrange the road map, and making a lot PR.
2. Find some participants who maybe dont have time or do not have that much knowledge to set up a  alpha user team to write guides, give idea to nuclear dev team.
3.Find some participants who will enjoy and good at managing the community and have many resources to running the BBS, and releated things.

marcelstoer, 2015-07-27:
Ok, here we go...

> 1, what is the good/popular branches setup. the clearly documented contribution guide?

Actually that's two questions.

Support as few branches as absolutely necessary. You already found out that long-lived branches are a pain to maintain. IMO besides master you should only have dev as long-term branches. Everything else should be short-lived. If, based on the reported issues, there's a critical bug in master for example you can create a patch branch that only lives for a few days until everything related to that issue is resolved.

Guidelines for contributors are extremely important and GitHub has good means to make that very painless: https://help.github.com/articles/setting-guidelines-for-repository-contributors/ (links to good examples).

> 2, how to sort out issues, what kind of issues can be opened and stay as a "real" issue here?

I suggest you close all those "help, why doesn't this Lua code work" issues quite aggressively. There are currently 160 open issues and it's quite demotivating for anyone trying to help to sift through them to find real bugs he/she could work on. Direct I-need-help issues to www.esp8266.com/viewforum.php?f=17 or http://stackoverflow.com/tags/nodemcu.
Anything that requires a discussion among firmware developers/contributors could be started at http://www.esp8266.com/viewforum.php?f=23. Or you could setup a group/mailing list (e.g. with Google) for developers and contributors.

> 3, I want to make this project a long-run project, but some times too busy for this, currently we still do part-time in this project. any suggestions of how to make this project a really good opensource iot project.

Of course you could always read a book on the subject by people who know ([How to Run a Successful Free Software Project](http://producingoss.com/), even [in Chinese](http://producingoss.com/zh/)). Below are just my 2 cents.

Fact: most OSS projects are run by passionate folks part-time in their spare time. You need at least enough time to nurture the community incl. the core contributors or you should find new maintainers. Not every contributor has to be a top-notch C-coder. I, for example, am not but I can still work on infrastructure (Travis-CI), help out managing the issue list or write documentation.

In order to reduce the distraction (i.e. unproductive time) for contributors the project should be self-explanatory for both users and developers. Good documentation is crucial so people don't have to ask. What's the first thing visitors to the GitHub repo see? The README.md. It must contain all relevant information and must always be up to date _for the respective branch_ (same for CONTRIBUTING.md)!
Personally I find the API documentation in the wiki less than ideal as you can't easily support multiple versions or multiple languages. For me it's a must for APIs documentations that they're maintained together with the source code in the repository. With the same commit you change the API you must also change the documentation.
One such system is [Read The Docs](http://read-the-docs.readthedocs.org/en/latest/features.html). It supports localization. You keep the documentation as .md files in a dedicated folder in the repository and whenever those files change the documentation is updated automatically. When you merge from dev to master the documentation is automatically merged with it. You don't have to tell the developers to update the wiki once their PRs are merged as they can/have to provide the documentation _together_ with the PR.
Reference: #486, #472, #494 all relate to missing or outdated documentation.

What is also very important for me is that what you ship must be well tested and stable. If not, you spend a lot of time just managing the issues or answering questions when you want to rather spend your time on coding. I realize automated testing isn't easy for IoT projects (see #516). Therefore, manual verification is even more important. I'm sometimes irritated how quickly PRs are merged. I suggest you only merge a PR if you have thoroughly verified it yourself or 2-3 people you trust haven written an "ok" comment.

As suggested by @MarsTechHAN you should introduce issue labels (e.g. bug, feature, one per branch, one per module).

> 4, what is a good police for those cool guys who want to participate more?

Not sure what policy you refer to.

> 5, for now me and others are from China, not English native speaker, some time it's a problem.

In what way? As [Terry said](https://github.com/nodemcu/nodemcu-firmware/issues/449#issuecomment-109959656) the English natives are certainly willing to help out when it comes to proof-reading and correcting documentation. To answer issues or write comments your English is certainly good enough.

nodemcu, 2015-07-29:
Thank you @marcelstoer @MarsTechHAN 
Will leave this issue here for some more days, to learn more.

eyaleb, 2015-07-29:
A few days ago I responded [elsewhere #449](https://github.com/nodemcu/nodemcu-firmware/issues/449#issuecomment-125068534). I will not repeat it here...

nfriedly, 2015-08-03:
> 2, how to sort out issues, what kind of issues can be opened and stay as a "real" issue here?
some issues like this one, can it be posted here or some where else like forum?

I agree with what several of the others have said about issue labels and directing "help me with lua" sort of issues to other sites.

> 3, I want to make this project a long-run project, but some times too busy for this, currently we still do part-time in this project. any suggestions of how to make this project a really good [...]

Choose some people that you trust and who have already contributed a bit and offer to make them [collaborators](https://github.com/nodemcu/nodemcu-firmware/settings/collaboration) on the repo (or [members](https://github.com/orgs/nodemcu/invitations/new) of the nodemcu organization). Then they can help you with the burden of sorting through issues, validating PRs, etc. Anyone who replied to this thread is probably a good candidate. Maybe set up a irc/gitter/whatever for your newly promoted team to be able to discuss things.

One other point about docs, the [api page](https://github.com/nodemcu/nodemcu-firmware/wiki/nodemcu_api_en) is becoming awkward due to how many modules there are. I think it should be split into one page per module (regardless of whether they remain wiki pages or are moved to some other location).

devsaurus, 2015-08-03:
> One other point about docs, the api page is becoming awkward due to how many modules there are. I think it should be split into one page per module (regardless of whether they remain wiki pages or are moved to some other location).

I'm currently fiddling with api updates for u8g - and fully agree with this one.
Really enjoyed contributing to [mqttwarn](https://github.com/jpmens/mqttwarn/wiki), this is a good example for module-wise structuring.

marcelstoer, 2015-08-03:
> mqttwarn, this is a good example for module-wise structuring

[OT]
I partially agree. The module wiki is really nice but IMO it's detrimental to overall clearity if the wiki is replicated in README.md. Example: it's irritating one has to keep  https://github.com/jpmens/mqttwarn/wiki/carbon and https://github.com/jpmens/mqttwarn#carbon in sync manually. With such a setup the README.md should only contain _pointers_ to the wiki.

devsaurus, 2015-08-03:
Right, I stumbled over this as well.
Having the README.md for more technical details and keep the user reference in the wiki might be an approach. However, are there other examples (on github?) where we could get inspiration and learn from? Especially with respect to automated api documentation. I'm too lazy in general to maintain things twice or even more times....

marcelstoer, 2015-08-03:
>  However, are there other examples (on github?) where we could get inspiration and learn from?

Let me pick up here, even at the risk of potentially further diverting from the core topics.

My stance on API documentation in general is that it belongs into the source code. Why? Because you need to document the source code anyway, right? And then you _generate_ the official API documentation (HTML, PDF, whatever) from the source code during the build.

A great example for such a system is [Swagger](http://swagger.io/) which I use to document REST APIs in my non-NodeMCU world. The idea is that you annotate the source code (implementation of your REST API) and document it. Swagger then generates a fully executable documentation based on the annotated source.

TerryE, 2015-08-05:
I feel that we really really need to take a simple firm line on the scope of this issues list.  The active community has an aim of contributing to the development of the firmware, but the majority of the issues raised here are by Lua developers who keep asking Qs like "My app doesn't work, so is there a bug in Lua?" or "Can I use Lua to solve world hunger?".  Answering all of these just takes up too much time.

I feel that we should have an issue or some other announcement which strictly defines the scope of the issues list as:
*   **Firmware bug**.  This should include a full description of the issue, expected result, actual test case demonstrating the error.
*    **Feature Enhancement**.  A possible enhancement within the scope of what could sensibly be implemented over the SDK defined in concrete terms and with a statement of benefit. 

and provides alternative links for issues outside this scope, e.g. for SDK feature enhancements (the Espressif bbs) and Lua application issues (e.g SO and esp8266.com) .

Any other issues  should be closed referencing this statement and we should avoid entering into discussion on these issues once closed.

marcoskirsch, 2015-08-05:
If these sort of guidelines were clearly stated in the README, this recurring problem would go away.

TerryE, 2015-08-05:
@marcoskirsch Marcos, I am a little more cynical. For this to go away (i) posters would have to read the README, and (ii) do as it suggests.  OK, some might do, so the "false issue" rate might frop by 30% or so, but it would help if we could just say "Please see XX " (some standard link), and better if I then had the permissions to close the issue myself.

nodemcu, 2015-08-06:
Thank you for all these suggestions! and thanks for everything!
so, there is a list I made to push things forward.
1, issues, only issues related to bugs or feature enhancement will stay in the list. others will direct to alternative links. as @TerryE and @marcelstoer suggested.
2, branches, master and dev will be the two major branches. create temporary branch for temp purpose.
3, docs, put docs in this repo. separated into modules. doc should be updated with code in PR and commit.
4, invite someone who willing&able to contribute as collaborators.

nodemcu, 2015-08-06:
Hard to find out contact info from github.  forgive my rudeness.

inversesandwich, 2015-08-08:
I think it might be worth using https://readthedocs.org/ for documentation.

marcelstoer, 2015-08-08:
> I think it might be worth using https://readthedocs.org/ for documentation.

Thanks. It might also be worth reading [other people's comments](https://github.com/nodemcu/nodemcu-firmware/issues/577#issuecomment-125326740) in this long thread before posting "suggestion dups" :wink:

inversesandwich, 2015-08-08:
Doh! I must have missed that when I skimmed though the comments.

boneskull, 2015-08-09:
I've never touched Lua before NodeMCU.  For some, it may be unclear if the problem they are having is a *NodeMCU* problem or a *Lua* problem.  A list of Lua-specific resources may be helpful in `CONTRIBUTING.md`.  

I feel that (on GitHub anyway) `CONTRIBUTING.md` is your best chance to guide people into sending the correct PRs and creating the correct issues.  It's worked well for [Mocha](/mochajs/mocha).

The maintainer closing all issues determined to be "not a NodeMCU problem" should provide a friendly link to `CONTRIBUTING.md` and/or Lua resources.

boneskull, 2015-08-09:
I'd go as far as recommending to drop the wiki entirely and use nodemcu.github.io as the central place for all documentation, tutorials, and links to other resources (like discussion boards).  GitHub wiki's are a bit obscure and can become stale quickly.

boneskull, 2015-08-09:
(you can throw a `CNAME` into there and point `nodemcu.com` at the GH pages site)

marcelstoer, 2015-08-09:
> I'd go as far as recommending to drop the wiki entirely and use nodemcu.github.io as the central place for all...

+-1 :wink: 

As much as I like [GitHub Pages](https://pages.github.com/) myself I still believe that the API documentation should be maintained in the same repository & branch as the source code. That is our best chance to enforce that each PR is complete and includes the necessary documentation updates.

As for `CNAME` and `nodemcu.com` I also find this quite nice, yeah GitHub rocks!, but IMO more essential than the 'form' i.e. the technology is the 'function' i.e. don't scatter information such as code samples all over the place (nodemcu.com,`README.md`, [/examples](https://github.com/nodemcu/nodemcu-firmware/tree/master/examples), [/lua_examples](https://github.com/nodemcu/nodemcu-firmware/tree/master/lua_examples), wiki).

TerryE, 2015-08-10:
I think that the tests here should be: 
*   Should anyone downloading the source need access to this documentation?  
*   (picking up a related point by @boneskull) Is this documentation already provided by an upstream source such as Lua or eLua?

If the answers are **yes** and **no** respectively then the documentation should be in (or buildable from) the nodeMCU/docs tree. eLua has a build process which constructs all of the standard eLua documentation in HTML format from its docs hierarchy.  OK, the latest version is also online but because this is in the repo, anyone using an old version or branch has the option of rebuilding the docs for that branch and also the normal git versioning, diff and blame functions work as standard.

The essential yes/no stuff here is the nodeMCU core and library API documentation.  Thins like white papers and FAQs (which by their nature evolve over time) are perhaps wiki stuff as they aren't part of the configuration controlled source base.  Any wiki would do, IMO as long as it is a single well known one and under general control of the project.

TerryE, 2015-08-25:
We have currently 115 open and 230 closed issues.  Ignoring the closed issues for the moment, the open issues probably fall into 4 categories:

1.   The issue relates to a valid bug in the core firmware.
2.   The issue relates to one of the library adding some enhanced support for a protocol or hardware device
3.   The issue relates to some underlying constraint of Lua 5.1, eLua 5.1, the SDK or the hardware itself and is therefore outside the scope of what can be addresses in the firmware.
4.   The issue is really an applications one, that is more of a "how do I use" rather than an issue that an expert user would consider as an underlying bug in the firmware itself.

My view on 3 and 4 is that these should be allocated to the "won't fix" and "help wanted" labels;  cross referenced to a standard advice issue and closed, but not locked (unless the conversation goes really off the rails).

The difference between 1 and 2 needs a bit of explanation, by comparison to how PHP is set up.  There, the core development team maintain the PHP runtime and a core set of libraries that are part of the distribution.  However there a lot of extensions supported through PECL (the PHP extension library) which is kept separate from the core.   I see that a lot of our "bug" issues aren't really to do with the core software, but to some module or "sketch" (in Arduino terms), so fixing any issues relating to these isn't really for the core team to get distracted by.

So what I propose is

1.    We agree a set categories, each with an associated label.  My list above could be a starting point for labels to use for these categories, but let's discuss this. 
2.    The collaborators work through all unlabelled open issues and assign each a label based on content.
3.    We wait a few weeks for comments and corrections then close all issues relating to categories that we have agreed as out of scope.
4.    We discuss the possibility of creating a  **nodemcu/firmware-extensions** repository for all non-core extensions and handle any issues relating to these in this repository.

Any comments or reactions?

inversesandwich, 2015-08-25:
I second Terry's proposal, especially the separate repo for extensions.

Only comment that I have is for the fourth category of issue. If a user has
to open a issue in order to find out how to use the software, the
documentation has not done it's job and requires improving and more clarity.

marcelstoer, 2015-08-25:
I agree with most of what Terry proposed.

> If a user has to open a issue in order to find out how to use the software

Most cat-4 issues _/_ remember stem from pure laziness to actually search for an read the available documentation.

> We discuss the possibility of creating a nodemcu/firmware-extensions repository for all non-core extensions

Don't complicate matters for now. I think there are enough other issues this project should tackle first (i.e. as discussed above in this meta issue). How would you define core / non-core? How do you build the firmware if extensions are in a separate repository?

boneskull, 2015-08-25:
"Help wanted" issues kind of suck. Some people like them in GH issues, but to my mind the more appropriate place is a mailing list, forum, or chat channel.  Of course there needs to be that place where users can go to get questions answered. 

Would like to hear other opinions on whether "help wanted" issues should be addressed (if you think they should be) instead of closed with links to other resources.

marcelstoer, 2015-08-25:
There _are_ places people can ask "help wanted" questions: http://www.esp8266.com/viewforum.php?f=17 or http://stackoverflow.com/tags/nodemcu to name two. IMHO such issues should be closed here right away with a standard answer.

devsaurus, 2015-08-25:
I'd also favour @TerryE's approach. And let's see how extensions would be defined, but that's a subsequent step. Getting the issue list somehow sorted is most important at the moment, IMO.

Aeprox, 2015-08-25:
> Most cat-4 issues / remember stem from pure laziness to actually search for an read the available documentation.

The available documentation is seriously lacking though, so I'd prefer TerryE's approach.
Most of the info is out there, but it's fragmented and in some instances outdated. For a beginner, that's a steep learning curve. Unless you're able to understand the C source code and/or compile your own firmware, the docs are your only hope to get a better understanding of the internal workings. When you're faced with an undocumented quirk/unexpected behavior it's easy to think it's bug in the firmware.
Right now the issue tracker is a kind of "unofficial documentation" with a lot of knowledge that's not found anywhere in the docs (for the people who aren't too lazy to look for it :wink: ). Until there's been a decision about how to document everything we shouldn't be too strict with the issues, this would only fragment the information more I think.

I am willing to lend a hand btw. This project taught me a lot about lua, C and who-knows-what-else, would love to give something back!

TerryE, 2015-08-27:
>The available documentation is seriously lacking though
This seems to be a common thread and improving the documentation should be an issue in its own right. Because a material rework of official / controlled documentation can take a lot of time and effort, my workaround was to start my [Unofficial FAQ](http://www.esp8266.com/wiki/doku.php?id=nodemcu-unofficial-faq) and it is on the esp8266 wiki so anyone is free to collaborate with me on this. I've got about 50 of what I want to put in, but as a page it's already too long, so I need to reorganise the content.

@nodemcu @vowstar @MarsTechHAN what do you think?

TerryE, 2015-09-07:
Just in case anyone is  interested I've pulled the open issues using the  github V3 API into a spreadsheet, so I can have a first pass at categorising them and labelling them.  Here is a [copy of the ODS](http://files.ellisons.org.uk/esp8266/nodemcu-open-issues-150907.ods) if anyone also wants to have a scan.  I'll report back in a couple of day giving my analysis and recommendations.

marcelstoer, 2015-12-12:
Things have changed for the better by now.

closed 2015-12-12 by marcelstoer