nodemcu/nodemcu-firmware #578: ESP8266 server looses connection... maybe...

nickich, 2015-07-26:
I have a simple http server running.
It seems that the ESP looses connection after certain period of time, if no request is made.
If I run it initially I have access to the web page contents. After some time the connection is lost. The ESP doesn't reply even to ping. The ESP is alive, as I have internal timer and it sends out some strings on UART periodically ( every 1s).
In the second case, if I start the ESP and run ping (from any other machine) continuously, it looks like the ESP doesn't loose the connection.
I'm using the Olimex boards, and I've tried the NodeMcu trunk and branch 096.

TerryE, 2015-07-26:
@nickich, this isn't support forum for application developers.  Please refer to esp8266.com and StackOverflow for this type of support.  If you have a specific repeatable bugrep with test cases and hopefully  reference to the source code responsible for the error then this is the sort of material to be posted here.

nickich, 2015-07-27:
@TerryE, I posted it here as it looks like a bug in nodemcu. I describe the issue and the branches where observed as well as the hardware used.
I don't think it's expected behaviour the connection to be lost if idle...

MarsTechHAN, 2015-07-27:
That's one of the basic feature of TCP . 
<hr>
TCP have a time-out, which means if you keep idle for a set of time(like 30s control by the server), the connection will automatcally drop by server.
<hr>
The solving way is to send some meaningless text to server.

nickich, 2015-07-27:
@MarsTechHAN Thanks for pointing it out. This might explain the behaviour. It makes sense as if I reconnect (wifi.sta.disconnect() followed by wifi.sta.connect()) works and I have the server alive again.
The thing that puzzles me is why the ping-ing the server (from outside) keeps it alive. 'ping' is not over TCP... Also when the server 'dies' the ping doesn't reply as well...

MarsTechHAN, 2015-07-27:
@mickich
I cannot fully understand your question.

1.You use ESP as server or some other thing.

2.What do you mean by 'the server dies'？If you cannot setup a new TCP connection to server, that's not a Time-Out problem.

3. If cannot get response from 'Ping ', that means the physics connection break, or server stop receiving message. That's strange.

nickich, 2015-07-28:
@MarsTechHAN
The ESP is a http server.
I'm not sure if the server literally 'dies'... 
The thing is: If I keep opening the page (on the ESP server) from a browser every few seconds it works great. If I 'ping' the ESP every second or so - it's great as well.
The problem starts if I don't attempt to open the webpage or 'ping' the ESP for a couple of minutes or so.
Then I just get timeout on the browser if I attempt to connect. The ESP doesn't respond to 'ping' at that time as well.
The WiFi status is 5 (STATION_GOT_IP) at all time. I'd expect if I loose connection the wifi status to change.

MarsTechHAN, 2015-07-28:
Which firmware，and can you paste the code？

(NodeMCU have a HTTP Server demo, why dont you try it out.)

jorgempy, 2015-07-28:
@MarsTechHAN 
> I'm using the Olimex boards, and I've tried the NodeMcu trunk and branch 096.

nickich, 2015-07-28:
It is same with the http_server demo from NodeMCU...
The strange thing is 'Destination host unreachable' message from the ping:
_Pinging 192.168.1.17 with 32 bytes of data:_
_Reply from 192.168.1.11: Destination host unreachable._
It looks the route cannot be found... 
Any ideas...?

thanhliem89dn, 2015-08-25:
I got a same issue with ESP 8266 12E with simple webserver example.

locchi93, 2015-08-25:
I have the same problem... after being connected for a certain period of time, the connection disappears.
If I try to send "print(wifi.sta.getstatus())", it returns " 5 " ("got ip" status), but if I try "print(wifi.sta.getip())" it returns "nil".

reischle, 2015-09-27:
Happens to me in stationap mode when sending more than a few single IP packets. I'm using a modified 0.9.5 version from around March 2015. Haven't tried a newer version yet.

nickandrew, 2015-10-03:
This doesn't look like a NodeMCU bug. Make sure your devices have enough power available (300mA?) to transmit WiFi packets. I found I could reliably crash my device just by pinging it a few times when I ran it from an unpowered USB hub.

Please close this issue.

TerryE, 2015-10-03:
I don't think that this is an issue with the nodemcu firmware.

closed 2015-10-03 by TerryE

sghazagh, 2015-10-29:
hi all,
I now facing this issue too.
Any solution for that?

TerryE, 2015-10-29:
@sghazagh, try the latest dev build.

sghazagh, 2015-10-29:
I am in latest dev build mate. (nodemcu_float_0.9.6-dev_20150704.bin)

I really do appreciate if you can help me fix this issue.
I also tried to just use the NodeMCU code for simple HTTP server. So it just lost the connection after a while!!!!

marcelstoer, 2015-10-29:
> nodemcu_float_0.9.6-dev_20150704.bin

That's not what Terry meant I reckon. You should try with a current build from the `dev` branch. Since there's no downloadable binary for that you can either use my [Docker image to build it yourself](https://github.com/marcelstoer/docker-nodemcu-build) or use my [cloud build service](http://frightanic.com/nodemcu-custom-build/) to have it built for you.

sghazagh, 2015-10-29:
I just downgraded the firmware to the one shipped with 'NodMCU-Flasher-Master' and it is:  nodemcu_integer_0.9.5_20150318.bin

From my last post which was 15 minutes ago it was alive when I refreshed the page.
I leave it for another 30 minutes untouched to see if it will be ok.
Will post the result here for others if it works fine....

Thanks for your prompt reply anyway...

sghazagh, 2015-10-29:
Hi,
It seems the Firmware downgrade just worked.
Hope it get fixed in next firmware update...

Thanks

marcelstoer, 2015-10-29:
> Hope it get fixed in next firmware update

That's why you should test with a build from the `dev` branch. No effort is spent towards fixing issues in the 0.9.5 and 0.9.6 branches AFAICT.

sghazagh, 2015-10-29:
To be honest I do not understand which options I should choose to build the firmware.
I think that would be good if you put an options, like "default", to select whatever needs to be in the package to be as same as the binary which you build for official Download.

In that case that would be easy to rebuild the latest dev version for testing...

TerryE, 2015-10-29:
@sghazagh,  this issue is already fixed in dev which is why I closed this.  The nodeMCU guys have "gone fishing", and the active community contributors are all working to rebaseline the firmware against a SDK 1.4 based build. No one who is actively working on the firmware will look at issues with a package that is stretching the esp8266 capabilities against an old SDK version wihch has loads of known bugs, so this issue falls directly in the "won't fix" category.

sghazagh, 2015-10-29:
Can you please be specific and tell me which dev version I can download/build  to have the latest changes which fixed this issue as well?
That would be good if you can address the already created dev version to download.If not, can you please specify what option I have to choose in Marcelstoer provide link then I will create one myself.

Many thanks,

sghazagh, 2015-11-02:
An update...
I just have built my own firmware version and have chosen the most common modules to be included and it was working fine too.

panzer7910, 2019-05-22:
Same problem happens to me:

**Config:**
_Arduino 1.8.9,
esp8266 2.5.2
ArduinoOTA.h
ESP8266WebServer.h_

`begin(){} ` 
>>connect to wifi and do basic initialization

`loop(){} `
>>Have a 2 mode. Mode can be change by IP address follow by predefined URL path :
>> "/CO2" Mode: plain text http webserver serving CO2 sensor output . 
>> "/OTA" Mode: to toggle OverThe Air mode to allow sketch upload over wifi. 

In "/CO2" mode, ESP8266 will passively wait for client request.

**Problem:**
I notice that after a while, the WIFI connection drop. The ESP8266 still doing sensor measurement except no WIFI functionality. 

**Suspected Cause:**
I guess this is my WIFI router feature to drop idle connection after certain timeout period to conserve resources.

**Solution:**
Ping the WIFI router periodically (say, every 10s) to keep Layer 1-3 connection alive.

nwf, 2019-05-22:
@panzer7910 This is a Lua-based ESP8266 firmware, not the Arduino one; you're in the wrong place.