nodemcu/nodemcu-firmware #579: disable spiffs cache code to free ram+flash

aguaviva, 2015-07-27:


BesitzeRuf, 2015-07-27:
?

dnc40085, 2015-07-27:
@aguaviva Can you please elaborate? Have you disabled the cache on your own? If so, does it cause a increase/decrease performance? How much RAM/flash is freed if the cache is disabled?

aguaviva, 2015-07-27:
The spiffs comes with a cache, thing that consumes ram and flash. There are a few #defines that can be used to disable it, I tried but it didn't work. Writing seems to work (wont crash) but reading resets the unit I didn't look much indeep. Will provide more details (going to bed now)

TerryE, 2015-07-29:
@aguaviva, uncached writing to SPI flash is strongly discouraged as the performance is terrible without the cache and the life of the Flash memory severely degraded.  This is not an option that will get support from the community.  There are a lot better ways to save memory, and we're in the process of implementing most of these.

aguaviva, 2015-07-29:
@TerryE As always... it depends! you may need the cache others may need more RAM. As everything, it should be configurable and we should let the user choose whatever fit better their needs.

TerryE, 2015-07-30:
> As everything, it should be configurable and we should let the user choose whatever fit better their needs. 

As I said, I don't think that this will get any support from the dev community.  Do the work yourself and take the performance measures.  It might then get merged.

marcelstoer, 2015-07-30:
> This is not an option that will get support from the community.

Fully agree.

aguaviva, 2015-07-30:
TerryE, I am not asking you to do it, I am putting this here so we know we have this option. And if anyone needs it can explore this path and implement it. That you dont like a feature doesnt mean others won't like it. Just dont implement it, others or myself might do it. Also any idea that can be ifdef'ed should be welcome.

closed 2015-07-31 by aguaviva