nodemcu/nodemcu-firmware #580: AP disconnections when phone is connected

wordsforthewise, 2015-07-27:
When I connect my phone to SOFTAP or STATIONAP mode, it constantly disconnects and reconnects.  It doesn't seem to do this on my computer (although sometimes it sporadically disconnects), but it always happens on my phone.  I'm just testing it with this small lua code:

wifi.setmode(wifi.SOFTAP)
local cfg = {}
cfg.ssid = "myfi"
cfg.pwd = "mystical"
wifi.ap.config(cfg)
local srv=net.createServer(net.TCP, 3000)
print('connect to \''..cfg.ssid..'\', password \''..cfg.pwd..'\', ip '..wifi.ap.getip())
cfg = nil
srv:listen(80,function(conn)
conn:on("connection",function(client)
print("conn")
end)
conn:on("disconnection",function(client)
print("discon")
end)
conn:on("reconnection",function(client)
print("recon")
end)
conn:on("sent",function(client)
print("sent")
end)
conn:on("receive", function(client,request)
    print("recieve")
end)
end)

I'm thinking that some push updates on my phone are screwing it up, but I have no idea if it's even possible to debug in lua, and I don't know where to start debugging it in c.  If anyone has advice for figuring out how to debug it in c let me know.

MarsTechHAN, 2015-07-27:
Because ESP's hot spot is not available to Internet. Some kind of phone will constantly drop connection and try other hot spot.

wordsforthewise, 2015-07-27:
Ah I see.  Is there a way to use the ESP into fooling the phone so it doesn't disconnect?

MarsTechHAN, 2015-07-28:
I dont have a good enough idea...

marcelstoer, 2016-02-26:
Still an issue with the SDK 1.5.1 used in the `dev` branch? If so please reopen.

closed 2016-02-26 by marcelstoer