nodemcu/nodemcu-firmware #582: many function definitions don't use the ICACHE_FLASH_ATTR

aguaviva, 2015-07-29:
Examples, all the Lua, mqtt, spifs, all the modules, coap...

If I understood correctly this keyword is necessary so the function gets executed from ROM and doesn't get copied to RAM before executing it.

creationix, 2015-07-29:
Isn't there a performance impact of this?  Are we still at a point where we want more free ram and less free CPU?

marcoskirsch, 2015-07-29:
@creationix I would say yes.

aguaviva, 2015-07-29:
it should be configurable so everyone can choose what they want, no need to make compromises.

Also the whole sdk uses that keyword all over the place, including the networking code

nodemcu, 2015-07-29:
@aguaviva 
we made some modification to the stock sdk from espressif,
put every function .text into ROM, even if there is no "ICACHE_FLASH_ATTR".
recently also .rodata into ROM, thanks to @jmattsson.
so, if you want your function in the RAM, you should add "ICACHE_RAM_ATTR" explicitly.

closed 2015-07-29 by nodemcu

aguaviva, 2015-07-30:
Excellent! Thanks for clarifying and for the great job.