nodemcu/nodemcu-firmware #583: SSD1306 128x64 OLED

fixingthingsguy, 2015-07-29:
Hello: It appears that the SSD1306 sold on ebay might use a different controller SH1106 (per Oli Kraus at http://forum.arduino.cc/index.php?topic=219419.45). Looks like Oli has made the change. Can you incorporate into nodemcu?  Here's the mail from the Arduino forum for your reference.
Thanks and Regards.
. 
Extract below from above link>>>>>>>>>>>>>>>
Your OLED does not contain the SSD1306 controller, but has the SH1106. In U8glib I added the SH1106 construcor with v1.16. This should fix your problem.

Recently, this has been asked more often here. I guess ebay sellers declare their OLEDs as SSD1306, but actually sell a display with SH1106 controller.

Oliver"
End Extract >>>>>>>>>>>>>>>>>>>>>>>>>>>>

devsaurus, 2015-07-29:
Support for SH1106 displays (I²C and SPI) was recently added to nodemcu per #572.
It's currently only in dev branch, so you could give it a try. You just need to recompile after enabling the driver in `u8g_config.h`.
Please note that I didn't verify these new integrations due to lack of hardware. Thus I'm very curious to learn about your results.

fixingthingsguy, 2015-07-29:
Thank you for your research. I have to say that I am not very well trained and would need some guidance

to recompile the code. I will be most happy to provide testing results when I can do that.

I will look around to find out how to recompile as it would be useful to me for the future as well.

I appreciate the huge amount of selfless work done by volunteers like yourself and others, including Oli.

Regards

closed 2015-07-29 by fixingthingsguy

fixingthingsguy, 2015-08-01:
I am happy to say I tested support for sh1106 and it works just fine.
I had load buil by "titanic".NodeMCU custom build by frightanic.com
	branch: master
	commit: 474ae33288f82f79d1e6cec907a6c842e188487b
	SSL: false
	modules: node,file,gpio,wifi,net,pwm,i2c,spi,tmr,adc,uart,ow,bit,u8g,cjson
 	built on: 2015-08-01 15:49
I used your examples for graphics.lua.
The hardware was purchased on ebay, but I used the graphics.lua software
UNCHANGED.
-ie although backof my hardware says 0x7a or 0x78, I still used 0x3c for sla.
as below: 
    sla = 0x3c
     i2c.setup(0, sda, scl, i2c.SLOW)
     disp = u8g.ssd1306_128x64_i2c(sla)

Fantastic support by volunteers. I can't say how much I appreciate this.
I used to work for a large software company and the effort in building a load was a HUGE
job. You all make it so simple.
Regards

F1CJN, 2017-09-14:
I confirm. Stange but OK. Don't uses 0x78 but 0X3C for the Ebay chineese OLED module (SH1106).
Great . Thanks a lot