nodemcu/nodemcu-firmware #590: on:receive event gets called twice if using HTTP/1.1 on api.pushingbox.com (works ok with HTTP/1.0)

8n1, 2015-08-03:
While playing around with Pushingbox I discovered this small bug. 
It's really ease to reproduce since you just need to create a connection to api.pushingbox.com(213.186.33.19) and print the payload in the on "receive" event.

Here is a short example:
```
pushConn = net.createConnection(net.TCP, 0)
pushConn:on("receive", function(__, payload)
    --print(payload)
    print(" Done.")
end)
pushConn:connect(80, "213.186.33.19")
pushConn:send("GET / HTTP/1.1\r\n"
    .."Host: api.pushingbox.com\r\n"
    .."Connection: close\r\n"
    .."Accept: */*\r\n"
    .."\r\n")
pushConn = nil
```
What I'm expecting is just a single "Done." but it always gets printed twice !
If you 'comment in' the print(payload) statement you can see that it is always 0 after the first "round". It took me quite a while to debug this because i thought the 0 belongs to the payload. 

Now the strange thing is if I set the protocol version to HTTP/1.0 it works as expected.

Í couldn't find any other website wich causes this bug and it seems only related to their subdomain(api.pushingbox.com). 

I've tested it on 0.9.5 ( 0.9.5_20150318 ) and the latest 0.9.6-dev (0.9.6-dev_20150704) release.

------
PS: I removed  the dns query and a http status extraction from the example above for readability, the original code (how I'm using it now) looks somthing like this:
```
dnsConn = net.createConnection(net.TCP, 0)
dnsConn:dns('api.pushingbox.com', function(pushConn, ip)
    pushConn:connect(80, ip)
    pushConn:send("GET / HTTP/1.0\r\n"
        .."Host: api.pushingbox.com\r\n"
        .."Connection: close\r\n"
        .."Accept: */*\r\n"
        .."\r\n")
    pushConn:on("receive", function(__, payload)
        http_status = string.sub(payload, 10, 12)
        if http_status == "200" then
            print(" SUCCESS.")
        else
            print(" Something went wrong.")
            print(payload)
        end
        http_status = nil
    end)
end)
dnsConn = nil
```

MarsTechHAN, 2015-08-03:
The TCP has an maximum amount of data in a single package. And the TCP won't attach those sepreat package together.

Here are two ways.
1. Recognize they are the same package(I wont explain here....Google it)
2. Drop the second package.

8n1, 2015-08-03:
Thanks for the hint, I think I got it. At least partly :)

You are taking about the chunked Transfer-Encoding, right?
So whenever the server trys to "chunk up" the response by sending a "Transfer-Encoding: chunked" header the on:receive function get's called for every "chunk".

And the reason why it works with HTTP/1.0 is because "chunked transfer-encoding" is not possible when using HTTP 1.0. 

Knowing what to look for I found a few other websites which uses the chunked Transfer-Encoding.
amazon.com is one them. But unlike api.pushingbox.com, www.amazon.com creates a lot more fragments.

```
function make_request(host)
    local counter = 0
    local dnsConn = net.createConnection(net.TCP, 0)
    dnsConn:dns(host, function(pushConn, ip)
        pushConn:connect(80, ip)
        pushConn:send("GET / HTTP/1.1\r\n"
            .."Host: " ..host .."\r\n"
            .."Connection: close\r\n"
            .."Accept: */*\r\n"
            .."\r\n")
        pushConn:on("receive", function(__, payload)
            counter = counter+1
            print("Counter: " ..counter)
        end)
    end)
end

make_request("www.amazon.com")
```
After running this code the Counter shows: ~265 (it took 17.457527 seconds to run and it doesn't work always, sometimes the counter just shows 1, mostly if the function gets called the first time)
Yes, parsing the amazon homepage directly is probably the last thing you want to do but it shows the "problem" :)

I will need some more time to read and understand this better. 

Your solutions:
1. By looking for the Transfer-Encoding and Content-Length Header?
2. So just ignore it or them by using something like a "already_done" variable and checking it?

MarsTechHAN, 2015-08-03:
The very first question @8n1
You want all the receive data or just an HTTP Head?

8n1, 2015-08-03:
In this case I only need the HTTP Head.

TerryE, 2015-08-04:
There's nothing in the HTTP/1.1 RFC that mandates that a server must not fragment its response over multiple IP packets.   OK, it shouldn't do  this if the response could be fitted into a single packed, but ***should not*** isn't ***must not***.  HTTP is quite a complex protocol to implement properly in nodeMCU lua.  I have yet to see a robust implementation that doesn't make all sorts of nasty assumptions about the other end.

MarsTechHAN, 2015-08-04:
So you use a flag
```lua
local flag = false

function make_request(host)
    local dnsConn = net.createConnection(net.TCP, 0)
    dnsConn:dns(host, function(pushConn, ip)
        pushConn:connect(80, ip)
        pushConn:send("GET / HTTP/1.1\r\n"
            .."Host: " ..host .."\r\n"
            .."Connection: close\r\n"
            .."Accept: */*\r\n"
            .."\r\n")
        pushConn:on("receive", function(__, payload)
     if flag == false then
             flag = true
            print(payload)
      end
        end)
    end)
end

flag = false
make_request("www.amazon.com")
```

TerryE, 2015-08-04:
@MarsTechHAN Martin, the point here for *this issue list* is that this **isn't** a bug in the nodeMCU firmware; it's a feature of the SDK that we shouldn't expect the firmware to compensate for.  So this issue should be **closed**, rather than getting into a discussion on application programming techniques. 

@8n1, if you want to understand the bear traps in what you are doing or in what Martin suggest have a read of my [Unofficial nodeMCU FAQ](http://www.esp8266.com/wiki/doku.php?id=nodemcu-unofficial-faq) or discuss this on one of the support forums,

8n1, 2015-08-04:
@MarsTechHAN This is exactly what I meant by using a 'already_done' variable. I'm using it like this:
```lua
function make_request(host)
    local already_done = false
    local dnsConn = net.createConnection(net.TCP, 0)
    dnsConn:dns(host, function(pushConn, ip)
        pushConn:connect(80, ip)
        pushConn:send("GET / HTTP/1.1\r\n"
            .."Host: " ..host .."\r\n"
            .."Connection: close\r\n"
            .."Accept: */*\r\n"
            .."\r\n")
        pushConn:on("receive", function(__, payload)
        if not already_done then
            print("Done")
            already_done = true
        end
        end)
    end)
end

make_request("www.amazon.com")
```

@TerryE I've already read your FAQ, twice. It helps, but it's not easy to understand all those concepts. But since I really love tinkering with NodeMCU I'll stick to it and try to understand as many as possible.

**Issue closed**. May be discussed as suggested in one of the support forums.

PS: Many thanks for your work guys, I really appreciate that.

closed 2015-08-04 by 8n1

TerryE, 2015-08-04:
Post on the esp8266 forum; I answer Qs there.  If you have constructive points on the FAQ I'd be more than welcome to discuss them / refine the FAQ. :smile: