nodemcu/nodemcu-firmware #591: monitor mode to probe nearby mac address

arthurtalkgoal, 2015-08-05:
Hi 

I need a WiFi probe device mac.   I think turning a nodemcu into monitor mode is a good soln. 

http://edwardkeeble.com/2014/02/passive-wifi-tracking/
Can we turn nodemcu into monitor mode?

clixx-io, 2015-08-05:
Yes.

I wrote such a tool in python, to discover new esp8266 on the LAN.

https://github.com/clixx-io/clixx.io/tree/master/source/tools/ip-notifier

So porting it to elua should also be possible.

TerryE, 2015-08-05:
This issues list is for reporting bugs in the ndeMCU firmware.  Your Q is not for here.  Please close this issue and use a support forum such as esp8266.com or StackOverflow.

FourQuarks, 2015-08-05:
You are boring TerryE. Nothing better to do?

marcelstoer, 2015-08-05:
> Nothing better to do?

I'm sure he did but instead his helping out here trying hard to bring some structure to this project. Every minute spent handling support requests that don't belong here is a minute spent that doesn't go into pushing the firmware forward.

arthurtalkgoal, 2015-08-05:
Hi Guys,

I understood if itz out of the scope, I should close the issue. 

Anyway, I think it may be something only fireware level can do. Let me know if @TerryE not agree.  

Thanks in advance

marcoskirsch, 2015-08-05:
> I'm sure he did but instead his helping out here trying hard to bring some structure to this project. 

+1

TerryE, 2015-08-05:
@arthurtalkgoal Arthur, if you read my [nodeMCU Unofficial FAQ](http://www.esp8266.com/wiki/doku.php?id=nodemcu-unofficial-faq) then you will see that the nodeMCU firmware essentially wraps a Lua scripting environment around the Espressif-supplied close source SDK and has work within the constraints imposed by the API exposed by the SDK.  

This community's priority is to improve and enrich the Lua environment so make it easier to use Lua to script IoT applications on the esp8266, so *this* list is for the firmware developers to exchange issues and ideas relating to the firmware internals.  There are other general resources to assist application developers in developing there applications such as the developer forums and my FAQ.

closed 2015-08-05 by arthurtalkgoal