nodemcu/nodemcu-firmware #596: Dev: smart config not working

3dot3, 2015-08-09:
never connected after run sample code on esp-13:
wifi.setmode(wifi.STATION)
wifi.startsmart(0,
    function(ssid, password)
        print(string.format("Success. SSID:%s ; PASSWORD:%s", ssid, password))
    end
)
-- above sdk v120, can get phone ip, must use esptouch v034.
wifi.startsmart(0,function(ssid, password,phoneip) print(ssid) print(password) print(phoneip) end)

MarsTechHAN, 2015-08-09:
I have already test it, and it works well. 
0.Which branch you are using? dev? There is not only one dev branch
1.Do you use correct app? Only newest "IOT Espressif" will work
2.What do you mean by "never connected after run"
3.You need to read  a book named"power questions"

3dot3, 2015-08-09:
I have the same code running on dev120 and it is fine for 2 android mobile devices. then I change the firmware to "dev", it doesn't connect at all. App version is 0.34

rxf992, 2015-08-17:
branch 'dev' use old version of SDK, which is not v1.2, and it can not work with latest ESP demo APP.

marcelstoer, 2015-08-17:
> branch 'dev' use old version of SDK, which is not v1.2

Correct, anyone who requires v1.2 must currently stick with the dev120 branch.

3dot3, 2015-08-31:
Which version of demo code it should be to working with this dev branch?

jmattsson, 2015-10-28:
An FYI, dev has been updated to SDK 1.4.0, and a the smartconfig module saw a little bit of work (which hasn't been tested since I don't have a setup for that). You may want to see how well (or if) things work on a recent dev build.

3dot3, 2015-10-28:
According sdk 1.40 docs, it has smartconfig 2.5.1. I've tested it with latest esptouch demo code from esp. 8266 can pick up packets in sniffer mode. But esptouch app can't receive udp broadcast from 8266 after it connected.

3dot3, 2015-10-28:
Latest esptouch app was announced only support smartconfig 2.4, I think that is why udp broadcast from smartconfig 2.5.1 not recognized.

3dot3, 2015-11-19:
As #715 descripted, this is not about the App not parsing the udp broadcast. It is actually the broadcast packet not sending out from ESP8266 at all.

marcelstoer, 2016-04-10:
Feel free to re-open if you're still seeing this with the latest `dev` branch. Note that smartconfig is now turned OFF by default.

closed 2016-04-10 by marcelstoer

Anudeep09, 2016-06-03:
I tried same with with ESP-12 module Firmware is NodeMCU 0.9.6 build 20150704  powered by Lua 5.1.4 and SDK id ESplorer  v0.1 build 206 by 4refr0nt  and App is ESP8266 SmartConfig version 0.3.4.3..  please any one can help..