nodemcu/nodemcu-firmware #598: luac.cross build process

TerryE, 2015-08-10:
This is more of a "note to sefl".  The starndard eLua distro has a **luac.cross** which if run most platforms will produce a host executable which will compile Lua code intended for an eLua target into an **lc** dumpfile that can be loaded onto the target or incomporated into a ROMFS or SPIFFS image.

My #468 patch changes the dumpfile format if line number info is retained, so to avoid having to maintain a standard eLua / luac patch as well as nodeMCU lua, I will see if this luac.cross build process can be added to nodeMCU.

TerryE, 2015-08-25:
I decided that I needed this to go with my #468, so I've been making progress on this.  A dozen or so warning left plus `liolib.c ` needs reworking to include the native and embedded versions of some routines.

It's a pity that the nodeMCU team decided to restructure the eLua hierarchy and change some common naming conventions which I've had to do a bit of macro wizardry to back out.  Still, time for bed and a job for tomorrow.  An example is that the standard libary includes are now of the form 

```C
#include C_STRING_INCLUDE
``` 

where `C_STRING_INCLUDE` is either `"c_string.h"` or `<string.h>` depending on which type of compile is taking place.

TerryE, 2015-08-28:
So I've got this working for my local nodeMCU fork. The version builds the same binary as master if `make` is run with  `LUA_CROSS_COMPILER` **not** defined.  However, if you run `lua tools/cross-lua.lua` on the host, then the same distro builds a runnable `luac.cross` host executable.

Does anyone else think this is worth adding to the core distro?

rxf992, 2015-08-30:
Yes, I like this feature~

TerryE, 2015-09-01:
Patch now incorporated into master

closed 2015-09-01 by TerryE