nodemcu/nodemcu-firmware #602: Bug with connecting iOS/OSX devices to ESP-8266 in AP mode

mazloumi, 2015-08-11:
Hi,

I have a problem with connecting any kind of iOS/OSX devices to the ESP-8266 when it is in AP mode. Here is my configuration:

- ESP-8266 ESP-12 on a NodeMCU DevKit v1.0
- NodeMCU 0.9.6 build 20150704  powered by Lua 5.1.4

Using a windows or android machine works without any problem but with Apple devices I get the error message that the password is incorrect. The bug is difficult to reproduce. I managed to connect an Apple device by explicitly calling 

ap.beacon = 100

But this may work 1 or 2 times then it does not connect again. Alternatively, I played with various SSIS names (with or w/o characters or digits) and same thing. Somethimes it works for a brief moment and somethimes not.

Any suggestions?

fernandorocha, 2015-08-14:
I can connect with apple devices, but can`t access the web pages served by the ESP8266

SpaceTeddy, 2015-08-15:
I have the same trouble if I set a custom IP address. I.e 10.10.10.1
If I leave this open with the standard 192.168.4.1 ip, then I have no issues with wifi connection

tiegeng, 2015-08-17:
I have the same problem with exactly the same setup. 

ESP-8266 ESP-12 on a NodeMCU DevKit v1.0
NodeMCU 0.9.6 build 20150704 powered by Lua 5.1.4

kubi57, 2015-08-20:
I have a similar problem.

ESP-01 with different builds

After starting the ESP in AP mode i can connect with iOS device, i get the html page from the ESP and after some seconds there is no network connection any more. When looking in the wifi settings of iOS it has connected to my routers wifi. When then trying to connect to the ESP wifi i get the error message that it can't connect to this network. After rebooting the ESP the a.m. happens again. As this happens with older builds of NodeMCu as well, I think it could be a new behaviour of the actual iOS update.

Windows woks well.

Denages, 2015-09-22:
I have the same bug. I'm trying to connect to nodeMCU with iPad IOS 8.1.2 and always get incorrect password error.

medarz, 2015-11-17:
Hi. Im also facing same issue. Password error on iOS devices (iphone 5) trying to connect to:
=node.info()
0	9	6	10616174	1392879	2048	2	40000000
Android works well

Darcade, 2016-02-06:
I have the same issue using iPhone 6 with iOS 9.2 and an ESP-03
`> =node.info()
0	9	6	16664214	1261768	512	0	40000000
`

jmattsson, 2016-02-13:
0.9.6 is old and no longer supported. Please see #719. Current master is on 1.4.0, and dev on 1.5.1.

joihn, 2016-03-04:
I still have the issue on 1.4.0. Any idea ?

TerryE, 2016-03-04:
@joihn, if there is an issue with 1.4.0 ten the truth is that we aren't going to backport any fix. `master` is frozen until resnapped to `dev`.  Work with dev and you might find someone interested in helping, otherwise you can do what we do and start with the source ...

joihn, 2016-03-04:
in fact I found the error : in the script I was using, the HTTP request "HTTP/1.1 200 OK\n\n" was absent. Desktop browser and android broser were tolerant, but not Iphone nor Ipad...

TerryE, 2016-03-04:
Oops. Pressed the wrong button. I shouldn't have reopened this  :(

closed 2016-03-04 by TerryE

ehsanzzz, 2016-06-24:
Oh  my God.I am confused .
I have a same problem .
in my case i can connect to esp8266 in AP  mod for the first time after reboot the module.
but when i disconnect from wifi  module and try to reconnect ....opppps.
My mobile phone is Huawei p8 lite and do not have any problem with other wifi devices.
I try with other mobile phones and tablets. they work perfectly.
iOS device and android also windows devices .
any body has a same problem here ???

tablatronix, 2017-02-21:
incorrect password, and cannot connect to network errors on IOS, looks like IOS does some kind of caching and gets stuck where it will not connect for some reason.

I also get a lot of, unable to join and 
`err already associed!`

I think softap is not negotiating this stuff properly all the time.

tablatronix, 2017-11-03:
Oh and I still get this on 2_3_0
But it seems to be fixed in RC , anyone know what the actual bug is/was ?

tablatronix, 2017-11-03:
Found this http://bbs.espressif.com/viewtopic.php?t=235