nodemcu/nodemcu-firmware #603: Feature request: date/time conversion

pastukhov, 2015-08-14:
Since we have sntp support i thing we need to have a date/time conversion library to convert timestamp to date/time and back.

pastukhov, 2015-08-18:
Anybody?

TerryE, 2015-08-18:
Patience Artem.  It's a new feature request and everyone has their priorities.  You shouldn't expect a quick response on something like this.

pastukhov, 2015-08-18:
Ok, thanks, i'll wait.

jmattsson, 2015-08-19:
Ah... time. Arguably the worst thing to deal with in computing, because there is no sense to it. See for example [Falsehoods programmers believe about time](http://infiniteundo.com/post/25326999628/falsehoods-programmers-believe-about-time), and the continuation (linked in said post).

In order to implement UTC<->date/time conversion it's necessary to keep up with the "timezone database" (which contains not only timezone information, but also things like leap seconds and daylight savings rules). Timezones are the playthings of governments and have any sane person tearing their hair out trying to deal with. And don't get me started on Daylight Savings... the rules around that change more frequently than the seasons in Melbourne (which is known for its "all four seasons in the same day" weather).

If anyone really wants to try and get all this onto an ESP they're of course welcome to do so, but I don't see myself or $work having the need, desire or time to do so any time soon. The recommended best practice for working with time stamps is to keep everything in UTC as long as possible, and only do date/time conversion right before it's presented to the user, which quite frequently is right in their browser.

marcelstoer, 2015-08-19:
> Since we have sntp support i thing we need to have a date/time conversion library

@pastukhov can you elaborate why you think SNTP-support mandates such a library? I don't see the necessity yet.

> The recommended best practice for working with time stamps is to keep everything in UTC as long as possible, and only do date/time conversion right before it's presented to the user

+1

pastukhov, 2015-08-19:
Ok.
I'm going to make an informer with oled/lcd screen similar to http://blog.squix.ch/2015/05/esp8266-projects-internet-connected.html, but it should be more universal.  And of course it should display current date and time. Can anyone advice me, how to display it? Any online service?

jmattsson, 2015-08-20:
A quick google led me to this: http://www.convert-unix-time.com/api

pastukhov, 2015-08-20:
@jmattsson Great, thanks!

closed 2015-08-20 by pastukhov

makefu, 2016-05-29:
somehow i refuse to believe that using a dedictated web api is the solution for this issue. it seems like https://github.com/JChristensen/Timezone does a quite good job for arduino projects to figure out the timezone, maybe this could be adopted for nodemcu?