nodemcu/nodemcu-firmware #605: iOS device connecting to Node (AP mode)?

solidremote, 2015-08-17:
I have trouble connecting nodemcu AP when using iPhone (both 4 with iOS 7 & 5s with iOS 8 tried), while using Android phones it is ok.

iPhone will get IP address (such as 192.168.4.2) subnet & Router(gateway), but there is no WIFI logo shown on iOS title bar.

When I input manually DNS address (any, for example 0.0.0.0) then the WIFI logo will show on iOS title bar.

It seems in AT firmware, 8266 will issue a invalid DNS address such as 4.1.192.168 that will solve this problem, can Nodemcu do this in similar way? thank you.

marcelstoer, 2015-08-17:
Very strongly smells like a dup of #602. If so please close this one.

TerryE, 2015-08-17:
Duplicate of #602

closed 2015-08-17 by TerryE

marcelstoer, 2015-08-17:
> TerryE closed this 4 minutes ago

So you're an official contributor with the respective permissions now? Congrats!

TerryE, 2015-08-17:
> So you're an official contributor with the respective permissions now? Congrats!

In practice just for issues management.  I've got to do a blitz of old issues in the next week or so.   I'll still be working pull request practices re code contribution. :smile: