nodemcu/nodemcu-firmware #607: "Move constants to ROM" patch breaks 1-Wire module

mludvig, 2015-08-19:
I just noticed that 1-Wire (ow) module doesn't work properly in the latest git version of the firmware. In particulat ow.read() always returns 0xFF, but for example ow.search() works fine.

It can be easily tested with the provided `lua_examples/onewire-ds18b20.lua` and a single DS18B20 connected to NodeMCU (I have it on PIN 4 instead of 9 - in case that makes a difference).

After some git-bisecting I found that the patch that broke it was "Move constants to ROM" (git b0f9788a6d). And sure enough after I reverted this commit in the latest git master 1-Wire begun working again.

Any pointers idea why that patch have broken the module? And more importantly how to fix it?

Thanks

TerryE, 2015-08-19:
This may be a timing issue.  AFAIK, the way that "ROM" works on the esp8266 if that the code and ROM data is actually fetched from an area of RAM that is in effect configured as a L1 cache which the chip firmware fetching any cache misses from SPIflash.  The limitation of this system is that this cache-miss process only works correctly for word aligned fetches so barfs on character fetches.  

What this means is that the code should be written to separate aligned ROMable data from character aligned data.  Unfortunately many modules having been written with this in mind, so Johnny's patch (see #504) is an exception handler which picks up these faults and handles them programmatically.  The upside is that we can move a lot of library modules ROM data off RAM and into Flash.  The downside is that this can timing issues with some code that is time critical and cause it to fail.  What we probably need is some form of attribute to say "keep this in RAM" or simply take local copies of ROM data before the time critical section.  It's a case of no free lunch, I guess.

jas39, 2015-08-19:
Ditto that bug. Seems like first command after search drops first bit(s), ie reading temperature
          present = ow.reset(pin)
          ow.select(pin, addr)
          ow.write(pin,0xBE,1)

becomes 0xBA (as checked on a logic analyzer) reverting fw corrects this. 

Bug seems valid for first byte in all reset->select->write(byte) sequencies

mludvig, 2015-08-20:
@jas39 which fw exhibits this? Or which git commit breaks it? Could that have something to do with timing caused by ROM-ification of constants?

mludvig, 2015-08-20:
@TerryE I don't think CRC is the problem - that kicks in only after the data are read from OW but the data read are all 0xFF. Nothing useful to do CRC on.
I guess the timing issue is caused by something outside of onewire.c. 
Perhaps the delayMicroseconds() is no longer precise? Can anyone check that with a logic analyser please?

jmattsson, 2015-08-20:
I thought the reported one-wire issues (this and #533) were already fixed by Thomas Shaddack's patch, but now I see that patch probably only went to me. As I'm not set up to test one-wire things here, could someone with this issue please try with said patch? I've pushed a patched version of dev to [this branch](https://github.com/DiUS/nodemcu-firmware/tree/onewire-powerup).

@TerryE There is indeed a macro for forcing a constant to reside in RAM, it's called RAM_CONST_ATTR. Your suggestion about caching on the stack before a critical section is also a good one. As for your last comment, the only driver with any potential const data is in fact the one-wire driver. Based on the description of the issue, the contents of Thomas Shaddack's patch, and what @mludvig just said, I suspect the timing issue actually lies outside the onewire.c driver itself, so a simple approach of "stuff all driver constants into RAM" might be counterproductive.

Regarding delayMicroseconds(), it itself would not have changed as it's a ROM function. Looking at the disassembly from onewire_{read,write}_bit, it also doesn't look like there would be any difference in the setup time, as all those contant values are immediates and thus loaded directly from the instruction cache.

mludvig, 2015-08-20:
@jmattsson great the `owDefaultPower` patch from your repo makes 1-Wire work again! Looking forward to see it in the main tree :)

jmattsson, 2015-09-01:
Since this has been merged into dev now, can we consider this issue resolved and close it?

closed 2015-09-01 by TerryE