nodemcu/nodemcu-firmware #610: ds18b20.lua | Handling difference between DS18S20 and DS18B20

windows97, 2015-08-21:
Looks like DS18B20 only works proper with ds18b20 (having FAMILY CODE == 0x28).
DS18**B**20  (FAMILY CODE == 0x28) has a 12bit resolution w/ 2^(-4) lsb = 0,0625 which is correclty applied.
[image]
*derived from http://datasheets.maximintegrated.com/en/ds/DS18B20.pdf*

DS18**S**20 (FAMILY CODE == 0x10) has a 9bit resolution w/ 2^(-1) lsb = 0,5 which is not handled correctly.
[image]
*derived from http://datasheets.maximintegrated.com/en/ds/DS18S20.pdf*

This factor should be handled according to the FAMILY CODE.

nickandrew, 2015-10-05:
I fixed the code in `lua_examples/onewire-ds18b20.lua` ... oh hang on, there's more code under `lua_modules` ...

nickandrew, 2015-10-05:
2nd commit pushed. Does this resolve your issue?

nickandrew, 2015-10-05:
3rd commit pushed; thanks @TerryE.

PR #681 (2015-11-01): Also support DS18S20
The DS18S20 has only 1 fractional bit whereas DS18B20 has 4, and their
temperature register alignment differs. Check the family code to choose
the correct multiplier for both devices.

Closes #610
commit 53eee168 (2015-11-01): Also support DS18S20

The DS18S20 has only 1 fractional bit whereas DS18B20 has 4, and their
temperature register alignment differs. Check the family code to choose
the correct multiplier for both devices.

Closes #610
PR #767 (2015-11-18): Support DS18S20 and fix negative temp handling
[2015-11-16: Fixes cleaned up and rebased against dev branch.]

The DS18S20 has only 1 fractional bit whereas DS18B20 has 4, and their
temperature register alignment differs. Check the family code to choose
the correct multiplier for both devices.

Closes #610

Negative temperatures (less than 0°C) are returned as a sign-extended
two's complement number. Subtract 0x10000 to recover the proper
negative value.
commit 0a7b730e (2015-11-18): Support DS18S20 and fix negative temp handling

The DS18S20 has only 1 fractional bit whereas DS18B20 has 4, and their
temperature register alignment differs. Check the family code to choose
the correct multiplier for both devices.

Closes #610

Negative temperatures (less than 0°C) are returned as a sign-extended
two's complement number. Subtract 0x10000 to recover the proper
negative value.
closed 2015-11-01 by vowstar