nodemcu/nodemcu-firmware #613: system param error @ ESP-01E

milkpirate, 2015-08-24:
I have the ESP-01E module handy and wanted to flash the lastest nodemcu firmware. I coundnt find a nodemcu_latest.bin or ESP8266Flasher.exe which was younger than 6 months. So I tried to build it on my own @ Ubuntu 14.04.3 LTS x64. I have the following toolchain installed: [esp-open-sdk](https://github.com/pfalcon/esp-open-sdk) (standalone) and its working well so far. The firmware builds and flashs via make flash well. But booting it up I get the following serial print out (75000 Bd):

```
 ets Jan  8 2013,rst cause:1, boot mode:(3,7)

load 0x40100000, len 31816, room 16 
tail 8
chksum 0x6c
load 0x3ffe8000, len 3016, room 0 
tail 8
chksum 0x83
load 0x3ffe8bc8, len 8, room 0 
tail 8
chksum 0x43
csum 0x43
system param error
0f 04 3c 00 00 06 6b 01 42 00 00 00 08 17 6c 65 6e 00 06 6c 01 37 00 00 00 0c 00 18 5a 1d 00 00 
10 06 67 01 55 05 00 00 19 2a 4b 00 00 06 68 01 ae 03 00 00 1a 74 73 76 00 06 6d 01 e5 04 00 00 
00 0d 5a 1d 00 00 06 6e 01 2f 05 00 00 16 14 06 78 01 b8 05 00 00 0f 35 a6 00 00 06 79 01 5a 04 
00 00 00 17 74 74 00 06 79 01 7f 03 00 00 04 0f d4 1b 00 00 06 79 01 7f 03 00 00 05 0f 15 25 00 
00 06 7a 01 54 06 00 00 08 17 65 6e 76 00 06 7b 01 54 06 00 00 0c 17 6c 65 6e 00 06 7c 01 37 00 
00 00 10 00 0e 35 1c 00 00 20 06 fd 01 54 06 00 00 0f 35 a6 00 00 06 fe 01 5a 04 00 00 00 17 74 
74 00 06 fe 01 7f 03 00 00 04 0f d4 1b 00 00 06 fe 01 7f 03 00 00 05 0f 8a 90 00 00 06 ff 01 7f 
03 00 00 06 0f 69 17 00 00 06 00 02 7f 03 00 00 07 0f 15 25 00 00 06 01 02 54 06 00 00 08 0f b8 
09 00 00 06 02 02 df 04 00 00 0c 0f 57 3c 00 00 06 03 02 d1 0a 00 00 10 0f 20 1c 00 00 06 04 02 
d1 0a 00 00 14 0f 0b 1e 00 00 06 05 02 5a 04 00 00 18 0f ef 3b 00 00 06 06 02 30 00 00 00 1c 00 
06 04 b8 05 00 00 18 e9 17 00 00 18 06 76 01 7f 06 00 00 19 2a 4b 00 00 06 77 01 ae 03 00 00 1a 
75 76 00 06 7d 01 61 05 00 00 00 0d e9 17 00 00 06 7e 01 5a 06 00 00 0e 7f 1b 00 00 4c 06 86 01 
bf 07 00 00 0f 35 a6 00 00 06 87 01 5a 04 00 00 00 17 74 74 00 06 87 01 7f 03 00 00 04 0f d4 1b 
00 00 06 87 01 7f 03 00 00 05 17 6b 00 06 88 01 df 04 00 00 08 0f 62 3f 00 00 06 89 01 bf 07 00 
00 0c 17 70 00 06 8a 01 c5 07 00 00 10 0f c1 19 00 00 06 8b 01 d1 07 00 00 14 0f 58 1b 00 00 06 
8c 01 0c 08 00 00 18 0f 34 18 00 00 06 8d 01 12 08 00 00 1c 0f f8 89 00 00 06 8e 01 18 08 00 00 
20 0f 30 18 00 00 06 8f 01 30 00 00 00 24 0f 84 1a 00 00 06 90 01 30 00 00 00 28 0f 27 18 00 00 
06 91 01 30 00 00 00 2c 0f bd 19 00 00 06 92 01 30 00 00 00 30 0f 8a 1a 00 00 06 93 01 30 00 00 
00 34 0f 54 1b 00 00 06 94 01 30 00 00 00 38 0f 0a 17 00 00 06 95 01 30 00 00 00 3c 0f 06 17 00 
00 06 96 01 30 00 00 00 40 0f 0b 1e 00 00 06 97 01 5a 04 00 00 44 0f 4c 1d 00 00 06 98 01 7f 03 
00 00 48 0f fd 19 00 00 06 99 01 7f 03 00 00 49 0f a4 19 00 00 06 9a 01 7f 03 00 00 4a 0f df 1d 
00 00 06 9b 01 7f 03 00 00 4b 00 06 04 b9 03 00 00 06 04 cb 07 00 00 06 04 8b 06 00 00 06 04 30 
00 00 00 0e be 1a 00 00 0c 06 a5 01 0c 08 00 00 0f 3f 1a 00 00 06 a6 01 18 08 00 00 00 0f 23 19 
00 00 06 a7 01 30 00 00 00 04 0f 5b 19 00 00 06 a8 01 30 00 00 00 08 00 06 04 d7 07 00 00 06 04 
18 08 00 00 06 04 55 05 00 00 0d 7f 1b 00 00 06 9c 01 8b 06 00 00 16 08 06 b6 01 4e 08 00 00 0f 
7c 17 00 00 06 b7 01 98 08 00 00 00 0f 35 a6 00 00 06 b8 01 98 08 00 00 04 00 0e d7 1a 00 00 20 
06 b1 01 98 08 00 00 0f 35 a6 00 00 06 b2 01 5a 04 00 00 00 17 74 74 00 06 b2 01 7f 03 00 00 04 
0f d4 1b 00 00 06 b2 01 7f 03 00 00 05 17 76 00 
6b 00 3a ec 91 24 40 82 20 00 00 2a 15 00 00 2e 01 53 02 08 76 00 2d 66 92 24 40 59 0e 00 00 2e 
01 52 02 7d 00 00 00 00 00 34 c0 0f 00 00 e7 15 00 00 32 63 6c 00 01 27 01 2b 13 00 00 44 c7 00 
00 43 f1 12 00 00 ec 92 24 40 d8 0f 00 00 01 29 01 39 06 13 00 00 57 c7 00 00 39 fd 12 00 00 6a 
c7 00 00 2a 09 93 24 40 22 00 00 00 9e 15 00 00 3d 15 13 00 00 7d c7 00 00 2d 20 93 24 40 59 0e 
00 00 2e 01 52 02 7d 00 00 00 34 f8 0f 00 00 d5 15 00 00 3d 20 13 00 00 90 c7 00 00 3a 46 93 24 
40 59 0e 00 00 c4 15 00 00 2e 01 52 02 7d 00 00 2d 64 93 24 40 59 0e 00 00 2e 01 52 02 7d 00 00 
00 2d fc 92 24 40 59 0e 00 00 2e 01 52 02 7d 00 00 00 00 2a 88 93 24 40 e1 00 00 00 d7 16 00 00 
32 74 68 00 01 2e 01 3d 02 00 00 a3 c7 00 00 46 31 13 00 00 a2 93 24 40 b4 00 00 00 01 33 01 39 
48 13 00 00 b6 c7 00 00 39 3e 13 00 00 c9 c7 00 00 2f a2 93 24 40 b4 00 00 00 3d 52 13 00 00 ea 
c7 00 00 3d 5c 13 00 00 fd c7 00 00 3d 68 13 00 00 10 c8 00 00 47 74 13 00 00 09 94 24 40 4d 00 
00 00 01 11 01 b0 16 00 00 39 89 13 00 00 23 c8 00 00 39 80 13 00 00 36 c8 00 00 2f 09 94 24 40 
4d 00 00 00 3d 94 13 00 00 49 c8 00 00 3d 9f 13 00 00 8a c8 00 00 3a 38 94 24 40 9c 20 00 00 9e 
16 00 00 2e 01 52 02 7c 00 00 2d 56 94 24 40 b2 20 00 00 2e 01 52 02 7c 00 00 00 00 3a b8 93 24 
40 59 0e 00 00 c4 16 00 00 2e 01 52 02 7d 00 00 2d ed 93 24 40 59 0e 00 00 2e 01 52 02 7d 00 00 
00 00 00 44 10 10 00 00 32 70 00 01 38 01 d3 13 00 00 a4 c8 00 00 43 ab 13 00 00 6e 94 24 40 28 
10 00 00 01 3a 01 39 c0 13 00 00 b7 c8 00 00 39 b7 13 00 00 ca c8 00 00 44 28 10 00 00 3d c9 13 
00 00 dd c8 00 00 3a a0 94 24 40 59 0e 00 00 2e 17 00 00 2e 01 52 02 7d 00 00 2d e5 94 24 40 59 
0e 00 00 2e 01 52 02 7d 00 00 00 00 00 00 35 57 2f 00 00 01 47 01 37 00 00 00 58 95 24 40 2c 00 
00 00 01 9c 89 17 00 00 31 67 00 01 47 01 d7 0d 00 00 11 c9 00 00 32 6d 00 01 48 01 37 00 00 00 
3d c9 00 00 2d 6e 95 24 40 d9 13 00 00 2e 01 52 02 7d 00 00 00 48 d9 30 00 00 01 80 37 00 00 00 
84 95 24 40 b0 00 00 00 01 9c 0f 18 00 00 29 4c 00 01 80 3d 02 00 00 67 c9 00 00 29 61 6c 6c 00 
01 80 30 00 00 00 88 c9 00 00 2b 67 00 01 81 d7 0d 00 00 b3 c9 00 00 49 d1 2f 00 00 01 82 37 00 
00 00 c6 c9 00 00 2b 70 00 01 83 ce 0b 00 00 f0 c9 00 00 49 5c 30 00 00 01 84 5a 04 00 00 19 ca 
00 00 2c cd 95 24 40 a1 1f 00 00 2d f0 95 24 40 d1 1f 00 00 2e 01 53 01 32 00 00 26 b9 2f 00 00 
01 11 02 01 3d 18 00 00 24 4c 00 01 11 02 3d 02 00 00 27 67 00 01 12 02 d7 0d 00 00 4a 36 30 00 
00 01 13 02 37 00 00 00 00 26 bc 30 00 00 01 07 02 01 60 18 00 00 24 67 00 01 07 02 d7 0d 00 00 
27 75 76 00 01 08 02 f6 09 00 00 00 21 e4 2f 00 00 01 73 01 7f 18 00 00 22 67 00 01 73 d7 0d 00 
00 3f 75 00 01 74 5a 04 00 00 00 26 64 2f 00 00 
74 00 06 35 74 03 00 00 04 09 d4 1b 00 00 06 35 74 03 00 00 05 00 06 04 b9 03 00 00 
```
I googled the `system param error` but I only found info related to Arduino builds but they suggested fixing wrong settings to:

> Eventually the solution for me was pretty simple: Make sure you set the flash size in the Arduino IDE to 1Mbyte when you flash the ESP-01E. These are my complete settings for flashing the chip: generic module / 80 Mhz / QIO / 40 MHz / 1M (64kb) / 230400

But I couldnt find anything related (like flash size and so on) to the in the `Makefile`. Did I miss something? Could you give me a little hint to fix that? I bet its just a stupid mistake and/or something obvious but I really appreciate any hint.

Best Lny

Aeprox, 2015-08-24:
There's the latest release available [here](https://github.com/nodemcu/nodemcu-firmware/releases) if you're looking for 'stable' versions. If you want newer builds you'll have to build your own as you did.

I haven't encountered that error myself, but you might try to use [nodemcu's flasher](https://github.com/nodemcu/nodemcu-flasher), which allows you to set the settings you've mentioned, right before flashing. AFAIK these settings are needed to specify the lay-out and speed of the flash chip while flashing, and aren't part of the compiling process, so no need to modify the makefile.

Hope this helps you along!

Aeprox

milkpirate, 2015-08-25:
Hey,

thank you very much. Does the `flash_size` parameter take Bit or Byte? I checked the marking on the SPI flash and it is 4Mbit ([datasheet](http://akizukidenshi.com/download/ds/sst/25051A.pdf)). But still `esptool`throws:
```
Connecting...
Traceback (most recent call last):
  File "./tools/esptool.py", line 542, in <module>
    assert len(args.addr_filename) % 2 == 0
AssertionError
```
and if I flash it with nodemcu's flasher (and corresponding advanced settings) the ESP gives:
```
 ets Jan  8 2013,rst cause:1, boot mode:(3,7)

load 0x40100000, len 31816, room 16
tail 8
chksum 0x41
load 0x3ffe8000, len 3016, room 0
tail 8
chksum 0xef
load 0x3ffe8bc8, len 8, room 0
tail 8
chksum 0x2f
csum 0x2f
system param error
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
6b 00 3a ec 91 24 40 82 20 00 00 2a 15 00 00 2e 01 53 02 08 76 00 2d 66 92 24 40 59 0e 00 00 2e
01 52 02 7d 00 00 00 00 00 34 c0 0f 00 00 e7 15 00 00 32 63 6c 00 01 27 01 2b 13 00 00 44 c7 00
00 43 f1 12 00 00 ec 92 24 40 d8 0f 00 00 01 29 01 39 06 13 00 00 57 c7 00 00 39 fd 12 00 00 6a
c7 00 00 2a 09 93 24 40 22 00 00 00 9e 15 00 00 3d 15 13 00 00 7d c7 00 00 2d 20 93 24 40 59 0e
00 00 2e 01 52 02 7d 00 00 00 34 f8 0f 00 00 d5 15 00 00 3d 20 13 00 00 90 c7 00 00 3a 46 93 24
40 59 0e 00 00 c4 15 00 00 2e 01 52 02 7d 00 00 2d 64 93 24 40 59 0e 00 00 2e 01 52 02 7d 00 00
00 2d fc 92 24 40 59 0e 00 00 2e 01 52 02 7d 00 00 00 00 2a 88 93 24 40 e1 00 00 00 d7 16 00 00
32 74 68 00 01 2e 01 3d 02 00 00 a3 c7 00 00 46 31 13 00 00 a2 93 24 40 b4 00 00 00 01 33 01 39
48 13 00 00 b6 c7 00 00 39 3e 13 00 00 c9 c7 00 00 2f a2 93 24 40 b4 00 00 00 3d 52 13 00 00 ea
c7 00 00 3d 5c 13 00 00 fd c7 00 00 3d 68 13 00 00 10 c8 00 00 47 74 13 00 00 09 94 24 40 4d 00
00 00 01 11 01 b0 16 00 00 39 89 13 00 00 23 c8 00 00 39 80 13 00 00 36 c8 00 00 2f 09 94 24 40
4d 00 00 00 3d 94 13 00 00 49 c8 00 00 3d 9f 13 00 00 8a c8 00 00 3a 38 94 24 40 9c 20 00 00 9e
16 00 00 2e 01 52 02 7c 00 00 2d 56 94 24 40 b2 20 00 00 2e 01 52 02 7c 00 00 00 00 3a b8 93 24
40 59 0e 00 00 c4 16 00 00 2e 01 52 02 7d 00 00 2d ed 93 24 40 59 0e 00 00 2e 01 52 02 7d 00 00
00 00 00 44 10 10 00 00 32 70 00 01 38 01 d3 13 00 00 a4 c8 00 00 43 ab 13 00 00 6e 94 24 40 28
10 00 00 01 3a 01 39 c0 13 00 00 b7 c8 00 00 39 b7 13 00 00 ca c8 00 00 44 28 10 00 00 3d c9 13
00 00 dd c8 00 00 3a a0 94 24 40 59 0e 00 00 2e 17 00 00 2e 01 52 02 7d 00 00 2d e5 94 24 40 59
0e 00 00 2e 01 52 02 7d 00 00 00 00 00 00 35 57 2f 00 00 01 47 01 37 00 00 00 58 95 24 40 2c 00
00 00 01 9c 89 17 00 00 31 67 00 01 47 01 d7 0d 00 00 11 c9 00 00 32 6d 00 01 48 01 37 00 00 00
3d c9 00 00 2d 6e 95 24 40 d9 13 00 00 2e 01 52 02 7d 00 00 00 48 d9 30 00 00 01 80 37 00 00 00
84 95 24 40 b0 00 00 00 01 9c 0f 18 00 00 29 4c 00 01 80 3d 02 00 00 67 c9 00 00 29 61 6c 6c 00
01 80 30 00 00 00 88 c9 00 00 2b 67 00 01 81 d7 0d 00 00 b3 c9 00 00 49 d1 2f 00 00 01 82 37 00
[... cut ...]
00 00 2c cd 95 24 40 a1 1f 00 00 2d f0 95 24 40 d1 1f 00 00 2e 01 53 01 32 00 00 26 b9 2f 00 00
01 11 02 01 3d 18 00 00 24 4c 00 01 11 02 3d 02 00 00 27 67 00 01 12 02 d7 0d 00 00 4a 36 30 00
00 01 13 02 37 00 00 00 00 26 bc 30 00 00 01 07 02 01 60 18 00 00 24 67 00 01 07 02 d7 0d 00 00
27 75 76 00 01 08 02 f6 09 00 00 00 21 e4 2f 00 00 01 73 01 7f 18 00 00 22 67 00 01 73 d7 0d 00
00 3f 75 00 01 74 5a 04 00 00 00 26 64 2f 00 00
74 00 06 35 74 03 00 00 04 09 d4 1b 00 00 06 35 74 03 00 00 05 00 06 04 b9 03 00 00
```

Aeprox, 2015-08-25:
My bad, I wrote those command down without checking them first. I wasn't at my linux machine at the time. These are the correct commands

```
make all
srec_cat -output ./bin/nodemcu.bin -binary ./bin/0x00000.bin -binary -fill 0xff 0x00000 0x10000 ./bin/0x10000.bin -binary -offset 0x10000
./tools/esptool.py write_flash --flash_freq 40m --flash_mode qio --flash_size 4m 0x00000 ./bin/nodemcu.bin
```

Looking at esptool's repo [here](https://github.com/themadinventor/esptool), which mentions the flash size parameter in slightly more detail in the file header section, I'm assuming the parameter is in bits?

However, this doesn't seem to be an issue related to the firmware itself, so I'm gonna quote marcelstoer before @TerryE comes along and closes the issue :)

> Welcome to the esp8266 world. This issue list is not a support forum, use http://www.esp8266.com/viewforum.php?f=17 for that. Other helpful resources:
>
>  - http://www.esp8266.com/wiki/doku.php
>  - http://nodemcu.com/index_en.html
>

milkpirate, 2015-08-25:
Ok great. Will switch over. Wasnt sure about the root of the problem. Thank you again!

PS: New commands work now. Still couldnt get the nodemcu to run correctly. Will ask in esp8266 forum again.

closed 2015-08-29 by milkpirate

silenceuncrio, 2016-01-06:
FYR,
I encounter the same problem and solve by the following parameters while flash my ESP-01E
[image]