nodemcu/nodemcu-firmware #614: nodemcu will reboot after dsleep (regardless of the parameters used for sleeptype and dsleep)

aguaviva, 2015-08-24:
My GPIO16 and the RST pins are connected.
This happens regardless of the time specified in the dsleep call (so it's not a watchdog issue)

Aeprox, 2015-08-24:
Isn't this default behavior? What does node.bootreason() say?

aguaviva, 2015-08-24:
It returns 2. No idea what it means, I can see the source code calling rtc_get_reset_reason but google didnt show up any results.

Aeprox, 2015-08-24:
Source code mentions 
```
// 2, 3 = reset (module dependent?), 4 = wdt
int rtc_get_reset_reason (void);
```
I too have no idea what that means. However, espressif docs read the following:

> Configures chip for deep-sleep mode. When the device is in deep-sleep, it
> automatically wakes up periodically; the period is configurable. Upon waking
> up, the device boots up from user_init

I believe this means that the whole lua system (nodeMCU) will reboot after deep-sleep.

TerryE, 2015-08-25:
> I believe this means that the whole lua system (nodeMCU) will reboot after deep-sleep.

Correct.  `user_int()` is part of the Lua startup on the nodeMCU firmware.  This cascades into lua_init which reinitialises memory and invokes `init.lua`, etc.  To get nodeMCU to support deep sleep correctly would be an enhancement.

aguaviva, 2015-08-25:
Ok, so to properly handle this I have to call node.bootreason() and then react appropriatelly (I might not have to reconnect to the wifi if the modem was left on) Any ideas? Answering to this in a wiki page is also welcome :)

jmattsson, 2015-08-28:
In deep sleep the chip is switched off, and all state is lost*. The only thing running is the RTC, which is the one responsible for restarting the chip at the end of the sleep period. When the chip is woken up again, it is effectively a reboot. 

The following boot reasons exist: (1) power-up, (2) reset-1, (3) reset-2, (4) watchdog reset, (5) rtc reset. I'm still hazy on reasons 2 and 3 and when you get which one. It's on my to-do list to work this out properly (and then add some named constants ideally). I'd love it if someone beat me to it...

What I can tell you is that you will never in practice see the rtc-reset cause (5). From what I have been able to deduce, this is what was meant to have happened after a deep sleep, but for whatever reason the boot loader in ROM fails to do something to the SPI controller and/or flash, and therefore fails to actually load the user software in this case. This is why GPIO16 has to be connected to the RST line as a workaround.

As an aside, there *might* be a possibility of a workaround if this code path is fully understood and then *possibly* compensated for right before going into deep sleep. For example, if the underlying cause is that the SPI flash is left in power-down mode when going to sleep and what the ROM boot loader doesn't do is bring it out of that, it might be possible to compensate for that. It's a long shot, but if someone has plenty of time and likes going through LX106 disassembly... :)

TL;DR: Yes, this is expected behaviour. Deep sleep discards state and does a restart on wake-up. Looking at node.bootreason() during startup is the best you can do I think (it's what we do here at work).

*) Except things squirreled away in RTC memory.

TerryE, 2015-12-22:
This is a feature of the SDK and the nodeMCU architecture. We can't "fix" this.

closed 2015-12-22 by TerryE

nickandrew, 2015-12-23:
@jmattsson Hmm, I've got a different list out of the SDK doc:

  *  0 = Normal startup by power on
  *  1 = Hardware watchdog reset
  *  2 = Exception reset
  *  3 = Software watchdog reset
  *  4 = Soft restart
  *  5 = Wake from deep sleep
  *  6 = External system reset

TerryE, 2015-12-23:
OK, but I think that what you are saying is that we need to fix the documentation.  The nodemcu firmware does reinitialise after deep sleep so any context needs to be stored in RTCmem or SPIFFS.

jmattsson, 2015-12-23:
@nickandrew Those are the values you get if you use the new `rst_info` struct in the more recent SDKs. The old node.bootreason() exposes the "raw" values. I think I saw some talk about extending that function to also return the new values, but don't quote me on specifics.