nodemcu/nodemcu-firmware #615: Cannot use MQTT with All 8 GPIOs

openhardwarecoza, 2015-08-25:
Hi.  I have a board/sketch that works with MQTT, switches 4 mosfets and (hopefully) reads 4 tactile buttons

If I upload this sketch, all works fine (3 buttons + 4 mosfets only)

If I add the 4th button, it doesnt complete the MQTT subscription.  Still shows 8986 free in node.heap?

Working sketch:  -- One time ESP Setup --
wifi.setmode(wifi.STATION)
wifi.sta.config ( "openhardwarecoza" , "novell1234" ) 
print(wifi.sta.getip())

Broker="192.168.1.200"

 --GPIOs connected to Mosfets
 gpio.mode(6,gpio.OUTPUT) 
 gpio.write(6,gpio.LOW) 

 gpio.mode(5,gpio.OUTPUT) 
 gpio.write(5,gpio.LOW) 

 gpio.mode(0,gpio.OUTPUT) 
 gpio.write(0,gpio.LOW) 

 gpio.mode(7,gpio.OUTPUT) 
 gpio.write(7,gpio.LOW) 

 --GPIOs connected to Buttons
 gpio.mode(3,gpio.INPUT,gpio.PULLUP) 
 gpio.mode(4,gpio.INPUT,gpio.PULLUP) 
 gpio.mode(1,gpio.INPUT,gpio.PULLUP) 
 gpio.mode(8,gpio.INPUT,gpio.PULLUP) 

 function reconnect()
   print ("Waiting for Wifi")
   if wifi.sta.status() == 5 and wifi.sta.getip() ~= nil then 
     print ("Wifi Up!")
     tmr.stop(1) 
     m:connect(Broker, 1883, 0, function(conn) 
     print("Mqtt Connected to:" .. Broker) 
     mqtt_sub()
   end)
 end
 end
 

m = mqtt.Client("STUDY", 180, "user", "password") 
 m:lwt("/lwt", "ESP8266-2", 0, 0) 
 m:on("offline", function(con) 
 print ("Mqtt Reconnecting...") 
 tmr.alarm(1, 10000, 1, function() 
 reconnect()
 end) 
 end)
 

 
 -- on publish message receive event 
 m:on("message", function(conn, topic, data) 
-- print("Recieved:" .. topic .. ":" .. data) 
 
 if (data=="ON") and (topic=="/dev2/ch1/com") then 
 gpio.write(6,gpio.HIGH) 
 m:publish("/dev2/ch1/state","ON",0,0) 
 elseif (data=="OFF") and (topic=="/dev2/ch1/com") then 
 gpio.write(6,gpio.LOW) 
 m:publish("/dev2/ch1/state","OFF",0,0)
 elseif (data=="ON") and (topic=="/dev2/ch2/com") then 
 gpio.write(5,gpio.HIGH) 
 m:publish("/dev2/ch2/state","ON",0,0) 
 elseif (data=="OFF") and (topic=="/dev2/ch2/com") then 
 gpio.write(5,gpio.LOW) 
 m:publish("/dev2/ch2/state","OFF",0,0)
 elseif (data=="ON") and (topic=="/dev2/ch3/com") then 
 gpio.write(0,gpio.HIGH) 
 m:publish("/dev2/ch3/state","ON",0,0) 
 elseif (data=="OFF") and (topic=="/dev2/ch3/com") then 
 gpio.write(0,gpio.LOW) 
 m:publish("/dev2/ch3/state","OFF",0,0)
 elseif (data=="ON") and (topic=="/dev2/ch4/com") then 
 gpio.write(7,gpio.HIGH) 
 m:publish("/dev2/ch4/state","ON",0,0) 
 elseif (data=="OFF") and (topic=="/dev2/ch4/com") then 
 gpio.write(7,gpio.LOW) 
 m:publish("/dev2/ch4/state","OFF",0,0)
 end 
 end) 

-- Pin to also toggle the status so you don't have to keep pulling out your phone to turn on device
 gpio.trig(3, "down",function (level)  
    local PinValue=gpio.read(6)  
    --Change the state  
    if (PinValue==0) then  
      --The read resets the output to 0, put it back  
      gpio.write(6,gpio.HIGH)  
   m:publish("/dev2/ch1/state","ON",0,0) 
    else  
      gpio.write(6,gpio.LOW)  
 m:publish("/dev2/ch1/state","OFF",0,0) 
     end  
 end)  

 gpio.trig(4, "down",function (level)  
    local PinValue=gpio.read(5)  
    --Change the state  
    if (PinValue==0) then  
      --The read resets the output to 0, put it back  
      gpio.write(5,gpio.HIGH)  
      m:publish("/dev2/ch2/state","ON",0,0) 
    else  
      gpio.write(5,gpio.LOW)  
      m:publish("/dev2/ch2/state","OFF",0,0) 
     end  
 end)  

 gpio.trig(8, "down",function (level)  
    local PinValue=gpio.read(0)  
    --Change the state  
    if (PinValue==0) then  
      --The read resets the output to 0, put it back  
      gpio.write(0,gpio.HIGH)  
      m:publish("/dev2/ch3/state","ON",0,0) 
    else  
     gpio.write(0,gpio.LOW)  
     m:publish("/dev2/ch3/state","OFF",0,0) 
     end  
 end)  

 gpio.trig(1, "down",function (level)  
    local PinValue=gpio.read(7)  
    --Change the state  
    if (PinValue==0) then  
      --The read resets the output to 0, put it back  
      gpio.write(7,gpio.HIGH)  
      m:publish("/dev2/ch4/state","ON",0,0) 
    else  
     gpio.write(7,gpio.LOW)  
     m:publish("/dev2/ch4/state","OFF",0,0) 
     end  
 end)  
 
 
 function mqtt_sub() 
 m:subscribe("/dev2/#",0, function(conn) 
 print("Mqtt Subscribed to OpenHAB feed for device STUDY") 
 end) 
 end 
 
 tmr.alarm(0, 1000, 1, function() 
 print ("Connecting to Wifi... ")
 if wifi.sta.status() == 5 and wifi.sta.getip() ~= nil then 
 print ("Wifi connected")
 tmr.stop(0) 
 m:connect(Broker, 1883, 0, function(conn) 
 print("Mqtt Connected to:" .. Broker) 
 mqtt_sub() --run the subscription function 
 end) 
 end
 
 end)

Not working sketch

pastukhov, 2015-08-25:
Try to make 1 subscribtion and 1 publish. And choose what the pin use with payload.

openhardwarecoza, 2015-08-26:
Its the m:connect that doesnt complete.  I commented out all the publish events and still same (i already only subscribe once)

karrots, 2015-11-10:
There have been a bunch of MQTT fixes merged into the `dev` branch since this issue opened. Please test again with a new build based on the `dev` branch and report back.

openhardwarecoza, 2015-11-10:
What is the quickest way to get a prebuilt .bin? (:  I am not a machine where I can get a dev environment setup in a flash.

marcelstoer, 2015-11-10:
@openhardwarecoza http://frightanic.com/nodemcu-custom-build/index.php or https://hub.docker.com/r/marcelstoer/nodemcu-build/

closed 2016-02-07 by openhardwarecoza