nodemcu/nodemcu-firmware #620: File.read .write and .seek behaviour when file is closed.

sfrederiks, 2015-08-29:
When these functions are called and the file is closed execution stops with an error message. In the documentation, it is stated that they should return with nil.

With Lua's ability to make some highly parallel programs, it is sometimes necessary to be able to test if a file is open or closed. If the code was implemented as documented this would be possible.

Example:

file.open('xxx.xxx', 'a+')
file.close()
if file.writeline('something') then
  print('File is open')
else
  print('File is closed')
end
print('something else')

This gives the following response:

NodeMCU 0.9.6 build 20150704  powered by Lua 5.1.4
lua: init.lua:18: open a file first

jmattsson, 2015-09-01:
Not addressing this issue per se, but here are two suggestions:
- use the return value of file.open()
- use Lua's pcall() function to catch runtime errors

sfrederiks, 2015-09-01:
Thanks for the advice on how to make workarounds.

Working more with the file library, it is clear that it ought to be both validated and verified for the return values from the different routines and the documentation.

TerryE, 2015-09-01:
> Working more with the file library, it is clear that it ought to be both validated and verified for the return values from the different routines and the documentation.

Sorry, but I think that this is your interpretation rather than what is said: "returns true: write OK. nil: there is error".

Looking at modules/

closed 2015-09-01 by TerryE