nodemcu/nodemcu-firmware #631: nodemcu reboot on builds 0.9.5 and 0.9.6 when counting down from 1000 to 0.

Seagoon, 2015-09-02:
Printing the heap in the example below doesn't show it decreasing. Esplorer error is "ESP module reboot detected! Event: internal NodeMCU exception or power fail. Please try again." I've tried this on two different NodeMCU modules (by different manufacturers) and see the same problem. Both firmwares are the float versions.

i = 1000
   repeat
     print(i)
     i= i-1
   until i == 0
print(i)

On both the builds mentioned in the title, nodeMCU reboots around the half way mark:

438
437
436
435
434
433
4bBÊ1
ü¤„ÿ1ÂJ¤äÎÆÍœ¦èè

NodeMCU 0.9.6 build 20150704  powered by Lua 5.1.4

679
678
677
676
675
6÷ü!,£<1ä)ÅäNä)]†1¤ñø

NodeMCU 0.9.5 build 20150318  powered by Lua 5.1.4

TerryE, 2015-09-02:
Sounds to me like you've got a problem with your Vcc. Have you a decent capacitor, sy 220uF between Vcc and Gnd?

Seagoon, 2015-09-02:
Hi,

No. I'm using a USB powered NodeMCU like the one linked below. I have a couple of versions that are clearly made by different manufacturers, but I guess it's possible they both skimped on the power side.
http://www.seeedstudio.com/depot/NodeMCU-v2-Lua-based-ESP8266-development-kit-p-2415.html

I've tried them both on a powered USB hub and a direct USB port on my PC. I assume this isn't an issue with a raw esp8266 module? I don't have one to test on.

Thanks,

pastukhov, 2015-09-02:
The reason of reboot is watchdog. 
```lua
    for i=1,10000 do 
      print(i)
      -- should call tmr.wdclr() in a long loop 
      --to avoid hardware reset caused by watchdog.
      tmr.wdclr()   
    end
```
from http://www.nodemcu.com/docs/timer-module/#timer-module-tmr-wdclr

Seagoon, 2015-09-02:
@pastukhov @TerryE Thank you both! That's exactly it. I'll close this.

closed 2015-09-02 by Seagoon

TerryE, 2015-09-02:
@pastukhov Artem, durrrhhh I missed that one.  I didn't think about the 10000!  The watchdog timer also serves another purpose.  See my discussion [When and why should I avoid using tmr.delay()?](http://www.esp8266.com/wiki/doku.php?id=nodemcu-unofficial-faq#when_and_why_should_i_avoid_using_tmrdelay).  If you sit in Lua for too long, then bad things start to happen.

Seagoon, 2015-09-03:
Thanks @TerryE. The example in my case was only 1000, which seems pretty aggressive. But I'm new to Lua and this device and I'll read your discussion. Thanks again to both of you. I should probably have posted this in a forum but I was convinced it was a bug.