nodemcu/nodemcu-firmware #632: cross-lua build error

eyaleb, 2015-09-03:
Now that https://github.com/nodemcu/nodemcu-firmware/issues/598 is committed I decided to do a build of master.

I used the command
```sh
$ lua tools/cross-lua.lua
The program 'lua' can be found in the following packages:
 * lua5.2
 * lua5.1
 * lua50
Try: sudo apt-get install <selected package>
```
I need lua installed... so did so
```sh
$ sudo apt-get install lua5.2
$ lua tools/cross-lua.lua
lua: ./tools/build.lua:5: module 'lfs' not found:
        no field package.preload['lfs']
        no file '/usr/local/share/lua/5.2/lfs.lua'
        no file '/usr/local/share/lua/5.2/lfs/init.lua'
        no file '/usr/local/lib/lua/5.2/lfs.lua'
        no file '/usr/local/lib/lua/5.2/lfs/init.lua'
        no file '/usr/share/lua/5.2/lfs.lua'
        no file '/usr/share/lua/5.2/lfs/init.lua'
        no file './lfs.lua'
        no file '/usr/local/lib/lua/5.2/lfs.so'
        no file '/usr/lib/x86_64-linux-gnu/lua/5.2/lfs.so'
        no file '/usr/lib/lua/5.2/lfs.so'
        no file '/usr/local/lib/lua/5.2/loadall.so'
        no file './lfs.so'
stack traceback:
        [C]: in function 'require'
        ./tools/build.lua:5: in main chunk
        [C]: in function 'require'
        tools/cross-lua.lua:2: in main chunk
        [C]: in ?
```
Now I had to guess.. tried this:
```sh
$ sudo apt-get install lua-filesystem
$ lua tools/cross-lua.lua
[COMPILE] .build/cross-lua/app__lua__lapi.o
[builder] Error building target
[builder] Last executed command was: 
gcc -O2 -Iapp/include -Iinclude -Iapp/lua -Wall -DLUA_CROSS_COMPILER -O2 -c -MD -MF .build/cross-lua/app__lua__lapi.d app/lua/lapi.c -o .build/cross-lua/app__lua__lapi.o
eyal@esp8266:/opt/Espressif/nodemcu-firmware-master$ gcc -O2 -Iapp/include -Iinclude -Iapp/lua -Wall -DLUA_CROSS_COMPILER -O2 -c -MD -MF .build/cross-lua/app__lua__lapi.d app/lua/lapi.c -o .build/cross-lua/app__lua__lapi.o
```

This is as far as I got, I could not find any error log to tell me what went wrong. Did some searching but still did not find an answer. This is what I have in the build directory, it did not go far:

```sh
$ ls -lRa .build/
.build/:
total 12
drwxrwxr-x  3 eyal eyal 4096 Sep  3 13:29 .
drwxrwxr-x 15 eyal eyal 4096 Sep  3 13:29 ..
drwxrwxr-x  2 eyal eyal 4096 Sep  3 13:31 cross-lua

.build/cross-lua:
total 40
drwxrwxr-x 2 eyal eyal  4096 Sep  3 13:31 .
drwxrwxr-x 3 eyal eyal  4096 Sep  3 13:29 ..
-rw-rw-r-- 1 eyal eyal  2108 Sep  3 13:31 app__lua__lapi.d
-rw-rw-r-- 1 eyal eyal 20992 Sep  3 13:31 app__lua__lapi.o
-rw-rw-r-- 1 eyal eyal   110 Sep  3 13:29 .builddata.comp
```

However, each time that I do `make clean` it leaves the .build alone, builds one other module and then fails as before. After the third attempt I have:

```sh
$ lua tools/cross-lua.lua
[COMPILE] .build/cross-lua/app__lua__lbaselib.o
[builder] Error building target
[builder] Last executed command was: 
gcc -O2 -Iapp/include -Iinclude -Iapp/lua -Wall -DLUA_CROSS_COMPILER -O2 -c -MD -MF .build/cross-lua/app__lua__lbaselib.d app/lua/lbaselib.c -o .build/cross-lua/app__lua__lbaselib.o
eyal@esp8266:/opt/Espressif/nodemcu-firmware-master$ ls -lRa .build/
.build/:
total 12
drwxrwxr-x  3 eyal eyal 4096 Sep  3 13:29 .
drwxrwxr-x 15 eyal eyal 4096 Sep  3 13:29 ..
drwxrwxr-x  2 eyal eyal 4096 Sep  3 14:16 cross-lua

.build/cross-lua:
total 104
drwxrwxr-x 2 eyal eyal  4096 Sep  3 14:16 .
drwxrwxr-x 3 eyal eyal  4096 Sep  3 13:29 ..
-rw-rw-r-- 1 eyal eyal  2108 Sep  3 13:31 app__lua__lapi.d
-rw-rw-r-- 1 eyal eyal 20992 Sep  3 13:31 app__lua__lapi.o
-rw-rw-r-- 1 eyal eyal  2678 Sep  3 14:15 app__lua__lauxlib.d
-rw-rw-r-- 1 eyal eyal 21368 Sep  3 14:15 app__lua__lauxlib.o
-rw-rw-r-- 1 eyal eyal  2395 Sep  3 14:16 app__lua__lbaselib.d
-rw-rw-r-- 1 eyal eyal 26928 Sep  3 14:16 app__lua__lbaselib.o
-rw-rw-r-- 1 eyal eyal   110 Sep  3 14:16 .builddata.comp
-rw-rw-r-- 1 eyal eyal    44 Sep  3 14:16 .builddata.link
```

I am running in a VirtualBox image with lubuntu-14.10-desktop-amd64.iso installed.

```sh
$ uname -a
Linux esp8266 3.16.0-31-generic #43-Ubuntu SMP Tue Mar 10 17:37:36 UTC 2015 x86_64 x86_64 x86_64 GNU/Linux
```

My guess: I am missing some very secret critical component.

TIA

cybersquid, 2015-09-03:
Hi Eyaleb:

luafilesystem (lfs) is a Lua package (called a Rock in Lua parlance).  If luarocks isn’t installed do “sudo apt-get install luarocks”.
Then you can “man luarocks” to learn how to install luafilesystem.

Also, I note you have installed Lua 5.2.  eLua (and nodemcu) are gbased on 5.1.
For comparison, here are all the deb packages on my system with “lua” in their names:

Desired=Unknown/Install/Remove/Purge/Hold
| Status=Not/Inst/Conf-files/Unpacked/halF-conf/Half-inst/trig-aWait/Trig-pend
|/ Err?=(none)/Reinst-required (Status,Err: uppercase=bad)
||/ Name                                 Version                              Architecture Description
+++-====================================-====================================-============-===============================================================================

ii  liblua5.1-0:i386                     5.1.5-5ubuntu0.1                     i386         Shared library for the Lua interpreter version 5.1
ii  liblua5.1-0-dev:i386                 5.1.5-5ubuntu0.1                     i386         Development files for the Lua language version 5.1
ii  liblua5.2-0:i386                     5.2.3-1                              i386         Shared library for the Lua interpreter version 5.2
ii  lua-mode                             20130419-3                           all          Emacs mode for editing Lua programs
ii  lua5.1                               5.1.5-5ubuntu0.1                     i386         Simple, extensible, embeddable programming language
ii  luarocks                             2.0.9-1                              all          deployment and management system for Lua modules
ii  luatex                               0.76.0-3ubuntu1                      i386         next generation TeX engine
ii  texlive-luatex                       2013.20140215-1                      all          TeX Live: LuaTeX packages

   Hope that helps,
	- dash

> 
> Now that #598 is committed I decided to do a build of master.
> 
> I used the command
> 
> $ lua tools/cross-lua.lua
> The program
> 'lua' can be found in
>  the following packages:
> 
> *
>  lua5.2
> 
> *
>  lua5.1
> 
> *
>  lua50
> Try: sudo apt-get install
> <selected package>
> I need lua installed... so did so
> 
> $ sudo apt-get install lua5.2
> $ lua tools/cross-lua.lua
> lua: ./tools/build.lua:5: module
> 'lfs'
>  not found:
>         no field package.preload[
> 'lfs'
> ]
>         no file
> '/usr/local/share/lua/5.2/lfs.lua'
> 
>         no file
> '/usr/local/share/lua/5.2/lfs/init.lua'
> 
>         no file
> '/usr/local/lib/lua/5.2/lfs.lua'
> 
>         no file
> '/usr/local/lib/lua/5.2/lfs/init.lua'
> 
>         no file
> '/usr/share/lua/5.2/lfs.lua'
[... 82 lines cut ...]
> drwxrwxr-x 15 eyal eyal 4096 Sep  3 13:29 ..
> drwxrwxr-x  2 eyal eyal 4096 Sep  3 14:16 cross-lua
> 
> .build/cross-lua:
> total 104
> drwxrwxr-x 2 eyal eyal  4096 Sep  3 14:16
> 
> drwxrwxr-x 3 eyal eyal  4096 Sep  3 13:29 ..
> -rw-rw-r-- 1 eyal eyal  2108 Sep  3 13:31 app__lua__lapi.d
> -rw-rw-r-- 1 eyal eyal 20992 Sep  3 13:31 app__lua__lapi.o
> -rw-rw-r-- 1 eyal eyal  2678 Sep  3 14:15 app__lua__lauxlib.d
> -rw-rw-r-- 1 eyal eyal 21368 Sep  3 14:15 app__lua__lauxlib.o
> -rw-rw-r-- 1 eyal eyal  2395 Sep  3 14:16 app__lua__lbaselib.d
> -rw-rw-r-- 1 eyal eyal 26928 Sep  3 14:16 app__lua__lbaselib.o
> -rw-rw-r-- 1 eyal eyal   110 Sep  3 14:16 .builddata.comp
> -rw-rw-r-- 1 eyal eyal    44 Sep  3 14:16 .builddata.link
> 
> I am running in a VirtualBox image with lubuntu-14.10-desktop-amd64.iso installed.
> 
> $ uname -a
> Linux esp8266 3.16.0-31-generic
> #43-Ubuntu SMP Tue Mar 10 17:37:36 UTC 2015 x86_64 x86_64 x86_64 GNU/Linux
> My guess: I am missing some very secret critical component.
> 
> TIA

eyaleb, 2015-09-03:
Thanks @cybersquid .

I removed the old packages and installed new ones as suggested (I still needed to guess that the missing `lfs.so` relates to the package `luafilesystem`).
```sh
$ sudo apt-get install lua5.1
$ sudo apt-get install luarocks
$ sudo luarocks install luafilesystem
$ make clean
$ rm -rf .build/*
$ lua tools/cross-lua.lua
[COMPILE] .build/cross-lua/app__lua__lapi.o
[COMPILE] .build/cross-lua/app__lua__lauxlib.o
[COMPILE] .build/cross-lua/app__lua__lbaselib.o
[COMPILE] .build/cross-lua/app__lua__lcode.o
[COMPILE] .build/cross-lua/app__lua__ldblib.o
[COMPILE] .build/cross-lua/app__lua__ldebug.o
[COMPILE] .build/cross-lua/app__lua__ldo.o
[COMPILE] .build/cross-lua/app__lua__ldump.o
[COMPILE] .build/cross-lua/app__lua__lfunc.o
[COMPILE] .build/cross-lua/app__lua__lgc.o
[COMPILE] .build/cross-lua/app__lua__llex.o
[COMPILE] .build/cross-lua/app__lua__lmathlib.o
[COMPILE] .build/cross-lua/app__lua__lmem.o
[COMPILE] .build/cross-lua/app__lua__loadlib.o
[COMPILE] .build/cross-lua/app__lua__lobject.o
[COMPILE] .build/cross-lua/app__lua__lopcodes.o
[COMPILE] .build/cross-lua/app__lua__lparser.o
[COMPILE] .build/cross-lua/app__lua__lrotable.o
[COMPILE] .build/cross-lua/app__lua__lstate.o
[COMPILE] .build/cross-lua/app__lua__lstring.o
[COMPILE] .build/cross-lua/app__lua__lstrlib.o
[COMPILE] .build/cross-lua/app__lua__ltable.o
[COMPILE] .build/cross-lua/app__lua__ltablib.o
[COMPILE] .build/cross-lua/app__lua__ltm.o
[COMPILE] .build/cross-lua/app__lua__lundump.o
[COMPILE] .build/cross-lua/app__lua__lvm.o
[COMPILE] .build/cross-lua/app__lua__lzio.o
[COMPILE] .build/cross-lua/app__lua__luac_cross__luac.o
[COMPILE] .build/cross-lua/app__lua__luac_cross__loslib.o
[COMPILE] .build/cross-lua/app__lua__luac_cross__print.o
[COMPILE] .build/cross-lua/app__lua__..__modules__linit.o
[LINK] luac.cross
.build/cross-lua/app__lua__luac_cross__loslib.o: In function `os_tmpname':
loslib.c:(.text+0x29c): warning: the use of `tmpnam' is dangerous, better use `mkstemp'
[builder] Done building target.
```

I now have a new file:
```
-rwxrwxr-x  1 eyal eyal 193024 Sep  3 15:11 luac.cross
```
Now it is time to use it (find how to upload the lc binary files)...

Thanks

TerryE, 2015-09-03:
Thanks. I need need to add some documentation on this. As nodeMCU Lua is version 5.1, you need Lua 5.1 on your host as well. As discussed the cross build also needs LFS. The easiest way to do this on most Debian varieties such as Ubuntu and RPi is simply to do

```Bash
sudo apt_get install lua-5.1 lua-fileesystem
```

eyaleb, 2015-09-03:
Next issue. Took me a few hours to drill all the way down....

First the good: I used to get, every now and then, an unexplained `wrong arg range` error.
Using the cross compiler I got a much nicer `funcs.lua:69: wrong arg range` which was now trivial to fix. I love it already.

The bad. My app runs mostly correctly, only one thing was not right. Here is what I discovered. I have a tiny private module (`app/modules/misc.c`)that provides rtc read/write functionality. As is the practice it stores a magic number to indicate that the rtc memory was initialised. With the cross compiler it always considered the magic to be wrong.

The magic value stored was `0xbadadoda`. Turns out it was truncated to `0x7fffffff`, and only got through when I shortened it to `0x6adadoda` (so 31 bit integer).

My module uses this to fetch the value from lua
```lua
uint32_t data;
data = luaL_checkinteger(L,2); /* accept value from lua */
lua_pushinteger( L, data); /* return value to lua*/
```
My app works correctly when compiled on the esp8266 itself so I suspect the cross compiler. Or maybe I was lucky to get away with (my) wrong code all this time?

And a quick note: my main workstation runs a later lua5.3. Will the nodeMCU software move forward too? ATM I have a VM just for this development.

cheers

TerryE, 2015-09-03:
luac.cross accepts the -l argument to provide a VM opcode listing so you can see what the lua compiler generates.  The VM code should be identical to the inbuild lua generated.  You can use node.compile or string.dump on the esp8266 to generate an lc and then upload it to the host and use ChunkSpy (you need a fix for it recognize the eLua types).  The code should be identical.  Worth checking.  Also try:

```
$ ./luac.cross -l -o /dev/null -
print(0xabababab)
^D   
main <stdin:0,0> (4 instructions, 16 bytes at 0x9166860)
0+ params, 2 slots, 0 upvalues, 0 locals, 2 constants, 0 functions
	1	[1]	GETGLOBAL	0 -1	; print
	2	[1]	LOADK    	1 -2	; 2880154539
	3	[1]	CALL     	0 2 1
	4	[1]	RETURN   	0 1
```
2880154539 = 0xabababab

So this is OK.  Incidentally I've build luac.cross on my headless 32-bit dev Ubuntu 14/.04-LTS VM and my 64bit Ubuntu Laptop, and both generate the same code here.

I suspect that the issue that you have is nothing to do with Lua, but it's the code that the `xtensa-lx106-elf-gcc` cross compiler generates on your build.  Integers in Lua are a (signed) long type -- that is signed 32-bit on the esp8266, not uint32_t, and the compiler generates implicit conversion code for 
uint32_t to signed 32 bit. Either make data type long or use 0x5a5a5a5a as your magic pattern.

But what you should really do is store a flag in the Lua registry to store this type of context.  You don't need magic patterns, a standard boolean will do. If the registry entry doesn't exist then you haven't initialised the RTC yet.  See [Pil Sec 27.3.1](http://www.lua.org/pil/27.3.1.html) for more info.

eyaleb, 2015-09-03:
Thanks @TerryE,

Interesting, I was not aware of a non-volatile registry kept by lua. I use the RTC memory for this. I keep more than just the magic (it just protects the other entries) in RTC memory, and flash is not the right place for data that is updated at a high rate.

I will look into this though.

cheers

TerryE, 2015-09-03:
>  non-volatile registry kept by Lua.
Nope it is volatile and is reininitalised on reboot and deep sleep reawakening.

eyaleb, 2015-09-03:
So it cannot replace the RTC magic - it needs to persist across deep sleeps.

The cross compiler is important as right now my app cannot compile on the esp (not enough memory) without severe trimming. This is one problem I can avoid. Thanks.

As for the possibility that the xtensa compiler may be involved: I do not see how. All I that changed was the lua compilation phase. The firmware is still the same one (latest master) and the xtensa compiler is not involved now. But I do not know all that is involved in the cross compiler.

To make the discussion concrete, here is a test program (`test.lua`):
```lua
rtc_magic_address = 188
rtc_magic = 0xbadad0da

misc.rtc_mem_write_int(rtc_magic_address, rtc_magic)
print (string.format("write %08x", rtc_magic))
print (string.format("read %08x", misc.rtc_mem_read_int(rtc_magic_address)))
```
Compile and run on the esp:
```lua
> dofile("test.lua")
write badad0da
read badad0da
> node.compile("test.lua")
> dofile("test.lc")
write badad0da
read badad0da
> file.format()
format done.
```
Uploaded cross compiled module:
```lua
> dofile("test.lua")
write badad0da
read badad0da
> dofile("test.lc")
write badad0da
read 7fffffff
```
cheers

TerryE, 2015-09-04:
Brilliant.  Thanks for this analysis.  It should help me to track down the error :smile:

TerryE, 2015-09-04:
OK, I've found the error and it is an error.  The issue is in the logic of `ldump.c:DumpNumber()`.  Lua treats all numbers as the C **double** type and uses 64bit floating operations to handle numbers.  This works because ANSI floating point will store integers up to 2^52 with a zero exponent, so your magic pattern `0xbadad0da` is actually the integer 3134902490, so this is stored as the double 3134902490.0 which in hex  is 0x00, 0x00, 0x40, 0x1b, 0x5a, 0x5b, 0xe7, 0x41, or your 0, 0, 64, 27, 90, 91, 231, 65.

As far as I see, the esp8266 code goes through exactly the same path but its internal double formats are different so such numbers get mangled.  I'll need to put in extra conditional logic to convert ANSI double to xtensa lx 106 doubles.  More research is needed :disappointed:

TerryE, 2015-09-04:
I think that this is a bug or a 'feature' in nodemcu that you've accidentally uncovered.  If you do the following on a PC Lua:

```Lua
print (0x00007adad0da)
print (0x0000badad0da)
```

you get `3134902490` followed by `2061160666` which is what you'd expect from the language spec.  in the case of nodemcu Lua you get  `-1160064806` followed by `2061160666`, that is hex integer constants are handled in some bizarre way and not converted properly.  Doing something like 

```Lua
if 0x123456789 == 0xffffffffff then print "Bizarre!"  end
```

prints **Bizarre!** on the esp8266!!

TerryE, 2015-09-04:
OK, I've found the issue.  If the tokenizer comes across a putative number it calls `luaO_str2d()` which in turn calls `c_strtod()` on an integer type including 0xNNNN... constants and this only works properly for 31 bit signed numbers.  So my issue is :+1: 

*    Do I accept that the behaviour of luac.cross and the esp8266 are different on this edge cases?
*    Do I put a fix in the  `c_strtod()` code and potentially break existing esp8266 apps by changing their behaviour?
*     Do I put a fix into luac,cross to mirror the (broken) behaviour of the esp8266 runtime.

Opinions gratefully received.

eyaleb, 2015-09-04:
It seems that the esp accepts overflow fro 31bits to 32 bits
```lua
> =string.format("%d %x", 0x0badad0da, 0x0badad0da)
-1160064806 badad0da
```
but not beyond 32 bits
```lua
> =string.format("%d %x", 0x100000000, 0x100000000)
-1 ffffffff
```
while cross does not accept these.

My opinion is that adjusting cross to allow 32 bit integers is a reasonable and acceptable change.

TerryE, 2015-09-05:
Gosh this is complicated.  The boot ROM includes a bunch of functions etx_\* that are exposed to the firmware via `ld/eagle.rom.addr.v6.ld:115-164` and then mapped to os_\* names in `include/osapi.h:11-45`. The firmware largely uses `c_*` names which are mapped to these os_\* equivalents in the various

TerryE, 2015-09-06:
@eyelab, if you don't mind I'll close this issue since the underlying issue is one with `c_strtod()` which I've raised in #637

eyaleb, 2015-09-06:
No problem, I see that you are already dealing with the issue.

cheers

TerryE, 2015-09-06:
@eyaleb, I've posted a commit ef735bd36948146f557f608562f25d5ff37c8be8 to my branch luac.cross.1 fixing this issue.  I will give it a day or so then for other feedback then I will raise the pull request

EnochTheWise, 2017-04-12:
i'm build luac.cross from current master branch and test it:
```
$ ./luac.cross -l -o /dev/null -
print(0xabababab)

main <stdin:0,0> (4 instructions, 16 bytes at 0x55d4c3212150)
0+ params, 2 slots, 0 upvalues, 0 locals, 2 constants, 0 functions
        1       [1]     GETGLOBAL       0 -1    ; print
        2       [1]     LOADK           1 -2    ; -1414812757
        3       [1]     CALL            0 2 1
        4       [1]     RETURN          0 1
```

Bug is still not fixed?

TerryE, 2017-04-19:
I've assigned this one to me.

TerryE, 2018-06-23:
Now dealt with in LFS patch

closed 2018-06-23 by TerryE