nodemcu/nodemcu-firmware #633: dht_lib.lua bug?

neophox, 2015-09-03:
I was reviewing dht support in nodemcu and i think there may be a bug in dht_lib.lua when decoding DHT temperature data (L106). Shouldn't it be byte_3 instead of byte_2? 
It seems to me that integral part gets overwritten by decimal part in that line.

vowstar, 2015-09-03:
Please test this code:
https://github.com/nodemcu/nodemcu-firmware/wiki/nodemcu_api_en#dhtread
```
pin = 5
status,temp,humi,temp_decimial,humi_decimial = dht.read(pin)
if( status == dht.OK ) then
  -- Integer firmware using this example
  print(     
    string.format(
      "DHT Temperature:%d.%03d;Humidity:%d.%03d\r\n",
      math.floor(temp),
      temp_decimial,
      math.floor(humi),
      humi_decimial
    )
  )
  -- Float firmware using this example
  print("DHT Temperature:"..temp..";".."Humidity:"..humi)
elseif( status == dht.ERROR_CHECKSUM ) then
  print( "DHT Checksum error." );
elseif( status == dht.ERROR_TIMEOUT ) then
  print( "DHT Time out." );
end
```

neophox, 2015-09-03:
@vowstar I'm having problems reflashing firmware in my esp8266-01, will try once reflashed.

Don't you think there is a typo in the [code](https://github.com/nodemcu/nodemcu-firmware/blob/master/lua_modules/dht_lib/dht_lib.lua#L106)? I mean it seems Byte[3] is being assigned to the wrong var, overwriting previous value (with 0 in dht11 case as there is no decimal part):

```lua
...

for i = 1, 8, 1 do -- Byte[2]
    if (bitStream[i+16] > 3) then
      byte_2 = byte_2 + 2 ^ (8 - i)     <-- THIS IS RIGHT
    end
  end

  for i = 1, 8, 1 do -- Byte[3]
    if (bitStream[i+24] > 3) then
      byte_2 = byte_2 + 2 ^ (8 - i)     <-- THIS SHOULD BE: byte_3 = byte_3 + 2 ^ (8 - i)  ??
    end
  end

...
```

I can't test it because my esp is stuck at boot and need reflashing, will try once recovered.

neophox, 2015-09-06:
@vowstar the code you posted works fine. But it's using internal dht module, not the lua version (which does have the bug). Am i wrong?

marcelstoer, 2015-09-06:
> it's using internal dht module, not the lua version

IMO you _should_ use the internal DHT module if at all possible. I'd even suggest to drop the Lua DHT lib and it's accompanying examples as we have an internal module now.

neophox, 2015-09-06:
I agree to drop the Lua DHT lib. I've never had success using it anyway :(
Internal module works fine for me (ESP-01 + DHT11).

marcelstoer, 2015-12-12:
Addressed with #789.

closed 2015-12-12 by marcelstoer