nodemcu/nodemcu-firmware #634: Implementation of printf

TerryE, 2015-09-05:
`app/libc/c_stdio.h` defines **c_printf()** as follows:

```C
#define c_printf(...) do {                                        \
        unsigned char __print_buf[BUFSIZ];          \
        c_sprintf(__print_buf, __VA_ARGS__);    \
        c_puts(__print_buf);                                    \
} while(0)
```

OK, it's only used a few dozen times in a non-debug build but inlining this with a 1K allocation on the stack for any routine which uses it is not the right thing to do.  If you are doing debug build then there are frequent NODE_DBG and NODE_ERR calls which use this macro to generate the debug print lines.  

Needs replacing by a memory efficient out of line routine.

jmattsson, 2015-09-14:
I concur this is quite sub-optimal as-is (just ask a colleague of mine who got bitten by out-of-stack issues while trying to debug another issue by c_printf'ing lots).

I *thought* the issue was that ets_printf might be disabled via system_set_os_print(0) and such. The overall print/debug stuff could do with a thorough tidying up by now I think. It's quite confusing when you're starting out developing to work out just how to reliably get some stuff out the serial port. I'm not sure what the proper/best approach here would be though.

TerryE, 2015-09-14:
> It's quite confusing when you're starting out developing to work out just how to reliably get some stuff out the serial port.
As we've seen elsewhere, it is very easy for developers new to ELF development to treat their coding as if they were developing on a \*nix terminal session and litter their code with printf statements.   Better they use something like the `NODE_DBG()` macro with `DEVELOP_VERSION` defined on a per module basis.  

Whatever the wrapper approach such diagnostics should have zero or near-zero impact on the two RAM-based segments, and as little as possible impact on Flash usage.  I don't think that we can do better than **ets_printf** as discussed above.

TerryE, 2015-12-22:
We've converted this one in the 1.4 migration discussions. The SDK 1.4 + macros largely tidy this up, but the ETS implementation doesn't support floating point formats and numbers.

closed 2015-12-22 by TerryE