nodemcu/nodemcu-firmware #637: Bug in c_strtod()

TerryE, 2015-09-06:
If you type the following on a normal (PC or ARM)) Lua, observe:

```Lua
> print (0xffffff*0x10000,0xffffff0000)
1099511562240	1099511562240
```

This is because Lua holds all numbers as doubles and hence integers up to 2^52 can be stored as exact values with a zero exponent.  You would expect the same with nodeMCU on a non-integer build since the RTS implements full ANSI double precision in software.  However what you actually get is:

```Lua
> print (0xffffff*0x10000,0xffffff0000)
1099511562240	-1
```

I've tracked this down ot a bug in the nodeMCU routine `c_strtod()`.  This routine:

*    returns the correct value up to `0x7fffffff`
*    returns the signed 32-bit 2-complement value up to `0xffffffff` that in **n-2^32**
*    returns -1 for higher values.

I am not sure whether this is just a bug or a deliberate "design feature" of the nodemcu firmware.  Certainly fixing it might impact existing programs which rely on this.

So should we fix it or accept it?   Either way I need the behaviour of lua.cross to be consistent (see #632 .)

eyaleb, 2015-09-06:
As I commented in a related thread, I think that fixing `luac.cross` is the way to go. It is new (to nodemcu) so all bets are off right now. But it is already in `master` so this should be the first step.

Only then should the (separate) question be debated: change nodemcu `c_strtod` (and cross at the same time) to accept large integers? I wonder how many people will actually notice such a change. This debate can be allowed more time.

Your call (the audience is rather quiet so far) - I am now avoiding the large value that failed my program :-)

cheers

TerryE, 2015-09-06:
OK, I've stepped through this on x86-32bit and also looked the firmware code.  the issues is that hex constants are handled in `lobject.c:luaO_str2d()`

```C
  //...
  *result = lua_str2number(s, &endptr);
  if (endptr == s) return 0;  /* conversion failed */
  if (*endptr == 'x' || *endptr == 'X')  /* maybe an hexadecimal constant? */
    *result = cast_num(c_strtoul(s, &endptr, 16));
  //...
```
`lua_str2number` is just a synonym for `c_strtod`.  The standard **strtod()** parses hex constants correctly up to double size.  **c_strtod** doesn't so the firmware code take the fallback path of using c_strtoul() which sets the value to ULONG_MAX (-1). 

My instinct is to leave this limitation and replicate it in `luac.cross` as @eyaleb suggests, probably just by calling **c_strtod**.

TerryE, 2015-12-12:
Done some time ago.

closed 2015-12-12 by TerryE