nodemcu/nodemcu-firmware #639: Wifi loses IP if kept busy with UPD/ICMP

mdillmann, 2015-09-07:
Hi,
my ESP-01 running the build NodeMCU 0.9.6 build 20150704  powered by Lua 5.1.4 loses AP association after receiving about 40-50 pakets.

If this happens, wifi.sta.status() still shows 5, but wifi.sta.getip() returns nil, ping requests from outside fail.

Steps to reproduce:
1. configure station mode once to access wifi
2. get and note IP
3. run a continous ping (preferably from MAC or LInux) against that IP
4. node.restart() freshly, ensure no init.lua

watch the ping and about 40-45 pakets later it will not respond anymore

Addtional notes: I am implementing a tftp server that receives tftp-write requests. The issue there is the same: tftp starts resending packets after about 40-45 UDP packets sent succesfully. To track that down I tested with the ping described above (which uses ICMP)

Thanks,
-Markus

mdillmann, 2015-09-07:
My access-point is telling me = daemon.info hostapd: wlan0: STA xx:xx:xx:xx:xx:xx IEEE 802.11: disconnected due to excessive missing ACKs
So it looks like it does not send ACKs when kept busy and after being thrown out no further communication is possible unless a wifi.sta.connect() is done again

TerryE, 2015-09-07:
@mdillmann, I think that this issue is in the grey area where what your are observing is down to characteristics of the SDK rather than the nodemcu firmware itself.  We're rather behind the SDK version so (0.96 vs 1.3) so I am not sure if rebaselining the SDK to current will address this.   Sorry that I can't be more definite, but there are quite a few TCP stack issues that a move from 0.96 to 1.3 will cure.

mdillmann, 2015-09-07:
Hi Terry, 
I browsed briefly thru the changlogs of the newer SDK's and found some changes that could affect/cure that bug.
Anyhow: if anyone else faces this issue and this is with an hostapd-based accesspoint: there is an option in hostapd.conf that you can set that prevents clients be thrown out that do not send proper acks: disassoc_low_ack 0
That worked around the problem here (openwrt-based AP with hostapd)

Thanks,
-Markus

TerryE, 2015-09-07:
Yup, there's a lot of bigfixes in the changelogs that we're missing :)

TheoHarbers, 2015-09-22:
Had the same problem between ESP and openwrt (Carambola RT3055). The solution was to add 'option disassoc_low_ack 0' to /etc/config/wireless in the config wifi-iface section.

<pre>config wifi-iface
        option device   radio0
        option network  lan
        option mode     ap
        option ssid     'ioThings'
        option key      'secret'
        option encryption 'psk-mixed'
        option disassoc_low_ack 0
</pre>

marcelstoer, 2016-08-07:
Feel free to reopen if you still see those symptoms with a recent 1.5.x firmware.

closed 2016-08-07 by marcelstoer