nodemcu/nodemcu-firmware #641: MQTT Client: Last Will and Testament RETAIN flag ignored

matteovigni, 2015-09-08:
I noticed that even if I set the retain flag to 1 while setupping the last will and testament it is still sent with retain flag to 0.
I quickly checked the source and I think that the problem is that the last will and testament function in app/modules/mqtt.c ignores the retain parameter (and maybe the qos too). It's not very clear to me where it should be used so I can't propose a patch, but I think that it should be quite easy for someone who knows how the MQTT module works.

karrots, 2015-11-06:
Should be fixed by pr #718 & #697. Please verify against the `dev` branch using the build service.

devsaurus, 2016-01-28:
Confirm that the retain flag for `client:lwt()` works with current `dev`.

closed 2016-01-28 by devsaurus