nodemcu/nodemcu-firmware #642: wifi.ap.getconfig patch

TerryE, 2015-09-09:
See my comment in 663158e4ab090f68bc1251801764501fba7b1d62

A one-line fix.

TerryE, 2015-09-09:
I've just run the following on master so I think that this is the only case of this.

```Lua
-- file list-entries.lua
local mods, list = {'adc','bit','coap','dht','gpio','net','ow','rc','tmr','uart',
   'ws2801', 'cjson','crypto','file','i2c','mqtt','node','pwm','spi','u8g','wifi','ws2812'}, {}

for i = 1, #mods do 
  if _G[mods[i]] then list[#list+1] = {mods[i], _G[mods[i]]} end
end

tmr.alarm(0,500,1, function()
  local name, t = unpack(table.remove(list,1) or {})
  if not name then tmr.stop(0) return end
  collectgarbage()
  print(node.heap())
  print ("=== " .. name .. " ===")
  for k, v in pairs(t) do
    print (k,v)
    if (type(v) == 'table' or type(v) == 'romtable' and k ~= '__metatable') then
      table.insert(list, {name.."."..k, v})
    end
  end
end)
```

jmattsson, 2015-09-11:
The getconfig-in-the-wrong-table was addressed in 6ad266618232f84e5dcf287303803b195aedee32 in dev. Is there still an issue?

TerryE, 2015-09-11:
Jonny, sorry but I hadn't picked this one up. This goes back to the policy question of what should go into what branches. Straight bug fixes like this should go into master as well as any dev branches, in my opinion so that people forking off master don't rediscover them. 

At the moment, master is too unstable to be usable IMO.

jmattsson, 2015-12-16:
Master has been thoroughly updated.

closed 2015-12-16 by jmattsson