nodemcu/nodemcu-firmware #645: wifi.ap channel limited to 1-13?

eku, 2015-09-11:
The NodeMCU API documentation of wifi.ap.config() says that the channel can be in the range of 1 to 13. 
That means channel 14 for Japan is not supported. Is this a software limitation in NodeMCU or a hardware limitation of the ESP8266?

jmattsson, 2015-09-14:
Looking at the code, I only see the NodeMCU wifi module rejecting channels >14, so I suspect the documentation may be wrong here. A quick google suggests that the ESP8266 is capable of 1-14, so I'd say give it a go! If it works, we'll get the documentation updated

marcelstoer, 2015-12-12:
@eku What's your conclusion?

eku, 2015-12-19:
Sorry that I did not report. The module also works on channel 14, We should correct the NodeMCU API documentation. BTW I had trouble with configuring the WEP key. It must be supplied in HEX. ASCII is not rejected by the API but NodeMCU cannot attach to the AP:

marcelstoer, 2015-12-19:
I updated the public wiki accordingly: https://github.com/nodemcu/nodemcu-firmware/wiki/nodemcu_api_en#wifiapconfig

> ...NodeMCU cannot attach to the AP

This irritates me. If you use `wifi.ap.config()` I must assume your NodeMCU *is* the AP, no? Are you talking about the `pwd` parameter for `wifi.ap.config()`? Please add any hints directly to the wiki.

closed 2015-12-19 by marcelstoer

wordsforthewise, 2016-02-02:
@eku Is there an easy way to convert the password to hex?  I'm not finding it so far.