nodemcu/nodemcu-firmware #648: Layering nodemcu properly on the Espressif SDK

TerryE, 2015-09-14:
This issue relates to a point that I raised in #625.  I've just rebuild my local dev120 branch to make over the Espressif SDK 1.3.  One real pain with the nodemcu build setup is that this forks and makes minor changes to the standard SDK for no clear reason that I can see. This causes unnecessary regression problems when upgrading SDKs when upgrading SDKs as you have to meld the two sets of updates to all include files.  Surely it would be a lot simpler if the nodemcu build was properly layered on top of the SDK -- that is there where no changes to any SDK files and any additions were in nodemcu-specific files.

At a minimum if proves essential to vary the Espressif-supplied files then this should be done with properly commented and delineated change blocks to simply any melding need on upgrade. 

I'll have a look at the feasibility of doing this in my tests.

Aeprox, 2015-09-14:
Hey TerryE

First off, let me thank you for your awesome work and effort you put into the project! I fully agree that this is a step that has to be taken to make future SDK updates easier to manage. I gave porting to 1.3.0 a shot myself, but I'm a freshly graduated programmer with very little experience with makefiles and linker scripts, so I gave up after resolving a few compiler errors to do more productive things.

Would you mind pushing your test branch to github, so that other interested folk (such as myself) could follow on your progress (and possibly learn a thing or two)? thanks!

TerryE, 2015-09-14:
Sure, but let me get it working first! :wink:   I want to see if I can create a little bit more headroom in irom0 so that others can at least try adding a few or the core extensions.  When I get to a sensible checkpoint, I'll push it to my copy rep and post some notes on constraints and issues, etc.

rxf992, 2015-09-15:
days ago I  ported dev130 myself but not really tested yet.
Just put my code link here hope there is something useful.
https://github.com/rxf992/nodemcu-firmware/tree/dev130

3dot3, 2015-09-15:
Hi rx992, I tried your 1.3 and I can't get it compiled by esp open sdk. What is your building environment?

rxf992, 2015-09-15:
@3dot3 I use Ai-thinker's cygwin,which can compiled espressif 's sdk successfully.
I remember that I have fixed compiling error, but linking may not.
The biggest issue is "can not fit in irom0.text  "

rxf992, 2015-09-15:
@3dot3 I suggest you just compare my code with original sdk 1.3 and nodemcu for reference.
And if you want more help, you could leave your error msg here.
I am busy developing with original sdk c APIs since using lua cost too much RAM for my project and I need sdk1.3.

TerryE, 2015-09-15:
@rxf992.  Thanks. I'll compare your patches with mine to do a compare and contrast.  //Terry

3dot3, 2015-09-17:
@rxf992 , Thanks, I have been using Ai-thinker's ide2.0 successfully build your code. After I write them into flash, it not start to run. Have you been try to get it up and running on your esp?

jmattsson, 2015-11-10:
Hey @TerryE, can/should we close this one now?

TerryE, 2015-11-10:
Bugger, having done #745, it struck me that the first change I'd like to propose to dev is to remove all of the SDK bits from this repository and use a suck process effectively cloning how the open-sdk does it. This is my next patch, so can we leave this open for now?

TerryE, 2016-01-20:
I just noticed this was still open, but we've now done this work.

closed 2016-01-20 by TerryE