nodemcu/nodemcu-firmware #651: Floating point rounding issue during exponential calculation

hazarkarabay, 2015-09-16:
This happens In float builds.

```
print(string.format("%.20f", 2^4), "\n")
> 16.00000000000000355271

print(string.format("%.20f", 2*2*2*2), "\n")
> 16.00000000000000000000	
```
which means
```
print(2^4 == 16)
> false
```
I don't know this is nodemcu or eLua or Lua related issue but this is causing unexpected behaviour especially using manual binary math functions like base64 encoding etc.

Wrapping a ```math.floor()``` around exponentials fixes it but I feel it needs more underlying solution. 
Any idea is appreciated.

TerryE, 2015-09-16:
Yup that's life with floating point.  The floating point routines use unnormalised R*8 which can hold integers without rounding up to 2^52, if I recall correctly, and simple operations such as add and multiply keep he unnormalised form.  the **^** operator calls an internal **pow** and I suspect that this returns a normalised representation of 16, and the low bit noise in the mantissa is as a result of the double to string conversion.

hazarkarabay, 2015-09-16:
Thank you for the answer. As you pointed, I see exponential function as form of simple math which in essence; multiplication. That's why I'm surprised to see ```2^4 ~= 16``` situation, because multiplication and division operations should be “safe” for floating point.

Is that internal pow call goes to Xtensa library or it is a standard C thing?

TerryE, 2015-09-17:
See [app/libc/c_math.c](/nodemcu/nodemcu-firmware/blob/master/app/libc/c_math.cc):pow()

hazarkarabay, 2015-09-19:
I see that algorithm based on ```a^b = exp(b ln(a))```, even for integer exponents.
Can I change that implementation to integer aware version, like e_pow.c from fdlibm? 

I tried but the compiler (or linker?) is not happy. ```../lib/libm.a(w_sqrt.o):(.literal+0x8): undefined reference to `__errno'
../lib/libm.a(w_sqrt.o): In function `sqrt':
....
collect2: error: ld returned 1 exit status
```
Probably I'm trying to do something absurd.

TerryE, 2015-09-20:
@hazarkarabay, what is your rationale for trying to do this? Compute time? accuracy?  This is an IoT chip, so an error of 1 in 2^51 is hardly important.  If you are that time critical then either select an integer build or use the explicit form.  I think that anyone involved in the firmware development would classify this as not worth fixing.

hazarkarabay, 2015-09-20:
Actually I just want a base64 decode function in Lua. None works of what I found. But all of that worked if I use a integer build. So I decided to find the reason and here we are.

Feel free to close this issue if it is not worth fixing.

marcelstoer, 2015-09-20:
> Actually I just want a base64 decode function

@jmattsson pointed out in https://github.com/nodemcu/nodemcu-firmware/issues/486#issuecomment-116376058 that he also thinks there should be a decode companion to [crypto.toBase64()](https://github.com/nodemcu/nodemcu-firmware/wiki/nodemcu_api_en#cryptotobase64) and I couldn't agree more.

TerryE, 2015-09-20:
@hazarkarabay, OK Hazar that someting that a lot easier to do.  The ROM already has a base64 encode and decode.  I'll have a look at how easier it is to add and give you the patch.

TerryE, 2015-12-12:
tBase64 superseded by #844

closed 2015-12-12 by TerryE