nodemcu/nodemcu-firmware #652: making Variables persistent after a restart

suraj022, 2015-09-16:
Hello,

I'm making a smart home system for myself. and i need  to store a retrieve data from the module.
I used the following function.

function save_settings(name,value)
    file.remove(name)
    file.open(name,"w+")
    file.writeline(value)
    file.close()
end

it works but its slow and the NodeMCU crashes if i trigger the above function rapidly... sometimes requiring a FS format. to be able to use it again.

So my question is.... is there any other way to make variables persistent between restarts?

eku, 2015-09-16:
Why don't you use the JSON modul?

suraj022, 2015-09-16:
any example on how to do that.. sry i have no idea what JSON is.

marcelstoer, 2015-09-16:
You should close this issue before one of the moderators has to. This is not a discussion forum but a place to report errors in the NodeMCU firmware. If you need help with your software either try http://www.esp8266.com/viewforum.php?f=17 or http://stackoverflow.com/tags/nodemcu. Good luck with your project.

closed 2015-09-16 by suraj022