nodemcu/nodemcu-firmware #653: wiif.startsmart() causes hard reset

rubberbird, 2015-09-17:
I have the below firmware

NodeMCU custom build by frightanic.com
	branch: master
	commit: bb9dd62882b89d40f5d0df2c009c972f64b7f965
	SSL: false
	modules: node,file,gpio,wifi,net,tmr,uart
 	built on: 2015-09-15 06:56
  powered by Lua 5.1.4

Using the ESP App v2.1

When my login details are found, the device hard resets.

marcelstoer, 2016-04-10:
Feel free to re-open if you're still seeing this with the latest `dev` branch. Note that smartconfig is now turned OFF by default.

closed 2016-04-10 by marcelstoer

Anudeep09, 2016-06-03:
How to enable samartconfig in firwware ..? i am using bin files to flash fimware

marcelstoer, 2016-06-03:
See http://nodemcu.readthedocs.io/en/dev/en/support/ and http://nodemcu.readthedocs.io/en/dev/en/modules/wifi/#wifistartsmart.