nodemcu/nodemcu-firmware #654: script crashes......unless I use the terminal

blowback, 2015-09-17:
On 0.9.6-dev_20150704, I have an init.lua that waits 5 seconds and starts main.lc, which fetches temperature from a BMP085 and uploads it thingspeak on a timer.

This will consistently crash when the main timer goes off, causing a panic and reboot.

....unless I type "print(1)" on the terminal, after init.lua has run, and before main.lc kicks in.

Utterly baffled....

blowback, 2015-09-17:
It's the "print" that's significant, not the "1". Other calls like adc.readvdd33() don't have the same crash-preventing magic.

Aeprox, 2015-09-17:
I've had issues simular to this one, where interecting with the module over serial prevents it from crashing. Hard to find the exact cause.

blowback, 2015-09-19:
Hmm judicious use of ` node.input([[print("1")]]) ` in the startup timer seems to fix it.

abhishek-kakkar, 2015-11-01:
+1. I spent hours debugging my MQTT based code to ultimately find this issue here. Unless I executed any command on the serial shell, the ESP8266 would reboot without any warning at almost the same time instants. Using node.input as above did fix the issue but it would be great if this was looked into and the root cause of this found out.

karrots, 2015-11-02:
You may want to try a build from the *dev* branch. Using the build server http://frightanic.com/nodemcu-custom-build/

TerryE, 2015-12-12:
See #719

closed 2015-12-12 by TerryE