nodemcu/nodemcu-firmware #656: Licensing statements in nodeMCU firmware

TerryE, 2015-09-18:
This issue arose following an issue in a fork of nodeMCU pvvx#2 where the repository owner has removed the nodeMCU copyright and substituted his own [pvvx/EspLua License](/pvvx/EspLua/blob/master/LICENSE) and where the nodeMCU user needs to consider whether to issue a DMCA notice to GitHub.  However this issue relates to the treatment of licensing within this repository.  Unlike @pvvx,as far as I can see @nodemcu hasn't removed copyright statements within the source materials but it has replaced earlier copyright overviews in the Lua and eLua sources.  I suggest that the content of the eLua LICENSE either be appended to LICENSE or included as a separate LICENSE.elua

jmattsson, 2015-09-24:
+1 on both accounts. I'd suggest using LICENSE.elua, as that makes it trivial if someone who for whatever reason wants to fork off just the eLua stuff.

@nodemcu, here's a helpful link: [how-to submit a takedown notice](https://help.github.com/articles/guide-to-submitting-a-dmca-takedown-notice/). @TerryE, I seem to recall I saw you already point out the transgression to the EspLua guys, yeah? So the "ask nicely" has already been done.

TerryE, 2015-09-24:
Yup, you can track the pvvx issue using the above link.  But since as far as I can see this is a removal of nodeMCU copyright, I think that we should leave it to them to take this further if they wish.

marcelstoer, 2016-04-10:
Just to get this right... Would adding https://raw.githubusercontent.com/elua/elua/master/LICENSE as `LICENSE.elua` to the root of the repository be enough? That'd be an easy fix.
The fact that eLua is [MIT just as NodeMCU](https://github.com/nodemcu/nodemcu-firmware/tree/dev#license) is handy.

TerryE, 2016-04-11:
Hummn, this lists quite a few libraries that we don't use. But a cut down version would be a good start and we could close this.

TerryE, 2018-06-22:
After 3 years of inactivity, I think that this one is really dead.  We can reopen it if there are any complaints + resource to remedy the requested actions.

closed 2018-06-22 by TerryE