nodemcu/nodemcu-firmware #657: wifi.sta.getap doesn't work inside tmr.alarm

alexandrevicenzi, 2015-09-19:
Hi,

I'm trying to list all access points every X seconds, but if I put `wifi.sta.getap` inside `tmr.alarm` is doesn't work.

Here's the code:

```lua
wifi.setmode(wifi.STATION)
wifi.sta.autoconnect(1)

wifi.sta.config("XXX", "YYY")

function scan_ap(t)
    print("Scan....")
    for k, v in pairs(t) do
      print(k..","..v)
    end
end

tmr.alarm(1, 2000, 1, function ()
    tmr.wdclr()
    wifi.sta.getap(1, scan_ap)
end)
```

It prints "Scan...", but the table `t` is empty. If I call `wifi.sta.getap(1, scan_ap)` outside the alarm it works fine.

Any ideas?

I'm using dev096 custom build with all modules.

TerryE, 2015-09-19:
Yup understand how Lua works on the esp8266.  Please read my [Unofficial FAQ](http://www.esp8266.com/wiki/doku.php?id=nodemcu-unofficial-faq) where I explain all this.   In this case you need to allow the scan to complete before overwriting it with another request.  So stretch the timer to 15 secs or make a one shot and requeue at the end of the scan_ap.

closed 2015-09-19 by TerryE

alexandrevicenzi, 2015-09-19:
Hi @TerryE, I was reading the article :smile:, but doesn't work in the first time too.

alexandrevicenzi, 2015-09-19:
It worked :smiley:

TerryE, 2015-09-19:
Good.  :smiley: