nodemcu/nodemcu-firmware #662: Hey guys, NodeMCU wants you, we need more administrators to manage this project.

MarsTechHAN, 2015-09-23:
<h2>#As the title said, the present managers and founders are too busy to hold all the works of NodeMCU.</h4>
<br>
<h4>So if you have time, ability, and really enjoy the NodeMCU, you can send a mail to my mail box, and we will pick the people we think fit this project to become</h2>
<br>
<h1>>>>>>>martin.zgsz@gmail.com<<<<<<<</h1>

TerryE, 2015-09-23:
Martin, I will put up my hand to offer to help here.  I've worked on various FLOSS projects, for example I was an admin for and largely set up the OpenOffice.org user forums; I was a contributor to the OOo Basic runtime.  I help set up and was a moderator for the VirtualBox user forums as well as being a contributor.  I woprk with the PHP dev team and contributed to the PHP runtime and especially their OPcache extension.  So I feel that I have some reasonable credentials.

However to be workable, we do need an active participants with *contributor* rights and at least one with *adminstration* rights on the repository.  I will write to you separately.  

One things that I would like to establish is whether is whether 

*    you guys will want to take back control of the repository in the future or 
*    the remaining contributors need to put their own roadmap and governance in place.

grahamehorner, 2015-09-23:
also happy to help in anyway I can

jmattsson, 2015-09-24:
Pinging @marcelstoer in addition to @TerryE - make sure you use the email address listed to volunteer!

I'm not having much time at the moment to spend on NodeMCU, but I sure like it :)

TerryE, 2015-12-12:
We've addressed this issue, now, so I am closing this.

closed 2015-12-12 by TerryE