nodemcu/nodemcu-firmware #666: How to improve NodeMCU development board?

GerwinJanssen, 2015-09-24:
Did you all know that we working on new improved ‪#‎ESP8266‬ development board? But we need your help. How can we improve our development board? Let us know!
[image]

ccxx72, 2015-09-24:
What about a divider on RX pin with a jumper to bypass it ?

udif, 2015-09-25:
How about adding 2 jumpers, one to disconnect the ESP8266 Tx and one to disconnect ESP8266 3v3 ?
This can be used to program ESP8266-01 modules using NodeMCU with nothing but a wire harness, and saves us from building this:
http://www.whatimade.today/content/images/2015/04/ESP8266_FTDI_RST_bb.jpg

If you don't want extra through hole stuff, just place 2 pairs of holes and have them jumpered by default, so anyone who wants this feature can cut the trace and install 2 jumpers.

eyaleb, 2015-09-25:
Improve power usage. Read on if deployment is being considered.

I use the board for development and find it unsuitable for deployment (on battery) as it uses too much power (over 400uA in deep sleep; I still use v0.9).

- I now connect a battery to the 5v pin (or through an LDO to the 3.3v pin). My bare setup (ESP+battery+LDO+ds18b20) uses an MCP1700 which has an Iq of 1.5uA and the whole thing draws 34uA then. If I needed more power I would probably use an RT9020.

- A way to disconnect the front-end.

- A jumper to turn off all LEDs (I now physically remove them).

emgab, 2015-09-26:
34uA - Wow - great achievement - would you mind to elaborate on how you achieved this with an ESP - i thought it would not be possible to run it on Batteries for a longer time.
Thanks for your insight and help in advance,
kind regards, mike

eyaleb, 2015-09-26:
@emgab , as I mentioned, I use a MCP1700 which has a 1.5uA Iq (quiescent current) and the ESP+ds18b20 use about 30uA. <b>Let me emphasize that this is the deep sleep current</b>. When running (with WiFi) the ESP consumes around 70mA, with much higher peaks for very short periods when communicating.

BTW, the more powerful RT9020 (500mA vs. 250mA) still has a typical Iq of 25uA, not that bad.
The deep sleep current matters when you wake up for only a few seconds every so many minutes. 

With my app the ESP wakes up once a minute and uses about 75mA*s (in less than 1s it takes to read and report the reading) so 1.25mAh/h. A bad LDO with, say 1mA Iq (I've seen worse), will consume another 1mAh/h, a significant load. If the app wakes up once every 5 minutes then the LDO adds 5mAh/h to the 1.25mAh/h - the bulk of the power is spent <b>sleeping</b>.

With a good LDO the app will still uses 1.25mAh/h working but the LDO will add only 0.0015mAh/h doing a 1min cycle and 0.0075mAh/h in a 5min cycle - the bulk of the power is spent when <b>awake</b> (doing work).

So the longer I sleep the more important the sleep power usage becomes. If the ESP is constantly on then the quality of the LDO is less critical. If the app wakes up once an hour (not uncommon) then most of the power is spent (wasted) when sleeping (doing no work).

cheers

fabianhu, 2015-09-27:
Hi, 
for the 3.3V supply, had some efficiency problems from 12V to 3.3 and therefore I built: [this](https://github.com/fabianhu/LM1117switcher).
It fits onto a LM1117-3.3 SOT-223 Footprint.
(I assume this is compatible to the LD33 shown at your board)
and works great from 12V, but may have some efficiency downsides during sleep compared to a LDO.
If you like to integrate a step-down converter onto your design, just let me know, if I can help.

greg-ware, 2015-09-28:
Hello all,
I've been thinking at this, i.e "how to design the ultimate Node-MCU board". My ideas have been:
  * 1) keep it hand-solderable. This means minimum package should be SOP.
     * 1a) This kind of implies keeping the 340CHG as USB/TTL as it has less pins and comes in SOP.
     * 1b) whenever possible, have both SMT and through-hole for LED, R,C components.
  * 2) make 'northbridge' decoupleable, i.e. leave the possibility of switching off USB/TTL and power (to run on 3.7V lithium alone):
    * 2a) one way to do this is to make power and USB/TTL stages optional, and replaceable/swappable. For this we can introduce a power+USB 'shield' that would piggy-back on top of the current location of the northbridge components (i.e. the stuff that's between the ESP and the USB socket). There would be two rows of pins (with a convenient wide-DIP pitch). I think we need 12-16 pins in all, to add a power+USB/TTL shield:
      * From USB socket: GND, +5V, D+/D-, reset and flash switches -> 6 pins
      * To ESP: +3.3V, Rx, TX, GPIO1, GPIO2, RESET -> 6 pins
      * optionally, bring out ESP's CH_PD, and GPIO 16 (wake-up trigger) -> 2 pins + GPIO15&ADC, 2 more pins.
    * 2b) Arrange the shield pinout so that it can be partially populated, e.g. group power mgmt pins together, so that a 'narrow' shield can be use when e.g. only power is needed.
    * 2c) Keep option to use same PCB for no-shield implementations. e.g. leave the tracks for 'default' implementation (xxx1117+(CH340G or CP2102)) under the shield (between the 2x8 W-DIP rows), and make the shield optional. Those who want the normal one would purchase a populated board (w/ xx1117 +CH340G but no power shield DIP headers), those who want optional would purchase a board with power shield headers and no power+USB/TTL components, and plug their own shield.
    * 2d) There are many advantages I can see to that design, only drawback is more complexity, i.e. that's two boards to ship with the piggy-back power-USB/TTL northbridge shield. But then, advantages are huge, you could have only a few northbridges and reuse them, and switching from programming use of a given node-MCU (xx117 reg+USB/TTL) to production use (3.7V Bat or high efficiency buck and no USB/TTL) would be a snap (LEDS would have to be on that shield, too). As technology evolves, the power stage could keep up and use new chips too without having to replace the whole ESP board.
  * 3) This is more a production issue, but node-MCU vendors should NOT have male pins soldered under the board by default, but provide a option for male or female headers, so that Node-MCU can be used a main board with shield plugged on top. This also means the headers soldering pads should be available on either side of the board (cause male is soldered on top, while female on the bottom)

greg-ware, 2015-09-28:
BTW, please do not change the form factor or pinout!
Current Node-MCU is becoming a de-facto standard, which is used by other board (rovers, ...) to integrate.
Changing this will fragment the market and drive costs higher, which would negate one of Node-MCU's and ESP8266's main advantage, which is ultra low cost.
[On this point, I'm at a loss as to why they came up with ESP-13 (AKA Wroom-02) yet-unseen and incompatible for factor???]

If you do have to change form factor, make it narrower but keep old pins aligned with new ones, i.e. same pin sequence

robertfoss, 2015-10-25:
Use a step up/down converter so that the board supports the full 3.0V-4.2V range of a LiPo.

TerryE, 2015-12-12:
This conversation really belongs as part of the [nodemcu-devkit-v1.0 issues](https://github.com/nodemcu/nodemcu-devkit-v1.0/issues) list, e.g. [Suggestions about new NodeMCU devkit based on ESP32](/nodemcu/nodemcu-devkit-v1.0/issues/12), hence falls under #719.

closed 2015-12-12 by TerryE