nodemcu/nodemcu-firmware #671: Document sntp module

karrots, 2015-09-27:
```sntp``` module not documented in wiki.

Syntax to my untrained eyes looks to be:
```sntp.sync(server, sucesscallback(unixtimestamp), errorcallback(string))```

jmattsson, 2015-09-28:
I documented it [in our local repo's wiki](https://github.com/DiUS/nodemcu-firmware/wiki/sntp-module-documentation) before it all got merged into NodeMCU. If you have the time, could you copy it over to the dev096 NodeMCU wiki page? Same goes for the RTC modules. Thanks!

karrots, 2015-09-28:
One note about the documentation mentioned. It looks like the word anycast is used where multicast should be used.

jmattsson, 2015-09-29:
@karrots That is fully intentional (see [RFC 2030](https://www.ietf.org/rfc/rfc2030.txt) for details). Anycast != multicast.

karrots, 2015-09-29:
ok I see multicast server discovery instead of the multicast service delivery I was thinking. Gets a bit confusing when there are two ways to use multicast.

closed 2015-10-24 by karrots