nodemcu/nodemcu-firmware #674: HSPI frequency

rickvanbodegraven, 2015-09-30:
I can't seem to find any way to set the frequency for the HSPI module. I've fixed it for my own project by dividing the frequency all the way down to 2MHz (my Max6675 only works up to 4.3MHz, which is too low for the default 10MHz speed of the HSPI module).

Could someone with more knowledge of the underlying driver code build in a function (or parameter) that allows users to set the communication speed from Lua code?

Serasidis, 2015-10-01:
The SPI frequency setting would be a good addition because many SPI slave devices do not work at 10 MHz (it's too high frequency).

devsaurus, 2015-10-01:
I was missing HSPI frequency control for the u8g and ucg modules as well since some displays might be restricted to <10 MHz.
Interestingly, the Lua API already supports the clock paramter for `spi.setup()`. But it's not used in the driver code with spi_master_init(). If someone could point me to a good documentation of the SPI registers, I would look into this. As well as supporting variable bit length...

Aeprox, 2015-10-02:
@devsaurus I think [this link](http://www.esp8266.com/viewtopic.php?f=13&t=2367) should be able to get you started? Seems like metalphreak already figured out most of it. I don't own a scope so can't implement these features.

devsaurus, 2015-10-04:
@Aeprox Thanks for the link! I referred to the information in this blog post http://d.av.id.au/blog/hardware-spi-clock-registers/ which is probably from the very same author.
@Serasidis Feel free to check whether #678 (against `dev`) works as expected for your hardware configuration.

Serasidis, 2015-10-05:
@devsaurus  First of all, thank you !
I saw in the nodemcu-firmware  bin folder that the bin firmware was updated 9 months ago. Could you update the bin firmware file or send me the new bin file to update my ESP8266 ?

devsaurus, 2015-10-05:
You'll find a temporary release from my fork here:
https://github.com/devsaurus/nodemcu-firmware/releases/tag/0.9.6-dev-hspi_clockdiv_20151005

Serasidis, 2015-10-05:
@devsaurus  IT WORKS !!! You are the best !!!
Thank you very much !

devsaurus, 2015-10-05:
Always glad to help :wink:
And thanks for giving this a try - it's very valuable to have a double checking on such fundamental things. I'll merge it soon once my other topic with HSPI is stable (variable databit length).

devsaurus, 2015-10-10:
I'm closing this issue for now, but stay tuned as there will be a major API update soon.
With the info from http://d.av.id.au/blog/hardware-spi-hspi-command-data-registers/ nodemcu will get much better HSPI support.

closed 2015-10-10 by devsaurus

pastukhov, 2015-10-15:
I'm geting a garbage on my ssd1306 with u8glib and
``spi.setup(1, spi.MASTER, spi.CPOL_LOW, spi.CPHA_LOW, spi.DATABITS_8, 8)``

devsaurus, 2015-10-15:
Thanks for the report. I've tested with ucglib back then - will give it a try with u8glib soon.

pastukhov, 2015-10-15:
Not sure that the problem is in this issue. I will make more tests tomorow.

devsaurus, 2015-10-15:
Just had a quick look - my ssd1306 executes u8g_graphics_test.lua flawlessly via SPI on sdk 1.4.0 fork.

commit 093a8959 (2015-10-06): Merge pull request #678 from devsaurus/dev-hspi_clockdiv

Add divider for arbitrary HSPI clock frequencies, resolves #674