nodemcu/nodemcu-firmware #682: dev build still having DNS Fail issues

UserXYZ, 2015-10-07:
I need to measure some temperature/humidity/etc stuff and send it over MQTT with time stamp.
Since there is no official support for NTP, nor are NTP servers they always available, I wanted to use simple HEAD HTTP requests to Google's (or any other always available) web servers and simply parse Date field in the response.
But, after some time, I start getting "DNS Retry" errors and after while, I'm hit with "DNS Fail" error.
When the first "DNS Retry" hits, free memory gets seriously decreased, and it never frees, so after some undetermined time, memory goes down to nil, error is reported about not enough memory  and the board reboots itself.

This has been observerd a long time ago here:
https://github.com/nodemcu/nodemcu-firmware/issues/234?_pjax=%23js-repo-pjax-container
and in examples/fragment.lua in every build (master, dev, dev096) there is the same piece of code to test for this error.

But, dev096 and master build pass this test without any problems, while dev build dies horribly.
I've built all of them on frightanic.com with the same setup, and have formatted the board.
Manually setting wifi and using this test with dev build gives zero, one, or two "DNS Retry" errors and the board resets itself - so it is not a problem of memory at the very beginning.

I also tried to compare net.c in dev and dev096, they are the same, and have a new, "enhanced" dns error reporting function (or whatever it is) unlike the origonal fix from march - so it' cant be that.
The only difference between dev096 and dev at this moment is in wifi.c, regarding some wifi setup functions - I couldn't find anything else.

Does anybody know what is going on here, since this bug is supposed to be fixed in march 2015?

BTW since I do not know how to install and use Espressif devkit 1.4.0 (which I downloaded) to build my own firmware, I am forced to use frightanic's online build system, but if somebody would like to point me to some detailer step-by-step howto build nodemcu devel system, I'd be very grateful.

pastukhov, 2015-10-07:
NTP and RTC clock is available in dev. Look at https://github.com/nodemcu/nodemcu-firmware/pull/545

UserXYZ, 2015-10-08:
Yes, I have seen that, but it's only available at dev branch and is not documented in any way (officially at least) and has no examples.
Plus, I'd have to build it on my own (having problems with that at the moment) because my preferred way, frightanic.com, does not offer to enable/disable all of the modules of each branch, so you always have the same modules...which kind of defeats the purpose of choosing dev build because you can not have all the new thing that are available in dev branch...

Finally, using ntp/rtc in dev does not solve the problem with dns leaking memory...it is still present in every branch, and replacing the functionality that I achieve with my solution (buggy as it is) with another does not solve the original problem...what if I was to send data to some mqtt broker, and I start getting dns fail and then memory low failures? 

So I'm begging somebody smarter to check the dns retry/fail code and try to fix the issue.

Cheers, Andrea

marcelstoer, 2015-10-08:
> rightanic.com, does not offer to enable/disable all of the modules of each branch, so you always have the same modules

True, but not for much longer. It's pretty high on my todo list.

UserXYZ, 2015-10-08:
Sweet, can't wait to get that.
I was actually thinking last night how it might be achieved, and I came with three-part process that  would enable you to change your site dynamically, based on what somebody wants to build (what branch that is)...

Since you get the source directly from github, the first part of the solution would be to get user_modules.h from selected branch, and parse it into a group of checkboxes so user can select the modules (like you do now, only the list would be directly related to what modules are available, and not some predefined set).
Then, next step would be selecting additional fonts etc and then finally, build.
In the build, you would use your dynamically generated user_modules.h and inject it into the source so the final firmware would have exactly what user wanted built.

Of course, warning about what modules to select and their dependancy should be presented to the user before triggering the build, so you'd be in the clean if something is not working right with the built firmware because the user chose wrong set of modules...

Does this sound silly/unachievable?

marcelstoer, 2016-02-26:
All of that has since been addressed:
- based on SDK 1.5.1 now (`dev` branch)
- docs at http://nodemcu.readthedocs.org/en/dev/ (all modules, building firmware, etc.)
- HTTP client module
- SNTP module

closed 2016-02-26 by marcelstoer