nodemcu/nodemcu-firmware #684: One wire search fails if is used in coroutine resumed by timer

sanyi, 2015-10-08:
I am using 3 DS18B20 wired to pin 3. I am trying to list this devices from a coroutine. The following code works:

```lua
function get_devices(pin)
     ow.reset_search(pin)
     repeat
          local addr = ow.search(pin)
          if(addr ~= nil) then
               print("device found")
          end   
          coroutine.yield(300)
     until (addr == nil)
     ow.reset_search(pin)
end

function main ()
     get_devices(3)
end

co = coroutine.create(main)
coroutine.resume(co)
coroutine.resume(co)
coroutine.resume(co)
coroutine.resume(co)
```

However if I want to trigger the coroutine resumes like this:

```lua
function resume (co)
	local _, delay = coroutine.resume(co)
	if coroutine.status(co) ~= 'dead' then
		tmr.alarm(0, delay, 0, function () resume(co) end)
	end
end
resume(coroutine.create(main))
```
the ow.search(pin) is correctly called but always returns nil.

What I am trying to achieve is to have a delay function, such that during the delay to give the oportunity to nodemcu firmware to do its job.

sanyi, 2015-10-10:
Something does not add up, right now does the same in the non timer variant. I have to check this more thoughtfully before bothering anyone else. I will close this for now. 

Cheers, Sanyi

closed 2015-10-10 by sanyi

TerryE, 2015-10-10:
@sanyl, look at the ds12b20.lua example.  `ow.search()` returns an 8 byte binary address as a character string. Just printing it as a varaible *will* produce gobbledegook.  Try something like
```Lua
local printable_addr = ''
for i = 1,8 do printable_addr = printable_addr .. (%02x-"):format(addr:byte(i)) end
```

sanyi, 2015-10-10:
@TerryE Thanks for the suggestion! 
I just started playing with NodeMCU from few days ago, pretty inexperienced with LUA too. 

I am working on a code to be able to read DS18B20 sensors. At high precision this sensors require quite long conversion time. I want to let the NodeMCU do it's thing during that wait time, while keeping the code simple.
Here is my attempt:

https://gist.github.com/sanyi/79df3f217c35e7526ff1

The code works. Well almost... To my dismay first attempt to list devices is not working. Every consecutive attempt works correctly. First I had the impression that is only happening when I am using coroutines. Then I understand that is always the same ever with the ds18b20 library packed in this repo. 
Once known about the problem is just a mater to call a dummy list address prior of doing something. However there is still a bug around, I am not sure if I should reopen this. Probably not.

sanyi, 2015-10-11:
@TerryE  My final conclusion with this is that the one wire search function fails from time to time, even on consecutive requests (NodeMCU 0.9.6 build 20150704  powered by Lua 5.1.4). Interestingly bigger the interval between searches increases the probability of failure. Any other operation works as should. So now I chose to work with fixed device addresses.

I had my first contraption built on NodeMCU, works happily for 24+ hours already. In the process I came up with some functions I packed into a library: https://github.com/sanyi/ndoemcu-ds18b20-gsa

Cheers, Sanyi