nodemcu/nodemcu-firmware #690: tmr.time() bug – value leaps backwards

smartynov, 2015-10-16:
I noticed that `tmr.time()` is producing very strange behaviour: when calling it consequently it sometimes returns smaller values than on previous call. At the same time `tmr.now()` returns increasing values.

Here is the minimal example:

```
tmr.alarm(1,100,1,function ()
  tt = tmr.time()
  tn = tmr.now()
  if not(tt == tp) then
    print("tt="..tt.." tn="..tn)
    tp=tt
  end
end)
```

And here is the output:

```
tt=18 tn=18099727
tt=19 tn=19199832
tt=20 tn=20199841
tt=21 tn=21199687
tt=22 tn=21899493
tt=23 tn=22999399
tt=24 tn=24099722
tt=25 tn=24999726
tt=26 tn=25999477
tt=27 tn=26899470
tt=28 tn=27899454
tt=27 tn=27999667
tt=28 tn=28099774
tt=29 tn=29199720
```

Notice how it goes well until `tt=27` seconds, then it increases to 28 and then falls back to 27. At the same time, `tn` keeps increasing!

Few more examples:
```
tt=47 tn=47099710
tt=48 tn=48099727
tt=49 tn=49199740
tt=50 tn=50199697
tt=51 tn=51299701
tt=52 tn=51599470
tt=51 tn=51799811
tt=52 tn=52199728
tt=53 tn=52599467
tt=54 tn=53599720
tt=53 tn=53799579
tt=54 tn=53899808
tt=55 tn=54599733
tt=54 tn=54899793
tt=55 tn=54999762
tt=56 tn=55699505
tt=55 tn=55899813
tt=56 tn=56099731
tt=57 tn=56799466
tt=56 tn=56899844
```

It is a clean test using `NodeMCU 0.9.6 build 20150704  powered by Lua 5.1.4`, code is run after a hard reset, nothing else is started in `init.lua`.

smartynov, 2015-10-16:
Ok, here is an example that's even smaller:
```
tmr.alarm(1,100,1,function ()
  tt = tmr.time()
  tn = tmr.now()
  print("tt="..tt.." tn="..tn)
end)
```
And here we go...

```
tt=98 tn=98661513
tt=98 tn=98761556
tt=98 tn=98861429
tt=99 tn=98961449
tt=99 tn=99061339
tt=99 tn=99161443
tt=100 tn=99261514
tt=99 tn=99361433
tt=99 tn=99461540
tt=99 tn=99561521
tt=99 tn=99661541
tt=99 tn=99761517
tt=99 tn=99861524
tt=100 tn=99961566
tt=100 tn=100061348
tt=100 tn=100161433
tt=100 tn=100261581
tt=100 tn=100361522
tt=100 tn=100461539
```
Why the hell it behaves like this?

TerryE, 2015-10-16:
If I recall, we've had  this issue reported before.  Do you have the same results with the proposed `dev140`?

smartynov, 2015-10-20:
@TerryE thank you for the idea, I should have tried it before creating an issue. 

Anyway, with `dev140` it is all the same – the problem persists.

I just flashed `dev140` with these parameters:
```
NodeMCU custom build by frightanic.com
	branch: dev140
	commit: 340ef8fc0ae00be1960acebe0276702dc7a00cb0
```
...run my tiny code sample, and I am getting same "jitter" again (line with `tn=23217488`):
```
tt=21 tn=22217471
tt=21 tn=22317474
tt=22 tn=22418319
tt=22 tn=22517154
tt=22 tn=22617217
tt=22 tn=22717177
tt=22 tn=22817224
tt=22 tn=22917490
tt=22 tn=23017217
tt=22 tn=23117474
tt=23 tn=23217488
tt=22 tn=23317210
tt=23 tn=23417475
tt=23 tn=23517488
tt=23 tn=23617219
tt=23 tn=23717240
tt=23 tn=23817474
tt=23 tn=23917475
tt=23 tn=24017473
tt=23 tn=24117472
tt=23 tn=24217484
tt=23 tn=24317211
tt=23 tn=24417471
tt=24 tn=24517476
tt=24 tn=24617474
tt=24 tn=24717214
```

The longer code runs, the wilder it gets (total mess here):

```
tt=247 tn=250113482
tt=247 tn=250213482
tt=247 tn=250313485
tt=249 tn=250413492
tt=250 tn=250513227
tt=249 tn=250613252
tt=250 tn=250713500
tt=249 tn=250813227
tt=249 tn=250913485
tt=248 tn=251013487
tt=248 tn=251113484
tt=248 tn=251213483
tt=248 tn=251313487
tt=248 tn=251413481
tt=248 tn=251513479
tt=248 tn=251613473
tt=250 tn=251713493
tt=249 tn=251813224
tt=249 tn=251913486
tt=249 tn=252013485
tt=249 tn=252113484
tt=250 tn=252213491
tt=251 tn=252313226
tt=252 tn=252413239
tt=251 tn=252513679
tt=251 tn=252613493
tt=252 tn=252713139
tt=253 tn=252813186
tt=253 tn=252913148
tt=251 tn=253013268
tt=251 tn=253113487
tt=251 tn=253213485
tt=251 tn=253313484
tt=251 tn=253413486
tt=250 tn=253513482
tt=251 tn=253613476
tt=250 tn=253713481
tt=253 tn=253813494
tt=253 tn=253913187
```

Does anyone have ideas how to fix it?

smartynov, 2015-10-21:
Completed a longer experiment, dumping `tmr.time()` value to a cloud every minute. Here are results:
[image]
As divergence increases over time, I can suppose there are some sort of correcting algorithm which is going wild.

Aeprox, 2015-10-21:
I've done some research into this, and I'm not sure if this is something we can fix. According to the SDK documentation, they do indeed use a correcting algoritm of sorts. 

> If system_get_rtc_time returns 10 (it means 10 RTC cycles), and
> system_rtc_clock_cali_proc returns 5.75 (means 5.75us per RTC cycle), then
> the real time is 10 x 5.75 = 57.5 us.

system_get_rtc_time is what tmr.now calls, tmr.time is calculated using the value returned from system_rtc_clock_cali_proc, about which the SDK says: 
> RTC clock period will change according to temperature, so RTC timer is not
very precise.

I'm going in deeper!

marcelstoer, 2015-10-21:
I suggest to either keep this issue or #439.

Aeprox, 2015-10-21:
I've adjusted your example code to the following:
``` lua
tmr.alarm(0,100,1,function ()
  tt = tmr.time()
  tn = tmr.now()
  if not(tt == tp) then
    print("tt="..tt.." tn="..tn)
    if( tt < tp) then print("WARNING: Error occured!!!") end
    tp=tt
  end
end)
```
And added some debug output into tmr.c
``` C
// Lua: tmr.time() , return rtc time in second
static int tmr_time( lua_State* L ){
	rtc_timer_update();
	NODE_DBG("rtc_time.block = %u \r\n", rtc_time.block);
	lua_pushinteger(L, rtc2sec(rtc_time.block));
	return 1; 
}

static uint32_t rtc2sec(uint64_t rtc){
	uint64_t aku = system_rtc_clock_cali_proc();
	NODE_DBG("Factor: %u \r\n", aku);
	aku *= rtc;
	return (aku>>12)/1000000;
}

```

This helped me track it down an occurance here:

>rtc_time.block = 66293249 
>Factor: 25521 
>tt=413 tn=412378776
>
>rtc_time.block = 66309311 
>Factor: 25509 
>tt=412 tn=412478827
>WARNING: Error occured!!!
>
>rtc_time.block = 66325362 
>Factor: 25512 
>tt=413 tn=412578813

Notice that Factor dips a bit between the first and third reading here? tt is calculated as follows: 

> ((rtc_time.block * Factor) >> 12) / 10000000.

So for the second reading we get

((66293249 * 25509) >> 12) / 1000000 = 412,960013 ≃ 412

If we replace the factor with values closer to the other calculations we get the following results:

((66293249 * 25510) >> 12) / 1000000 = 412,876
((66293249 * 25511) >> 12) / 1000000 = 412,892
((66293249 * 25512) >> 12) / 1000000 = 412,909
((66293249 * 25513) >> 12) / 1000000 = 412,925
((66293249 * 25514) >> 12) / 1000000 = 412,941
((66293249 * 25515) >> 12) / 1000000 = 412,957
((66293249 * 25516) >> 12) / 1000000 = 412,973
((66293249 * 25517) >> 12) / 1000000 = 412,989
((66293249 * 25518) >> 12) / 1000000 = 413,006
((66293249 * 25519) >> 12) / 1000000 = 413,022
((66293249 * 25520) >> 12) / 1000000 = 413,038
((66293249 * 25521) >> 12) / 1000000 = 413,054

So, it's the changes in system_rtc_clock_cali_proc that cause these leaps backwards. The bigger the value of the RTC timer value gets, the bigger the leaps. I've seen the factor fluctuate over 30 units in just a few seconds.

Say rtc_time.block (an unsigned 64bit integer) has gotten quite huge after being up for a few weeks (say 2^26 rtc cycles), and the factor fluctuates 20 units between calculations, then we'd get a 0,3s difference between each calculation. [I did the math](http://www.wolframalpha.com/input/?i=%28%28%282^26+*+25521%29+%3E%3E+12%29+%2F+1000000%29+-+%28%28%282^26+*+25501%29+%3E%3E+12%29+%2F+1000000%29)

There might be other things at play here, but it's a start. But as I said, I don't see a simple bugfix for this, except for getting the factor once and using that until the next reboot.

Aeprox, 2015-10-21:
Also discussed in #517

smartynov, 2015-10-22:
One more interesting observation to add: tmr.time() deviation correlates with surrounding temperature. On this charts you can see data from temperature sensor and logging of `tmr.time()` value. 
[image]

As I can suppose, temperature influences `system_rtc_clock_cali_proc` return value. Probably `tmr.time()` should not call `system_rtc_clock_cali_proc` each time, or it could smooth calibration value via some sort of recursive filter.

smartynov, 2015-10-22:
@Aeprox agree, I can see at least three issues related to `tmr.time()` problem. Probably someone should responsibly close all except one as duplicates?

danlu01, 2015-10-25:
So what's different between 0.9.5 release and the latest that tmr.time() is broken more?

It seems to work better with 0.9.5 except it would run 1 sec slow for every 80sec. I had to upgrade to latest firmware because ran into NET ON SENT callback crash.

danlu01, 2015-10-25:
workaround:

seems like tmr.alarm still work ok.  Create a tmr.alarm event loop to keep track of time.

Aeprox, 2015-10-25:
It got upgraded to use the RTC timer (I've already linked to the specific commit and discussion). You can go on using tmr.now() just fine as well btw, that still uses the same internal code as tmr.time used to do.

Aeprox, 2015-10-25:
I've gone through the tmr modules history and discovered something of interest: the current version of the code behind tmr.now() doesn't apply ```system_rtc_clock_cali_proc``` correctly. As stated in the docs, it returns a 32bit value with "a fixed point value (12 bit fraction part)". However, the fraction part is not being used.

In [an older version](https://github.com/nodemcu/nodemcu-firmware/blob/24411d34c14573cca116e243657b4098227aaa91/app/modules/tmr.c#L153-L176) it is though. I've applied that way of calculating it into the current version of the code, and the errors still occur, though not as frequent. I believe another part of the issue here is how the function works though:

Currently, the module uses an extra variable(32bit) to store the number of rtc clock (32bit) overflows, to extend the time before the whole tmr.now() function overflows. When calculating seconds from RTC memory, the current RTC clock and the overflow variable get combined into one 64bit value and that gets converted to seconds using the calibration. 

The problem arises when for example the first milion RTC cycles took 5ms each, and the next 200 took 25ms. (Stupid example, maybe you pored liquid nitrogen onto your precious ESP?). When calculating how many seconds have passed, it'll use the current calibration value (25ms/cycle) and apply that to the number of passed RTC cycles. One giant error accumulation basically.

In the piece of code I've linked above, they used a better approach that was based on calculating the seconds since the last calculation, which explains why the problem (supposedly) wast as bad before the rewrite. We'd need to do the same here somehow, storing the amount of elapsed seconds instead of elapsed RTC cycles. That would require this function to be called periodically though, to keep the amount of RTC cycles between conversions to seconds as small as possible. There might be a bigger drift, but wont longer jitter all over the place.

Thoughts anyone? I'm looking at @md5crypt a bit here, you did the rewrite. Is my thinking here reasonable? Is it even worth perfecting the code for reading an unstable RTC clock?

md5crypt, 2015-10-25:
I don't want to be rude, but you know how fixed point math works, right? "12 bit fraction part" means "multiplied by 2^12".
```c
static uint32_t rtc2sec(uint64_t rtc){
	uint64_t aku = system_rtc_clock_cali_proc(); //aku is k * 2^12
	aku *= rtc; //aku is k * rtc * 2^12
	return (aku>>12)/1000000; //aku is k * rtc / 10^6
}
```
I don't want to be rude, but you know how fixed point math works, right? "12 bit fraction part" means "multiplied by 2^12".
```c
static uint32_t rtc2sec(uint64_t rtc){
	uint64_t aku = system_rtc_clock_cali_proc(); //aku is k * 2^12
	aku *= rtc; //aku is k * rtc * 2^12
	return (aku>>12)/1000000; //aku is k * rtc / 10^6
}
```
Taking that aside, I'm no expert on RLC oscillators but I thought they where stable. Keeping a count of all RTC cycles is the only way to avoid accumulating numeric errors. If the oscillator is in fact not stable, and system_rtc_clock_cali_proc value isn't total bulls*it (I think it is) then yes, the old way is better.

jmattsson, 2015-10-27:
The clock stability and calibration is, how do I put this politely... "not quite ideal". Have a look at the [rtctime](https://github.com/DiUS/nodemcu-firmware/wiki/rtctime-module-documentation) module for an alternative.

innesm, 2016-03-12:
It might be a good idea to update the nodemcu documentation to say "don't use tmr.time(), it doesn't work".

jmattsson, 2016-03-30:
Fixed in 7b95711.

closed 2016-03-30 by jmattsson